m-labs/artiq #117: classes inheriting from HasEnvironment can't set_result()

jboulder, 2015-08-28:
Inheriting from HasEnvironment doesn't give access to the results database rdb. 

```python
__author__ = 'rabi'

from artiq import *

class Auxiliary(HasEnvironment):
    def build(self):
        self.attr_device("core")

    @kernel
    def go(self):
        delay(10*ms)
        r = [1, 2, 3]
        #self.handle_results(r)
        return r

    def handle_results(self, r):
        #not possible to set results here
        # error is ValueError: Result database not present
        self.set_result("help_results_test",
            r, store=False, realtime=False)
        pass

class ResultsTest(EnvExperiment):
    """Results Test Program"""

    def build(self):
        self.attr_device("core")
        self.aux = Auxiliary(parent=self)

    @kernel
    def run(self):
        delay(1*ms)
        r = self.aux.go()
        self.handle_results(r)

    def handle_results(self, r):
        # this does work
        self.set_result("help_results_test",
            r, store=False, realtime=False)

    def analyze(self):
        pass
```

sbourdeauducq, 2015-08-28:
fixed in ebfd8c937cb2a365c7dac19bfaf211e42b098e99

closed 2015-08-28 by sbourdeauducq

sbourdeauducq, 2015-08-28:
NB: This problem was easy to fix, but for future ones please trim your test code to the minimum required to show the issue.