m-labs/artiq #118: scan on core produces error artiq.transforms.tools.NotASTRepresentable

jboulder, 2015-08-28:
The following produces error 
```bash
  File "/home/rabi/anaconda3/lib/python3.4/site-packages/artiq/transforms/tools.py", line 65, in value_to_ast
    raise NotASTRepresentable(str(value))
artiq.transforms.tools.NotASTRepresentable: <artiq.language.scan.LinearScan object at 0x7ff979bb25c0>
```

```python
from artiq import *

class ResultsTest3(EnvExperiment):
    """Results Test Program"""

    def build(self):
        self.attr_device("core")
        self.attr_argument("xs",
                   Scannable(default=LinearScan(0, 30e-3, 50)))

    @kernel
    def run(self):
        for x in self.xs:
            delay(1*ms)
```
It would appear this precludes doing a scan on the core device. And, there is a long delay overhead.
```python
from artiq import *

class ResultsTest4(EnvExperiment):
    """Results Test Program"""

    def build(self):
        self.attr_device("core")
        self.attr_argument("xs",
                   Scannable(default=LinearScan(0, 30e-3, 50)))

    @kernel
    def run_kernel(self):
        delay(1*ns)

    def run(self):
        for x in self.xs:
            self.run_kernel()
```
```bash
rabi@britton-VirtualBox:/media/sf_git/nistpenning/calc/artiq$ time artiq_run exps/help/results_test4.py
ts: [None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None]

real	0m3.994s
user	0m3.284s
sys	0m0.152s
```
Not sure why my attempt at using ARTIQ to report timing always returned None.

sbourdeauducq, 2015-08-28:
How do you get that "ts" list?

jboulder, 2015-08-28:
I'm following the flopping_f_simulation.py example.

sbourdeauducq, 2015-08-28:
Well, there is no ``ts`` result in flopping_f_simulation, and it does not show any similar problem AFAICT.

jboulder, 2015-08-28:
Sorry. Please disregard the last bit following $ time.

sbourdeauducq, 2016-01-26:
@whitequark How much time will it take to implement scans on core?

whitequark, 2016-01-26:
@sbourdeauducq The bulk of work (support for generating calls to methods when all you have is syntax) was already done when I added support for context managers, so I'd say around two days to implement and test.

whitequark, 2016-01-26:
Oh, that's without generators, i.e. you will need to define `__iter__`/`__next__` methods. Generators will take a *lot* more time, I wouldn't even take it onto myself to estimate it because I'll need to read up more closely on how they work.

jordens, 2016-01-26:
Ack. Let's delay this.

sbourdeauducq, 2016-06-18:
Also support ``MultiScanManager``: ec3e7792dc4fba547f9812d5e4d921089916185b

sbourdeauducq, 2016-06-18:
For now, a workaround is to compute a list on the host and pass it to the device. After #478 is fixed, this workaround is also applicable to situations with several ``MultiScanManager``s.

sbourdeauducq, 2017-10-10:
Maybe an option is to implement only support for ``__iter__`` and ``__next__`` (without generators) on the core device, and rewrite the scan code accordingly. ``MultiScanManager`` may be a bit tricky though.

sbourdeauducq, 2017-10-10:
Other areas that need work:
* rewriting ``RangeScan`` so that the scan points are not pre-computed at class creation (otherwise we gain essentially nothing compared to the current solution).
* supporting ``random.shuffle`` on the device.

whitequark, 2017-10-10:
There's native support for generators now in Rust and Clang so I need to reevaluate this.

jordens, 2017-10-10:
Just doing iterators and 1d scans would go a long way.

sbourdeauducq, 2017-10-19:
@jordens OK.

sbourdeauducq, 2021-12-13:
Transferred to NAC3.

closed 2021-12-13 by sbourdeauducq