m-labs/artiq #120: Force override for DDS' in gui

raghavendrasrinivas, 2015-08-28:
Would be possible to have a force override for the DDS frequencies in the GUI similar to the TTLs? Sometimes useful when troubleshooting AOMs for example.

raghavendrasrinivas, 2015-08-28:
Is the DDS panel in the GUI also supposed to update the frequencies of the DDS as they are set in various experiments? If so, that doesn't seem to be happening. I loaded the gui and ran an experiment when I just set the DDS' frequencies to 100 MHz and it still shows as 0 MHz in the GUI.

dhslichter, 2015-08-28:
This was originally planned but it was decided at the last site visit in June that actually implementing DDS override in the same way as TTL override was going to be tricky to handle, so we would not pursue it.  The use case where one wants to set DDS frequencies manually for AOM debugging should be accomplished either by writing an experiment that just sets DDS frequencies, or by modifying the code of an existing experiment for the desired DDS behavior.  If this is an issue down the road, we can revisit the idea.

raghavendrasrinivas, 2015-08-28:
Okay. Would it be possible to just have a DDS panel that lets you set frequencies for various DDS? This wouldn't be an override to any experiments that run, just a panel that lets you send set frequency commands to the DDS.

So basically every DDS would have an input that you could enter a frequency to and a set frequency button that changes the DDS frequency to that value.

sbourdeauducq, 2015-08-29:
Write a new "experiment" and use the argument system (attr_argument and friends).

sbourdeauducq, 2015-08-29:
Please keep each issues to a one topic. If the frequency meters are not updating in the DDS panel (though it does work correctly here), this is a different problem than how to set frequencies for debugging.

sbourdeauducq, 2015-10-04:
I guess that what you are seeing is that the frequency monitor is reset after the end of the kernel execution (it is otherwise correctly updated here while the kernel is running). I will fix this, reopen if the problem is different.

commit e46ba835 (2015-10-04): rtio/dds: use rio_phy domain to reset FTW tracker. Closes #120
closed 2015-10-04 by sbourdeauducq