m-labs/artiq #124: artiq_flash.sh -t pipistrello fails to find board

jboulder, 2015-09-01:
udev rule applied. problem persists following reboot. 
```bash
rabi@britton-VirtualBox:~$ artiq_flash.sh -t pipistrello
Flashing failed. Maybe you do not have permission to access the USB device?
To fix this you might want to add a udev rule by doing:
$ sudo cp /home/rabi/anaconda3/lib/python3.4/site-packages/artiq/misc/99-papilio.rules /etc/udev/rules.d
Then unplug/replug your device and try flashing again

Other reason could be that you chosed the wrong target
Please make sure you used the correct -t option (currently: pipistrello)
rabi@britton-VirtualBox:~$ conda list artiq
# packages in environment at /home/rabi/anaconda3:
#
artiq                     0.0                     np_1521  
llvmlite-artiq            0.5.1                    py34_4  
rabi@britton-VirtualBox:~$ dmesg
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.19.0-26-generic (buildd@lgw01-04) (gcc version 4.8.2 (Ubuntu 4.8.2-19ubuntu1) ) #28~14.04.1-Ubuntu SMP Wed Aug 12 14:09:17 UTC 2015 (Ubuntu 3.19.0-26.28~14.04.1-generic 3.19.8-ckt4)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-3.19.0-26-generic root=UUID=6f255724-bf75-4ddf-b329-d78c630c080c ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000005dbeffff] usable
[    0.000000] BIOS-e820: [mem 0x000000005dbf0000-0x000000005dbfffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[    0.000000] Hypervisor detected: KVM
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] AGP: No AGP bridge found
[    0.000000] e820: last_pfn = 0x5dbf0 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR variable ranges disabled:
[    0.000000] PAT configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- UC  
[    0.000000] CPU MTRRs all blank - virtualized system.
[    0.000000] found SMP MP-table at [mem 0x0009fff0-0x0009ffff] mapped at [ffff88000009fff0]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff880000099000] 99000 size 24576
[    0.000000] init_memory_mapping: [mem 0x00000000-0x000fffff]
[    0.000000]  [mem 0x00000000-0x000fffff] page 4k
[    0.000000] BRK [0x01fd4000, 0x01fd4fff] PGTABLE
[    0.000000] BRK [0x01fd5000, 0x01fd5fff] PGTABLE
[    0.000000] BRK [0x01fd6000, 0x01fd6fff] PGTABLE
[    0.000000] init_memory_mapping: [mem 0x5d800000-0x5d9fffff]
[    0.000000]  [mem 0x5d800000-0x5d9fffff] page 2M
[    0.000000] BRK [0x01fd7000, 0x01fd7fff] PGTABLE
[    0.000000] init_memory_mapping: [mem 0x40000000-0x5d7fffff]
[    0.000000]  [mem 0x40000000-0x5d7fffff] page 2M
[    0.000000] init_memory_mapping: [mem 0x20000000-0x3fffffff]
[    0.000000]  [mem 0x20000000-0x3fffffff] page 2M
[    0.000000] init_memory_mapping: [mem 0x00100000-0x1fffffff]
[    0.000000]  [mem 0x00100000-0x001fffff] page 4k
[    0.000000]  [mem 0x00200000-0x1fffffff] page 2M
[    0.000000] init_memory_mapping: [mem 0x5da00000-0x5dbeffff]
[    0.000000]  [mem 0x5da00000-0x5dbeffff] page 4k
[    0.000000] BRK [0x01fd8000, 0x01fd8fff] PGTABLE
[    0.000000] RAMDISK: [mem 0x35a2c000-0x36d0dfff]
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000E0000 000024 (v02 VBOX  )
[    0.000000] ACPI: XSDT 0x000000005DBF0030 00003C (v01 VBOX   VBOXXSDT 00000001 ASL  00000061)
[    0.000000] ACPI: FACP 0x000000005DBF00F0 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
[... 466 lines cut ...]
[60543.090362] 00:00:00.042197 automount VBoxServiceAutoMountWorker: Shared folder "virtualBoxShared" was mounted to "/media/sf_virtualBoxShared"
[60544.788786] init: plymouth-upstart-bridge main process ended, respawning
[60544.818546] init: plymouth-upstart-bridge main process ended, respawning
[60569.823069] audit_printk_skb: 105 callbacks suppressed
[60569.823072] audit: type=1400 audit(1441120767.456:47): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/lib/cups/backend/cups-pdf" pid=2177 comm="apparmor_parser"
[60569.823078] audit: type=1400 audit(1441120767.456:48): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=2177 comm="apparmor_parser"
[60569.823817] audit: type=1400 audit(1441120767.456:49): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=2177 comm="apparmor_parser"
[60702.145425] usb 1-2: USB disconnect, device number 3
[60702.145533] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[60702.145544] ftdi_sio 1-2:1.0: device disconnected
[60702.149074] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
[60702.149087] ftdi_sio 1-2:1.1: device disconnected
[60733.653000] usb 1-2: new high-speed USB device number 4 using xhci_hcd
[60733.947547] usb 1-2: New USB device found, idVendor=0403, idProduct=6010
[60733.947551] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[60733.947552] usb 1-2: Product: Pipistrello LX45
[60733.947554] usb 1-2: Manufacturer: Saanlima
[60733.947555] usb 1-2: SerialNumber: 400100527227
[60733.955805] ftdi_sio 1-2:1.0: FTDI USB Serial Device converter detected
[60733.955826] usb 1-2: Detected FT2232H
[60733.957812] usb 1-2: FTDI USB Serial Device converter now attached to ttyUSB0
[60733.960159] ftdi_sio 1-2:1.1: FTDI USB Serial Device converter detected
[60733.960265] usb 1-2: Detected FT2232H
[60733.963175] usb 1-2: FTDI USB Serial Device converter now attached to ttyUSB1
```

fallen, 2015-09-01:
Can you run:
`$ lsusb -t | grep ftdi_sio`
It should print something like:
`|__ Port 1: Dev 13, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 480M`
But with different "Port" and "Dev" value.
Can you then do: 
`$ ls -lh /dev/bus/usb/001` if Port is 1

fallen, 2015-09-04:
It's difficult to find the real issue without your help.

sbourdeauducq, 2015-09-07:
Do you have xc3sprog installed? Also, the script should display a better error message when xc3sprog is not found.

jboulder, 2015-09-09:
I'm away from Boulder. I'll be back on Thursday and will check this then.
Joe

sbourdeauducq, 2015-10-17:
Just tested it OK with the latest conda package (1626).

closed 2015-10-17 by sbourdeauducq