m-labs/artiq #125: clang: error: unknown argument: '-mffl1' (...-maddc)

sbourdeauducq, 2015-09-04:
Following the Clang instructions currently in the manual results in a compiler that does this when compiling the runtime:
```
clang: error: unknown argument: '-mffl1'
clang: error: unknown argument: '-maddc' 
```

whitequark, 2015-09-07:
I have just built LLVM and Clang from scratch according to instructions in the manual and the following works:
```
touch x.c
./bin/clang -c -target or1k -mffl1 x.c
```

sbourdeauducq, 2015-09-07:
@raghavendrasrinivas Can you let us know exactly what you did to get that broken compiler?

raghavendrasrinivas, 2015-09-10:
I did this,

1. 
$ cd ~/artiq-dev
$ git clone https://github.com/m-labs/migen
$ cd migen
$ python3 setup.py develop --user

2. $ cd ~/artiq-dev
$ git clone https://github.com/openrisc/llvm-or1k
$ cd llvm-or1k/tools
$ git clone https://github.com/openrisc/clang-or1k clang
$ cd ..

$ mkdir build
$ cd build
$ cmake .. -DCMAKE_INSTALL_PREFIX=/usr/local/llvm-or1k -DLLVM_TARGETS_TO_BUILD="OR1K;X86" -DCMAKE_BUILD_TYPE=Rel -DLLVM_ENABLE_ASSERTIONS=ON
$ make -j4
$ sudo make install

3. 
cd ~/artiq-dev/artiq/soc/runtime && make runtime.fbi

Was step 3 that generated the error.

whitequark, 2015-09-11:
Ohh. I see now. common.mak assumes that `clang` in PATH has the right target in it, which is not the case if you install everything to /usr/local/llvm-or1k. Let me amend the instructions...

whitequark, 2015-09-11:
Oh, it's already there.

> Note Make sure that /usr/local/llvm-or1k/bin is first in your PATH, so that the clang command you just built is found instead of the system one, if any.

sbourdeauducq, 2015-10-21:
Just tested again with a fresh LLVM install built as the manual says, everything (BIOS and runtime) compiles fine. @raghavendrasrinivas reopen if you can reproduce the problem.

closed 2015-10-21 by sbourdeauducq