m-labs/artiq #126: pxi6733: ping returns False

kcmccorm, 2015-09-04:
File "C:\Users\rabi\AppData\Local\Continuum\Anaconda3\lib\site-packages\artiq-
0.0+dev-py3.4.egg\artiq\devices\pxi6733\driver.py", line 54, in ping
    self.daq.DAQmxGetDevSerialNum(self.device, data)
AttributeError: 'DAQmx' object has no attribute 'device'

fallen, 2015-09-04:
I just pushed this change: https://github.com/m-labs/artiq/commit/7dfd11e2f0c47b0b2b7303c4c76ee6ffa7e9d39e
Could you give this a try when the Windows conda package will be uploaded? (hopefully in a few minutes)
In any case, if it fails or not, please post the console output of the controller.

fallen, 2015-09-04:
The Windows package is uploaded, you can update by doing : conda update artiq

fallen, 2015-09-04:
I updated again so that the debug message is only printed when controller is run in verbose mode.
So please when you update your artiq package, in order to test run the controller in verbose mode with `-vv` argument

sbourdeauducq, 2016-01-26:
It seems no one really cares about the PXI6733, so I'd remove support for it from the stable branches until we have a plan for testing.

jordens, 2016-01-26:
@kcmccorm @dhslichter @jboulder ping

dhslichter, 2016-02-01:
Are other groups, e.g. JQI, using these devices?

jboulder, 2016-02-01:
I don't know of anyone using it except Katie. Katie, do you have continued
interest in the NI 6733? -Joe

dhslichter, 2016-02-01:
After further discussion, we have decided that we will only use the PDQ with ARTIQ for this system, and will not try to use the PXI6733 with ARTIQ, so it is OK to remove it from the stable distributions.  Perhaps other groups using NI hardware may benefit from aspects of the code already in place.

sbourdeauducq, 2016-02-01:
What about putting it into a separate repository?

jboulder, 2016-02-01:
I just had a conversation with everyone about this. The plan is to switch
over to PDQs when we make the switch to ARTIQ.

Katie

dhslichter, 2016-02-01:
If putting it into a separate repository is helpful to you, that should be OK from our end.

jordens, 2016-02-02:
Leave it in a branch. We can resurrect it later.

sbourdeauducq, 2016-02-02:
I don't like stale branches laying around, and the repository hints that you can have external ARTIQ controllers/drivers. People regularly ask how to extend ARTIQ; even if the code is untested, it should be easy to have a good structure so that the repository can be used as example.

sbourdeauducq, 2016-02-02:
https://github.com/m-labs/aq_ni6733

closed 2016-02-02 by sbourdeauducq

jordens, 2016-02-02:
A non-working example of how to extend artiq? Doesn't sound like a good idea and seems less "likeable" than a branch. And why would would it be less "stale"? Most people would prefer branches very much because they are much easier to handle then a bunch of additional repositories. There is nothing bad about branches.

jordens, 2016-02-02:
You also need to migrate the issues.