m-labs/artiq #128: fancy scanning widget

jordens, 2015-09-05:
Prototype design for a fancier scanning widget.

https://goo.gl/photos/AG7fPLGoACcZ8h4W9

sbourdeauducq, 2015-09-18:
The purpose of this widget is to visually edit the main scan parameters: minimum, maximum and number of points.

The graduated axis represents all real numbers. The minimum and maximum of the scan are defined by two triangles. Those triangles can be dragged using the mouse to modify the corresponding value.

In addition to the graduation, the axis has dots that represent the values (iteration points) that the scanned variable will take during scanning.

Increasing or decreasing the number of points is done using shift+drag. The drag feels like ejecting or pulling points: when in the right part of the widget, dragging from left to right decreases the point count, dragging right to left increases it. When in the left part, the drag directions are reversed.

Zooming on the whole widget is done using the mouse wheel like in pyqtgraph. There is also a button to fit the view to the figure similar to the A button in pyqtgraph, and a button to set the scan min/max to 1/3 and 2/3 of the current view.

The widget should be synchronized with the three spin boxes.

jordens, 2016-02-23:
https://github.com/m-labs/scanwidget

jordens, 2016-02-25:
preview: https://github.com/m-labs/artiq/tree/scanwidget
todo: https://github.com/m-labs/scanwidget/blob/master/todo.md

commit d34d83f3 (2016-03-11): Merge branch 'scanwidget' (closes #128)

* scanwidget:
  gui: fix scanwidget usage
  scanwidget: apply changes as of 10439cb
  scanwidget: apply changes as of 98f0a56
  missing parts of 59ac567
  scanwidget: disable unmodified wheel on axis and slider
  scanwidget: wire up signals better, set values late, take scanwidget from 7aa6397
  gui: use scanwidget
  scanwidget: add from current git
closed 2016-03-11 by jordens