m-labs/artiq #133: conda install fails due to unsatisifiable dependency

jboulder, 2015-10-09:
```bash
rabi@68810artiq1:~/Downloads/pycharm-community-4.5.4/bin$ conda install artiq
Fetching package metadata: ......
Solving package specifications: .
Error: Unsatisfiable package specifications.
Generating hint: 
[      COMPLETE      ]|########################################################################| 100%

Hint: 'artiq' has unsatisfiable dependencies (see 'conda info artiq')
```

Looks like this may be python3.5 dependency. See below. 

```bash
rabi@68810artiq1:~/Downloads/pycharm-community-4.5.4/bin$ conda install pyqtgraph quamash
Fetching package metadata: ......
Solving package specifications: .
Error: Unsatisfiable package specifications.
Generating hint: 
[      COMPLETE      ]|########################################################################| 100%

Hint: the following packages conflict with each other:
  - pyqtgraph
  - quamash

Use 'conda info pyqtgraph' etc. to see the dependencies for each package.
rabi@68810artiq1:~/Downloads/pycharm-community-4.5.4/bin$ conda info pyqtgraph
Fetching package metadata: ......

pyqtgraph 0.9.10.1036edf py35_0
-------------------------------
file name   : pyqtgraph-0.9.10.1036edf-py35_0.tar.bz2
name        : pyqtgraph
version     : 0.9.10.1036edf
build number: 0
build string: py35_0
channel     : http://conda.anaconda.org/m-labs/channel/dev/linux-64/
size        : 899 KB
arch        : x86_64
binstar     : {'package_id': '55cc4b6959da985898fd289f', 'channel': 'dev', 'owner_id': '55cc3c85d2626f681e0ee47f'}
license     : MIT License
machine     : x86_64
md5         : df2e08c4990fd5094a8b88373830feef
operatingsystem: linux
platform    : linux
subdir      : linux-64
target-triplet: x86_64-any-linux
installed environments:
dependencies:
    numpy
    pyqt >=4.7
    python 3.5*       <<<<<<<<<<==========================
rabi@68810artiq1:~/Downloads/pycharm-community-4.5.4/bin$ conda install quamash
Fetching package metadata: ......
Solving package specifications: ............
Package plan for installation in environment /home/rabi/anaconda3:

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    sip-4.16.9                 |           py34_0         222 KB
    qt-4.8.7                   |                1        36.4 MB
    pyqt-4.11.4                |           py34_0         3.5 MB
    quamash-0.5.1              |           py34_0          29 KB
    ------------------------------------------------------------
                                           Total:        40.2 MB

The following NEW packages will be INSTALLED:

    quamash: 0.5.1-py34_0 

The following packages will be UPDATED:

    pyqt:    4.11.3-py34_1 --> 4.11.4-py34_0
    qt:      4.8.6-3       --> 4.8.7-1      
    sip:     4.16.5-py34_0 --> 4.16.9-py34_0

Proceed ([y]/n)? n^[[A^C
Operation aborted.  Exiting.
rabi@68810artiq1:~/Downloads/pycharm-community-4.5.4/bin$ conda info quamash
Fetching package metadata: ......

quamash 0.5.1 py34_0
--------------------
file name   : quamash-0.5.1-py34_0.tar.bz2
name        : quamash
version     : 0.5.1
build number: 0
build string: py34_0
channel     : http://conda.anaconda.org/m-labs/channel/dev/linux-64/
size        : 29 KB
arch        : x86_64
binstar     : {'channel': 'dev', 'package_id': '55cc48b459da985879fd28cf', 'owner_id': '55cc3c85d2626f681e0ee47f'}
license     : BSD
machine     : x86_64
md5         : ccd404708c51a0701227f41a6d0a3557
operatingsystem: linux
platform    : linux
subdir      : linux-64
target-triplet: x86_64-any-linux
installed environments:
dependencies:
    pyqt 4.*
    python 3.4*

```

whitequark, 2015-10-09:
OK, this is something I caused. There are [both py34 and py35 builds](https://anaconda.org/m-labs/pyqtgraph/files) of pyqtgraph uploaded, but it seems that conda's dependency resolver is unable to select the right one.

In retrospect the py35 migration should have been done in a separate channel.

I should finish the migration in a moment.

whitequark, 2015-10-12:
The migration has been finished.

jboulder, 2015-10-12:
I confirm that the install now works.

Note for other users of conda deployment. Due to ARTIQ's new python 3.5
dependency you'll have to create a new environment for python 3.5 and
install ARTIQ in it.

```bash
(root)rabi@688penningA:~/git/m-labs/artiq$ conda create --name py5
python=3.5
(root)rabi@688penningA:~/git/m-labs/artiq$ source activate py5
(py5)rabi@688penningA:~/git/m-labs/artiq$ conda install artiq
```

Changes in configuration files:
* ddb.pyon -> device_db.pyon
* pdb.pyon -> dataset_db.pyon

Changes in program files:
* attr_device -> setattr_device
* attr_argument -> setattr_argument
See: https://github.com/m-labs/artiq/issues/136

closed 2015-10-12 by whitequark