m-labs/artiq #134: ARTIQ network addresses can map to localhost

jboulder, 2015-10-11:
Suppose the IP address of my ARTIQ Master machine is x.x.x.101 and in DNS it is called x.x.artiq_master. The user may well reasonably expect that the following are synonymous. 
* x.x.x.101
* x.x.artiq_master

For example, the following works on the same machine. 
$ artiq_master --bind x.x.x.101 
$ artiq_ctlmgr --bind x.x.x.101 --server x.x.x.101

The following does not work on the same machine. 
$ artiq_master --bind x.x.x.101 
$ artiq_ctlmgr --bind x.x.artiq_master --server x.x.artiq_master

The following does not work on the same machine. 
$ artiq_master --bind ::1
$ artiq_ctlmgr --bind 127.0.0.1 --server 127.0.0.1

Recommendation: Bind to network-facing IP addresses, never to 127.0.0.1 or ::1 or "localhost" (or DNS names).

sbourdeauducq, 2015-10-12:
::1 is the IPv6 address and 127.0.0.1 is the v4 address - binding to the IPv6 address and connecting on the v4 address predictably does not work.

sbourdeauducq, 2015-10-12:
The second example works correctly here, both with the IP address and the hostname in the device database. What are the contents of that DNS entry?

sbourdeauducq, 2015-10-12:
Binding to network-facing IP addresses is problematic:
* when running an ARTIQ server on a untrusted network (e.g. public wifi) there are security issues.
* there can be several network cards in the same machine (though we can bind on all).

jboulder, 2015-10-12:
>The following does not work on the same machine.
>$ artiq_master --bind x.x.x.101
>$ artiq_ctlmgr --bind x.x.artiq_master --server x.x.artiq_master

The following are run on the same machine.
```
(artiq_local)rabi@688penningA:~/gitlab/nistpenning/artiq$ artiq_master --bind 132.163.82.151&
(artiq_local)rabi@688penningA:~/git/m-labs/artiq/examples/master$ artiq_ctlmgr --bind 688penninga --server 688penninga
WARNING:artiq.frontend.artiq_ctlmgr:Connection to master failed (ConnectionRefusedError: [Errno 111] Connect call failed ('127.0.1.1', 3250))
WARNING:artiq.frontend.artiq_ctlmgr:Retrying in 5.0 seconds
WARNING:artiq.frontend.artiq_ctlmgr:Connection to master failed (ConnectionRefusedError: [Errno 111] Connect call failed ('127.0.1.1', 3250))
WARNING:artiq.frontend.artiq_ctlmgr:Retrying in 5.0 seconds
```

```
(artiq_local)rabi@688penningA:~/git/m-labs/artiq/examples/master$ ping 688penninga
PING 688penningA (127.0.1.1) 56(84) bytes of data.
64 bytes from 688penningA (127.0.1.1): icmp_seq=1 ttl=64 time=0.039 ms
64 bytes from 688penningA (127.0.1.1): icmp_seq=2 ttl=64 time=0.066 ms
```
DNS 
```
688penninga.bw.nist.gov 	132.163.82.151
```

sbourdeauducq, 2015-10-12:
Use ``artiq_ctlmgr --bind 688penninga.bw.nist.gov --server 688penninga.bw.nist.gov``?
Is ``688penninga.bw.nist.gov`` resolving to the right address (``132.163.82.151``), not ``127.0.1.1``?

jboulder, 2015-10-12:
Yes. That works!

If it would help others avoid these pitfalls it might be good to update the
docs to require complete domain names.

closed 2015-10-12 by sbourdeauducq

jboulder, 2015-10-13:
> Closed #134 <https://github.com/m-labs/artiq/issues/134> via ceb18d0

Great!

commit ceb18d05 (2015-10-12): manual/management: add warning about setups that resolve the current hostname to localhost. closes #134