m-labs/artiq #135: ARTIQ build (still) broken

jboulder, 2015-10-12:
https://anaconda.org/m-labs/artiq/files

linux-32/artiq-0.0-np_1536.tar.bz2 is 1 month 2 days old

whitequark, 2015-10-12:
There is now a build 1579 that should work.

jboulder, 2015-10-12:
Confirmed build now works. I can install 1589 from conda. 

Is there a way of seeing the relationship between commit ID and build number?
https://github.com/m-labs/artiq/commits/master
https://anaconda.org/m-labs/artiq/files

whitequark, 2015-10-12:
Not really, since while build number is derived from the output of `git describe`, it is ambiguous. We'd need to include the hash into the build number, conda allows to do that.

sbourdeauducq, 2015-10-20:
@whitequark Can you make the platform-independent packages noarch in conda-recipes (most of them AFAICT), and upload linux32 and win32 versions for the other ones in addition to linux64?

whitequark, 2015-10-20:
Will do.

whitequark, 2015-10-20:
Well, every package that builds for linux32 and win32 is already uploaded, so that does not have to be done.

sbourdeauducq, 2015-10-21:
@whitequark Nevertheless ARTIQ still cannot be installed on linux32 and win32. I tried to make it noarch, which would be the proper way of fixing this, but this tickles yet another conda bug. Can you have a look at it?

sbourdeauducq, 2015-10-21:
https://github.com/conda/conda-build/issues/542

whitequark, 2015-10-21:
I don't really see a point in bothering with noarch then, just build it thrice.

sbourdeauducq, 2015-10-21:
The bitstream being included in the same package makes this approach somewhat slow and annoying (especially with the non-reproducibility of bitstream compilation by ISE and Vivado). How is #146 progressing?

whitequark, 2015-10-22:
Everything in still issue should be done, #146 is in progress.

closed 2015-10-22 by whitequark