m-labs/artiq #136: m-labs.hk/artiq/manual rebuild is not automatic

jboulder, 2015-10-12:
Please automatically update manual as part of build. 

http://m-labs.hk/artiq/manual

Out of date manual made it harder to notice https://github.com/m-labs/artiq/commit/50a5a3baf0b3e4524b2f5b7d47ce62b4bf4212f1 .
```
environment: attr_* -> setattr_* 
```

sbourdeauducq, 2015-12-02:
@whitequark can you implement automatic doc upload on the new build-server?

whitequark, 2015-12-02:
@sbourdeauducq yes but I would like you to set up a separate account for doc upload

sbourdeauducq, 2015-12-03:
You should be able to use a dedicated SFTP account (SSH file transfer) with the credentials in my home folder on the buildserver.

whitequark, 2015-12-17:
It is actually impossible to use `sftp` in batch mode using password authentication.

sbourdeauducq, 2015-12-18:
Use ``expect`` or some other hack.

whitequark, 2016-01-16:
Done

closed 2016-01-16 by whitequark