m-labs/artiq #138: conda: aiohttp, dateutil dependencies unsatisfied

jboulder, 2015-10-16:
```bash
(py35)rabi@mahxu:~/gitlab$ conda install artiq
Fetching package metadata: ......
Solving package specifications: ..........................
Error: Could not find some dependencies for artiq: aiohttp, dateutil

Did you mean one of these?

    artiq

Did you mean one of these?

    dateutil

You can search for this package on anaconda.org with

    anaconda search -t conda dateutil

 (and similarly for the other packages)

You may need to install the anaconda-client command line client with

    conda install anaconda-client
```

```bash
(py35)rabi@mahxu:~/gitlab$ conda info
Current conda install:

             platform : linux-64
        conda version : 3.18.3
  conda-build version : 1.11.0
       python version : 3.4.3.final.0
     requests version : 2.8.1
     root environment : /home/rabi/anaconda3  (writable)
  default environment : /home/rabi/anaconda3/envs/py35
     envs directories : /home/rabi/anaconda3/envs
        package cache : /home/rabi/anaconda3/pkgs
         channel URLs : http://conda.anaconda.org/m-labs/channel/dev/linux-64/
                        http://conda.anaconda.org/m-labs/channel/dev/noarch/
                        https://repo.continuum.io/pkgs/free/linux-64/
                        https://repo.continuum.io/pkgs/free/noarch/
                        https://repo.continuum.io/pkgs/pro/linux-64/
                        https://repo.continuum.io/pkgs/pro/noarch/
          config file : /home/rabi/.condarc
    is foreign system : False
```

whitequark, 2015-10-16:
I've been rebuilding all of ARTIQ's dependencies on our new, (more) automated build infrastructure, based on Appveyor and Travis CI servers.

As a part of this, I've changed the way channels are used: packages deemed 'stable' are put in the `main` branch, and only the automatically built migen and artiq packages are put in the `dev` branch. So please add the `main` channel as well.

Instructions somewhere probably need to be updated; I will look into it.

jboulder, 2015-10-16:
http://www.m-labs.hk/artiq/manual/installing.html#installing-using-conda

closed 2015-10-16 by whitequark

jboulder, 2015-10-16:
OK. I can install using the main branch. But main is built for an ARTIQ
revision that predates several of the recent API changes. For example, the
main build still wants ddb.pyon.

And it looks like there are two copies of artiq in the main build: 1599 and
np_1420.

https://conda.anaconda.org/m-labs/linux-64

whitequark, 2015-10-16:
I mean you need both:
> So please add the main channel as well.

whitequark, 2015-10-16:
This is the same general idea as Ubuntu main repository (stable stuff) and ppa's (automatic builds, possibly unstable).

commit f3e61e26 (2015-10-16): conda: move everything except artiq to conda-recipes (closes #138).