m-labs/artiq #139: artiq_gui PyQt4.QtCore has no attribute QSortFilterProxyModel

kgilmo, 2015-10-16:
Hi I'm Kevin. I'm a new student on the Penning trap. Working with Joe on ARTIQ. 

```bash
(py35)rabi@688penningA:~/gitlab/nistpenning/artiq$ artiq_gui --help
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/py35/bin/artiq_gui", line 4, in <module>
    from artiq.frontend.artiq_gui import main
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/frontend/artiq_gui.py", line 20, in <module>
    from artiq.gui.log import LogDock
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/gui/log.py", line 70, in <module>
    class _LevelFilterProxyModel(QtCore.QSortFilterProxyModel):
AttributeError: module 'PyQt4.QtCore' has no attribute 'QSortFilterProxyModel'
```

```bash
artiq 0.0 1619
--------------
file name   : artiq-0.0-1619.tar.bz2
name        : artiq
version     : 0.0
build number: 1619
build string: 1619
channel     : http://conda.anaconda.org/m-labs/channel/dev/linux-64/
size        : 1.8 MB
arch        : x86_64
binstar     : {'package_id': '55cc47427d5d4d56ff3e4029', 'owner_id': '55cc3c85d2626f681e0ee47f', 'channel': 'dev'}
license     : 3-clause BSD
machine     : x86_64
md5         : ca686116cb1a68cbec8b78cb6ebd4c3b
operatingsystem: linux
platform    : linux
subdir      : linux-64
target-triplet: x86_64-any-linux
installed environments:
    /home/rabi/anaconda3/envs/py35
dependencies:
    aiohttp
    binutils-or1k-linux
    dateutil
    flterm
    h5py
    llvmlite-artiq
    numpy
    prettytable
    pydaqmx
    pyelftools
    pygit2
    pyqtgraph
    pyserial
    python >=3.5.0
    quamash
    scipy
    sphinx
    sphinx-argparse
```

```bash
(py35)rabi@688penningA:~/gitlab/nistpenning/artiq$ conda info
Current conda install:

             platform : linux-64
        conda version : 3.17.0
  conda-build version : 1.14.1
       python version : 3.4.3.final.0
     requests version : 2.7.0
     root environment : /home/rabi/anaconda3  (writable)
  default environment : /home/rabi/anaconda3/envs/py35
     envs directories : /home/rabi/anaconda3/envs
        package cache : /home/rabi/anaconda3/pkgs
         channel URLs : http://conda.anaconda.org/m-labs/channel/main/linux-64/
                        http://conda.anaconda.org/m-labs/channel/main/noarch/
                        http://conda.anaconda.org/m-labs/channel/dev/linux-64/
                        http://conda.anaconda.org/m-labs/channel/dev/noarch/
                        https://repo.continuum.io/pkgs/free/linux-64/
                        https://repo.continuum.io/pkgs/free/noarch/
                        https://repo.continuum.io/pkgs/pro/linux-64/
                        https://repo.continuum.io/pkgs/pro/noarch/
          config file : /home/rabi/.condarc
    is foreign system : False

```

sbourdeauducq, 2015-10-17:
What is happening here is that some Qt bindings have ``QSortFilterProxyModel`` in ``QtGui``, whereas others have it in ``QtCore`` ...

sbourdeauducq, 2015-10-17:
da83212c79576734bf80fc3bfc0ae9b14057c06a autodetects the correct Qt module, and I checked the GUI now works with both the PyQt5 and PyQt4 bindings. Can you upgrade, test and close the issue if OK?

jboulder, 2015-10-17:
Confirmed fixed. I don't own the issue so can't close it.

closed 2015-10-17 by whitequark