m-labs/artiq #142: conda version numbers

jboulder, 2015-10-16:
Use git commit hash instead as build number for conda. 
Assign incremented counter as conda version number e.g 0.0.2000.
Discussed in #135. 

This permits roll-back to specific version using conda. 
http://conda.pydata.org/docs/faq.html#managing-packages

Use case, right now I'd like to use conda to roll back to a version where the GUI isn't broken. #139

whitequark, 2015-10-16:
@jboulder, I will do this, but it is not at all enough. The proper way to fix this issue is to fix the versions of *all* dependencies on deploy (or publishing, in this case). You can see this being done in package managers such as rubygems or opam or even npm. Unfortunately it seems that conda does not have a decent way of doing that, so even if you roll back artiq package itself, all dependencies will remain at the same version, and if the bug was in one, you will not be able to return to the working state.

justboh, 2015-10-19:
This kind of 'roll back' feature is attractive for me as a scientist trying to use the development versions until we get to the point of a 'stable release' situation.  If the conda rollback is not enough, what do we need to do to get this feature? Seems important for getting more people to use ARTIQ.

sbourdeauducq, 2015-10-20:
@justboh You should be able to use the environment feature of Anaconda for this:
http://conda.pydata.org/docs/using/envs.html
For example, keep an environment with a known-good version of ARTIQ and its dependencies, and another one with the latest.

whitequark, 2015-10-21:
> Use git commit hash instead as build number for conda. 
> Assign incremented counter as conda version number e.g 0.0.2000.

You cannot give `conda build` a build number, so git commit hash will become a part of the version number instead. The build number will stay unchanged.

whitequark, 2015-10-21:
Done for migen.

whitequark, 2015-10-21:
Actually, I just realized this is a bad idea, because conda first looks at the "most recent" version, and now we have a version like 0.0.gabcdef, which would be considered "more recent" than 0.0.g012345, even if commit 012345 is more recent. So we should also put the build number into the version string, which is really awkward.

jboulder, 2015-10-21:
Can you use a version number like 0.vn.hash where vn is the version number
that gets incremented? -Joe

Here's what it looks like from my end presently.
```bash
(py35)jwbritto@68810artiq2:~$ conda install migen
Fetching package metadata: ........
Solving package specifications: ...............
Package plan for installation in environment
/home/jwbritto/anaconda3/envs/py35:

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    migen-0.1                  |         py35_103         148 KB

The following NEW packages will be INSTALLED:

    migen: 0.1-py35_103

Proceed ([y]/n)? y

Fetching packages ...
migen-0.1-py35 100% |################################| Time: 0:00:00 896.51
kB/s
Extracting packages ...
[      COMPLETE      ]|###################################################|
100%
Linking packages ...
[      COMPLETE      ]|###################################################|
100%
```bash

On Wed, Oct 21, 2015 at 11:28 AM, whitequark <notifications@github.com>
wrote:

> Done for migen.
>
> —
> Reply to this email directly or view it on GitHub
> <https://github.com/m-labs/artiq/issues/142#issuecomment-149969499>.
>

whitequark, 2015-10-21:
That's odd, it should have installed `migen-0.1.gecd04cf-py_111`. Why did conda not look for the noarch package, I wonder?

whitequark, 2015-10-22:
So the reason conda doesn't select the noarch package is because it considers `0.1.gecd04cf` as a lower version than `0.1`, presumably because it assumes semver and alphanumeric identifiers in semver mean prerelease versions.

whitequark, 2015-10-22:
Ok, so it is actually possible to install a specific build using undocumented syntax `conda install "migen=0.1=py35_93"`. https://groups.google.com/a/continuum.io/d/msg/conda/s6GbcODB8D0/NgR4nX1qiZIJ

sbourdeauducq, 2015-10-22:
What's wrong with 0.1.build_number.git_hash ?

whitequark, 2015-10-22:
Nothing technically; it is less idiomatic to conda, it is convenient to be able to do `conda install migen=0.0`, and it will break anaconda.org's UI when there are thousands of builds because of how they treat versions vs build numbers.

whitequark, 2015-10-22:
Done in artiq and migen.

closed 2015-10-22 by whitequark