m-labs/artiq #148: IPv6 address in network device support packages reference

kgilmo, 2015-10-20:
In controller usage examples (and presumably elsewhere) "::1" is used, mapping to localhost

See #134

sbourdeauducq, 2015-10-20:
True. What is the problem with that?

kgilmo, 2015-10-20:
For pedagogical reasons, writing examples with the full domain name / IP address would be much clearer. Since the localhost may not be the correct server and someone new to this sort of thing may not recognize "::1" as corresponding to the localhost, being a bit more clear in the documentation would be helpful.

sbourdeauducq, 2015-10-20:
The IP address and hostnames are obviously site-dependent, such examples would work only on one machine.

jboulder, 2015-10-20:
https://github.com/m-labs/artiq/commit/ceb18d05b3ebf47bc2a02875b2b869fc0f6297ad

ACK

Another new ARTIQ user ran into this problem. It would help if NDSP
examples use an obvious placeholder for fully specified DNS name. e.g.
artiqmaster.mydomain.com

commit d35bc5a4 (2015-10-20): examples/device_db: add comment about controller hosts. Closes #148
closed 2015-10-20 by sbourdeauducq