m-labs/artiq #149: thorlabs_tcube.driver.get_encoder_counter() returns None

kgilmo, 2015-10-20:
Confirmed that get_encoder_counter() returns int when using stand-alone thorlabs_tcube_controller and artiq_rpctool. However, when accessing device in experiment a call to self.thor_tcube.get_encoder_counter() returns None. This is also true of self.thor_tcube.get_dc_status_update().

jboulder, 2015-10-20:
```python
from artiq import *

class ThorFun(EnvExperiment):
    """Thorlabs Fun"""

    def build(self):
        #self.init_logger()
        #  self.setattr_device("scheduler")
        self.thor_tcube = self.get_device("thor_tcube")

    def run(self):
        move = self.thor_tcube.move_relative(10000)
        count = self.thor_tcube.get_encoder_counter()
        count2 = self.thor_tcube.get_encoder_counter()
        # x = count*2.9e-9
        s = "Distance in m: {}".format(type(count2))
        print(s)
```

sbourdeauducq, 2015-10-21:
Are you using "best_effort: true" in device_db?

sbourdeauducq, 2015-10-21:
My guess is you are, and it fails to connect and therefore can only return None (it should print a warning in the log too). I just checked that BestEffortClient returns the correct values when the connection is successful.

kgilmo, 2015-10-21:
Yes, I was using "best_effort:true" in device_db. I see, so it's a connection issue that's suppressed with BestEffortClient and returns None.

sbourdeauducq, 2015-10-21:
If you have confirmed that, can you close the issue?

closed 2015-10-21 by kgilmo