m-labs/artiq #151: gui log scroll bar stuck at top

kgilmo, 2015-10-20:
When buffer is getting full, the scroll bar in the log gets stuck at the top while new messages are being output.

[image]

sbourdeauducq, 2015-10-21:
Works fine here. The scrollbar is not "stuck" at the top, I can move it down. Are you referring to being unable to view previous messages? The log is limited to the latest 1000 messages; we can easily increase that limit.

kgilmo, 2015-10-21:
Sorry, I mean 'stuck' in the sense that with incoming messages the scrollbar goes to the top which makes the most recent messages in the log difficult to read. I can move it down too, but it goes right back up immediately.

sbourdeauducq, 2015-10-21:
It works correctly here with both pyqt4 and pyqt5.

sbourdeauducq, 2015-10-21:
Is the scrollbar's cursor actually moving to the top when a new message arrives? Or do you just see a scrolling happening because old messages get discarded when the buffer is full?

kgilmo, 2015-10-21:
The cursor seems to be moving to the top.

kgilmo, 2015-10-27:
Joe sees this too in today's build of ARTIQ. Do you have a scenario that produces 1000's of lines of log content when you try to reproduce what we observe?

kgilmo, 2015-10-27:
[image]

sbourdeauducq, 2015-10-28:
Yes, I wrote an experiment that printed a new message every couple milliseconds.

sbourdeauducq, 2015-10-28:
I can actually reproduce it on my other computer...

sbourdeauducq, 2015-10-28:
Looks like a Qt bug. It happens even when ARTIQ doesn't touch the scrolling...

jboulder, 2015-10-28:
How about trimming old messages out of the log beyond whatever depth the QT
widget can handle?

sbourdeauducq, 2015-10-28:
The trimming is actually the source of the problem. When you remove row 0 of the table (which is what the current code already does to limit the log to 1000 entries), Qt intermittently scrolls to the top.

sbourdeauducq, 2015-10-30:
This is totally a Qt issue. The program below also has the problem. Seems to be triggered by switching the window focus.

```python
import sys
from PyQt5 import QtCore, QtWidgets

class _Scroller(QtCore.QAbstractTableModel):
    def __init__(self, parent):
        QtCore.QAbstractTableModel.__init__(self, parent)

        self.headers = ["Test"]
        self.depth = 20
        self.entries = []
        self.counter = 0

        timer = QtCore.QTimer(self)
        timer.timeout.connect(self.timer_tick)
        timer.start(100)

    def headerData(self, col, orientation, role):
        if (orientation == QtCore.Qt.Horizontal
                and role == QtCore.Qt.DisplayRole):
            return self.headers[col]
        return None

    def rowCount(self, parent):
        return len(self.entries)

    def columnCount(self, parent):
        return len(self.headers)

    def insertRows(self, position, rows=1, index=QtCore.QModelIndex()):
        self.beginInsertRows(QtCore.QModelIndex(), position, position+rows-1)
        self.endInsertRows()

    def removeRows(self, position, rows=1, index=QtCore.QModelIndex()):
        self.beginRemoveRows(QtCore.QModelIndex(), position, position+rows-1)
        self.endRemoveRows()

    def timer_tick(self):
        to_add = [self.counter]
        self.counter += 1

        nrows = len(self.entries)
        self.entries.extend(to_add)
        self.insertRows(nrows, len(to_add))
        if len(self.entries) > self.depth:
            start = len(self.entries) - self.depth
            self.entries = self.entries[start:]
            self.removeRows(0, start)

    def data(self, index, role):
        if index.isValid():
            if role == QtCore.Qt.DisplayRole:
                return str(self.entries[index.row()])

def main():
    app = QtWidgets.QApplication(sys.argv)

    w = QtWidgets.QMainWindow()
    table = QtWidgets.QTableView()
    w.setCentralWidget(table)
    w.show()

    table.setModel(_Scroller(table))

    sys.exit(app.exec_())

if __name__ == '__main__':
    main()

```

jboulder, 2015-10-30:
Where are you getting your PyQt5 with support for python 3.5? -Joe

commit 0a36b03a (2015-10-31): gui/log: work around Qt scrolling bug. Closes #151
closed 2015-10-31 by sbourdeauducq