m-labs/artiq #152: set_dataset(key, value, broadcast=True) problem for persistent plots

jboulder, 2015-10-20:
A common plotting usage scenario for ARTIQ will be the following. One histogram will be used by multiple experiments to display the status of ion fluorescence. We don't necessarily want to record all the histogram data for posterity but do want GUI plot configurations to persist between GUI launches.

Example:
```python
from artiq import *
import logging
import numpy as np
import time

class Scratch(EnvExperiment):
    """scratch exp"""

    def build(self):
        pass

    def run(self):
        while True:
            try:
                self.hist_nbins = 100
                self.hist_nmax = 200
                counts = []
                for i in range(10):
                    new_counts = np.random.poisson(100, size=10)
                    counts.append(new_counts)
                    hist, hist_bins = np.histogram(counts, self.hist_nbins,
                        range=(0, self.hist_nmax))
                    hist = hist.tolist()
                    hist_bins = hist_bins.tolist()
                    self.set_dataset("common_hist1_bins", hist_bins,
                                     broadcast=True, persist=True, save=False)
                    self.set_dataset("common_hist1_counts", hist,
                                     broadcast=True, persist=True, save=False)
                    time.sleep(0.1)

            except TerminationRequested:
                # record in Parameter DB the latest frequency which ought to be self.f
                self.set_artiq_datasets_if_changed()
                logging.info("Terminated gracefully")
```
What works: plotting between restarts of ARTIQ Master and GUI. Good.

What doesn't work: I had to mark persist=True even though I don't care to retain the data in common_hist1_bins and common_hist1_counts dataset. 

Expected behavior: If broadcast=True and persist=False, data is broadcast to Master and made available in dataset database. Upon closure of the GUI the common_hist1_bins and common_hist1_counts dataset are set to None in the dataset. GUI plotting should check for None and generate an empty plot. This paves the way for persistent plots dependent on immutable dataset entries with mutable data.

sbourdeauducq, 2015-10-21:
"What doesn't work: I had to mark persist=True even though I don't care to retain the data in common_hist1_bins and common_hist1_counts dataset."

This is not a description of something that doesn't work. What happened when you used persist=False?

sbourdeauducq, 2015-10-21:
Also, there is no such thing as a "persistent plot". A "persistent dataset" is one that gets written to dataset_db.pyon so it is saved between master restarts - there is no other difference.

closed 2016-01-15 by sbourdeauducq