m-labs/artiq #154: dataset_db.py can't be reset

jboulder, 2015-10-20:
I've deleted the following files from my ARTIQ home folder.
* dataset_db.pyon
* artiq_gui.pyon
* results folder
I start artiq_master and artiq_gui. Looking in GUI dataset pane it's not empty! Looking in dataset_db.pyon on disk, it's not empty! Where did the entries come from?

Expected behavior is that all datasets are deleted upon removing dataset_db.pyon.

```python
$ cat dataset_db.pyon

{
    "geonium_cyc_f": 7000000.0,
    "hk_default_ion_present": true,
    "master_cpu_load": 0.05,
    "wall_f": 100000.0,
    "flopping_freq": 1499.9994777763654,
    "common_hist1_counts": nparray((100, ), "int64", "AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAEAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAEAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAEAAAAAAAAAAwAAAAAAAAABAAAAAAAAAAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="),
    "geonium_com_f": 1000000.0,
    "total": npscalar("int64", "ZicAAAAAAAA="),
    "counts": [0.03478169174721513, 0.26395262749628967, 0.30233966709371685, 0.32820304008235734, 0.8491522117152708, 0.7481885099930505, 0.4613533368054358, 0.25352027270837074, 0.9486720651208157, 0.6282172129198613],
    "common_hist1_bins": nparray((101, ), "float64", "AAAAAAAAAAAAAAAAAAAAQAAAAAAAABBAAAAAAAAAGEAAAAAAAAAgQAAAAAAAACRAAAAAAAAAKEAAAAAAAAAsQAAAAAAAADBAAAAAAAAAMkAAAAAAAAA0QAAAAAAAADZAAAAAAAAAOEAAAAAAAAA6QAAAAAAAADxAAAAAAAAAPkAAAAAAAABAQAAAAAAAAEFAAAAAAAAAQkAAAAAAAABDQAAAAAAAAERAAAAAAAAARUAAAAAAAABGQAAAAAAAAEdAAAAAAAAASEAAAAAAAABJQAAAAAAAAEpAAAAAAAAAS0AAAAAAAABMQAAAAAAAAE1AAAAAAAAATkAAAAAAAABPQAAAAAAAAFBAAAAAAACAUEAAAAAAAABRQAAAAAAAgFFAAAAAAAAAUkAAAAAAAIBSQAAAAAAAAFNAAAAAAACAU0AAAAAAAABUQAAAAAAAgFRAAAAAAAAAVUAAAAAAAIBVQAAAAAAAAFZAAAAAAACAVkAAAAAAAABXQAAAAAAAgFdAAAAAAAAAWEAAAAAAAIBYQAAAAAAAAFlAAAAAAACAWUAAAAAAAABaQAAAAAAAgFpAAAAAAAAAW0AAAAAAAIBbQAAAAAAAAFxAAAAAAACAXEAAAAAAAABdQAAAAAAAgF1AAAAAAAAAXkAAAAAAAIBeQAAAAAAAAF9AAAAAAACAX0AAAAAAAABgQAAAAAAAQGBAAAAAAACAYEAAAAAAAMBgQAAAAAAAAGFAAAAAAABAYUAAAAAAAIBhQAAAAAAAwGFAAAAAAAAAYkAAAAAAAEBiQAAAAAAAgGJAAAAAAADAYkAAAAAAAABjQAAAAAAAQGNAAAAAAACAY0AAAAAAAMBjQAAAAAAAAGRAAAAAAABAZEAAAAAAAIBkQAAAAAAAwGRAAAAAAAAAZUAAAAAAAEBlQAAAAAAAgGVAAAAAAADAZUAAAAAAAABmQAAAAAAAQGZAAAAAAACAZkAAAAAAAMBmQAAAAAAAAGdAAAAAAABAZ0AAAAAAAIBnQAAAAAAAwGdAAAAAAAAAaEAAAAAAAEBoQAAAAAAAgGhAAAAAAADAaEAAAAAAAABpQA=="),
    "wall_m": 1.0
}

```

sbourdeauducq, 2015-10-21:
Probably from the memory of the master process that was still running. dataset_db is owned by the master while it is running and should not be modified manually during that time.

closed 2015-10-21 by sbourdeauducq

jboulder, 2015-10-22:
Yes. This was the problem.