m-labs/artiq #155: dataset_db.py & ARTIQ in a state where dataset can't be updated

jboulder, 2015-10-20:
For the following states of the dataset_db.py and artiq_gui.py the following program does not set xxx in the dataset_db. There are no errors. 

```python
from artiq import *
import logging
import numpy as np
import time

class Scratch2(EnvExperiment):
    """scratch exp2"""

    def build(self):
        pass

    def run(self):
        # confirm: resetting datasets can change type
        xxx = 10 + np.random.normal(0, 1)
        self.set_dataset("xxx", xxx,
                         broadcast=True, persist=True, save=False)
        logging.debug("scratch2 get_dataset(xxx) = {}".format(self.get_dataset('xxx')))
```

```python
{
    "ExplorerDock": {
        "Novatech Wall": {
            "expanded": [],
            "argument_values": {
                "wall_order_m": "2",
                "on": false,
                "reset_dds_phases": false,
                "scan": {
                    "max": 101000.0,
                    "ty": "LinearScan",
                    "npoints": 10,
                    "min": 100000.0
                },
                "reset_novatech": false,
                "gain": 0.0,
                "dfdt": 1000.0,
                "dfdt_max": 1000.0
            }
        },
        "scratch exp": {
            "expanded": [],
            "argument_values": {}
        },
        "default": {
            "expanded": [],
            "argument_values": {
                "detect_t": 0.001,
                "hist_nmax": 200,
                "hist_nbins": 60,
                "simulation_mode": false,
                "no_cool_t": 0.03,
                "nreps": 100,
                "cool_t": 0.003
            }
        },
        "default experiment using std_include": {
            "expanded": [],
            "argument_values": {
                "detect_t": 0.001,
                "hist_nmax": 200.0,
                "simulate ions": true,
                "hist_nbins": 60.0,
                "no_cool_t": 0.01,
                "nreps": 100.0,
                "cool_t": 0.003,
                "repump_t": 0.001
[... 142 lines cut ...]
    "MainWindow": b'\x01\xd9\xd0\xcb\x00\x01\x00\x00\x00\x00\x00\x08\x00\x00\x00^\x00\x00\x07\x87\x00\x00\x04W\x00\x00\x00\x08\x00\x00\x00^\x00\x00\x07\x87\x00\x00\x04W\x00\x00\x00\x00\x00\x00'
}

```

```python
{
    "xx": 9.20917720834072,
    "counts": [0.03478169174721513, 0.26395262749628967, 0.30233966709371685, 0.32820304008235734, 0.8491522117152708, 0.7481885099930505, 0.4613533368054358, 0.25352027270837074, 0.9486720651208157, 0.6282172129198613],
    "master_cpu_load": 0.05,
    "total": npscalar("int64", "ZicAAAAAAAA="),
    "geonium_cyc_f": 7000000.0,
    "wall_f": 100000.0,
    "wall_m": 1.0,
    "common_hist1_counts": [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 3, 0, 2, 0, 2, 1, 1, 1, 5, 3, 3, 5, 3, 2, 3, 0, 2, 1, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    "flopping_freq": 1499.9994777763654,
    "x": 3,
    "hk_default_ion_present": true,
    "y": nparray((10, ), "float64", "AAAAAACAYEDNzMzMzIxWQGdmZmZmJl1AzszMzMzMXkDNzMzMzIxWQAEAAAAA4FxAAQAAAABgV0ABAAAAAOBcQDQzMzMzs11AAQAAAADAWEA="),
    "common_hist1_bins": [0.0, 2.0, 4.0, 6.0, 8.0, 10.0, 12.0, 14.0, 16.0, 18.0, 20.0, 22.0, 24.0, 26.0, 28.0, 30.0, 32.0, 34.0, 36.0, 38.0, 40.0, 42.0, 44.0, 46.0, 48.0, 50.0, 52.0, 54.0, 56.0, 58.0, 60.0, 62.0, 64.0, 66.0, 68.0, 70.0, 72.0, 74.0, 76.0, 78.0, 80.0, 82.0, 84.0, 86.0, 88.0, 90.0, 92.0, 94.0, 96.0, 98.0, 100.0, 102.0, 104.0, 106.0, 108.0, 110.0, 112.0, 114.0, 116.0, 118.0, 120.0, 122.0, 124.0, 126.0, 128.0, 130.0, 132.0, 134.0, 136.0, 138.0, 140.0, 142.0, 144.0, 146.0, 148.0, 150.0, 152.0, 154.0, 156.0, 158.0, 160.0, 162.0, 164.0, 166.0, 168.0, 170.0, 172.0, 174.0, 176.0, 178.0, 180.0, 182.0, 184.0, 186.0, 188.0, 190.0, 192.0, 194.0, 196.0, 198.0, 200.0],
    "geonium_com_f": 1000000.0,
    "detect_bright_mean": 60.47
}

```

sbourdeauducq, 2015-10-22:
I cannot reproduce the problem; it works fine here (except that the pyqtgraph fails to restore the dock state from the DockArea entry in the artiq_gui.pyon you provided).

* Are you sure you are not just experiencing some network issue? Can you try again, and does the problem persist?
* Is the value not updated in the GUI, or in the master? Can you try reading it back with another experiment that prints it? Note that while the master is running, there is a delay of up to 30 seconds for dataset_db.pyon to be updated.

jboulder, 2015-10-22:
> Are you sure you are not just experiencing some network issue? Can you try again, and does the problem persist?

Yes. Confirmed that the problem persists even after restarting all ARTIQ processes.

>Note that while the master is running, there is a delay of up to 30 seconds for dataset_db.pyon to be updated.

Noted. It doesn't appear in the GUI after 1 minute. 

OK. So with some playing around... It looks like the dataset_db.pyon is getting updated correctly but the GUI Dataset pane is not being updated. See below for some examples of interaction via artiq_client. I've run the code snippet above from the GUI which generates xxx in the on-disk dataset_db.pyon but not in the GUI. 

```python

(py35)jwbritto@68810artiq2:~/gitlab/nistpenning/artiq$ artiq_client --server localhost set-dataset -p common_hist1_counts 10
(py35)jwbritto@68810artiq2:~/gitlab/nistpenning/artiq$ cat dataset_db.pyon 
{
    "common_hist1_counts": [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    "detect_bright_mean": 255.21,
    "common_hist1_bins": [0.0, 3.3333333333333335, 6.666666666666667, 10.0, 13.333333333333334, 16.666666666666668, 20.0, 23.333333333333336, 26.666666666666668, 30.0, 33.333333333333336, 36.66666666666667, 40.0, 43.333333333333336, 46.66666666666667, 50.0, 53.333333333333336, 56.66666666666667, 60.0, 63.333333333333336, 66.66666666666667, 70.0, 73.33333333333334, 76.66666666666667, 80.0, 83.33333333333334, 86.66666666666667, 90.0, 93.33333333333334, 96.66666666666667, 100.0, 103.33333333333334, 106.66666666666667, 110.0, 113.33333333333334, 116.66666666666667, 120.0, 123.33333333333334, 126.66666666666667, 130.0, 133.33333333333334, 136.66666666666669, 140.0, 143.33333333333334, 146.66666666666669, 150.0, 153.33333333333334, 156.66666666666669, 160.0, 163.33333333333334, 166.66666666666669, 170.0, 173.33333333333334, 176.66666666666669, 180.0, 183.33333333333334, 186.66666666666669, 190.0, 193.33333333333334, 196.66666666666669, 200.0]
}
(py35)jwbritto@68810artiq2:~/gitlab/nistpenning/artiq$ artiq_client --server localhost --help
usage: artiq_client [-h] [-s SERVER] [--port PORT]
                    {submit,delete,set-dataset,del-dataset,show,scan-devices,scan-repository}
                    ...

ARTIQ CLI client

positional arguments:
  {submit,delete,set-dataset,del-dataset,show,scan-devices,scan-repository}
    submit              submit an experiment
    delete              delete an experiment from the schedule
    set-dataset         add or modify a dataset
    del-dataset         delete a dataset
    show                show schedule, log, devices or datasets
    scan-devices        trigger a device database (re)scan
    scan-repository     trigger a repository (re)scan

optional arguments:
  -h, --help            show this help message and exit
  -s SERVER, --server SERVER
                        hostname or IP of the master to connect to
  --port PORT           TCP port to use to connect to the master
(py35)jwbritto@68810artiq2:~/gitlab/nistpenning/artiq$ artiq_client --server localhost set-dataset -p newvar 10
(py35)jwbritto@68810artiq2:~/gitlab/nistpenning/artiq$ cat dataset_db.pyon 
{
    "common_hist1_counts": [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    "common_hist1_bins": [0.0, 3.3333333333333335, 6.666666666666667, 10.0, 13.333333333333334, 16.666666666666668, 20.0, 23.333333333333336, 26.666666666666668, 30.0, 33.333333333333336, 36.66666666666667, 40.0, 43.333333333333336, 46.66666666666667, 50.0, 53.333333333333336, 56.66666666666667, 60.0, 63.333333333333336, 66.66666666666667, 70.0, 73.33333333333334, 76.66666666666667, 80.0, 83.33333333333334, 86.66666666666667, 90.0, 93.33333333333334, 96.66666666666667, 100.0, 103.33333333333334, 106.66666666666667, 110.0, 113.33333333333334, 116.66666666666667, 120.0, 123.33333333333334, 126.66666666666667, 130.0, 133.33333333333334, 136.66666666666669, 140.0, 143.33333333333334, 146.66666666666669, 150.0, 153.33333333333334, 156.66666666666669, 160.0, 163.33333333333334, 166.66666666666669, 170.0, 173.33333333333334, 176.66666666666669, 180.0, 183.33333333333334, 186.66666666666669, 190.0, 193.33333333333334, 196.66666666666669, 200.0],
    "detect_bright_mean": 265.51,
    "newvar": 10
}
(py35)jwbritto@68810artiq2:~/gitlab/nistpenning/artiq$ artiq_client --server localhost scan-repository
(py35)jwbritto@68810artiq2:~/gitlab/nistpenning/artiq$ cat dataset_db.pyon 
{
    "xxx": 10.42538582963125,
    "common_hist1_counts": [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    "common_hist1_bins": [0.0, 3.3333333333333335, 6.666666666666667, 10.0, 13.333333333333334, 16.666666666666668, 20.0, 23.333333333333336, 26.666666666666668, 30.0, 33.333333333333336, 36.66666666666667, 40.0, 43.333333333333336, 46.66666666666667, 50.0, 53.333333333333336, 56.66666666666667, 60.0, 63.333333333333336, 66.66666666666667, 70.0, 73.33333333333334, 76.66666666666667, 80.0, 83.33333333333334, 86.66666666666667, 90.0, 93.33333333333334, 96.66666666666667, 100.0, 103.33333333333334, 106.66666666666667, 110.0, 113.33333333333334, 116.66666666666667, 120.0, 123.33333333333334, 126.66666666666667, 130.0, 133.33333333333334, 136.66666666666669, 140.0, 143.33333333333334, 146.66666666666669, 150.0, 153.33333333333334, 156.66666666666669, 160.0, 163.33333333333334, 166.66666666666669, 170.0, 173.33333333333334, 176.66666666666669, 180.0, 183.33333333333334, 186.66666666666669, 190.0, 193.33333333333334, 196.66666666666669, 200.0],
    "detect_bright_mean": 277.09,
    "newvar": 10
}

```

sbourdeauducq, 2015-10-23:
Regarding delayed updates: I was talking about the PYON file on the disk, not the GUI. The GUI is always updated immediately.

Is ``artiq_client show-datasets`` also affected by this problem?

closed 2016-01-15 by sbourdeauducq