m-labs/artiq #156: auto generation of empty *.pyon

jboulder, 2015-10-22:
Find that ARTIQ does not automatically generate an empty dataset_db.pyon if dataset_db.pyon is missing. Auto generation is better default behavior vs throwing exception. Same goes for artiq_gui.pyon and device_db.pyon.

artiq_gui.pyon might better be called gui_db.pyon to match new naming scheme.

sbourdeauducq, 2015-10-23:
Ok for dataset_db.

I do not see the point of having an empty device_db.pyon. How would you use ARTIQ with no devices?
Note that this file is read-only for the master, i.e. the way to add new devices is to modify this file and trigger a rescan.

The GUI already creates artiq_gui.pyon when it is not present.

closed 2015-10-24 by sbourdeauducq

jboulder, 2015-10-25:
For both debugging and to eliminate accumulated dead datasets, it's
convenient to simply delete dataset_db.pyon and fire up ARTIQ again.

Thank you for making this change.

commit 3044a053 (2015-10-24): databases: assume empty dataset_db if file not present. Closes #156