m-labs/artiq #157: conda broken

jboulder, 2015-10-22:
(py35)rabi@688penningA:~$ conda update artiq
Fetching package metadata: ....Error: HTTPError: 500 Server Error: INTERNAL SERVER ERROR: http://conda.anaconda.org/m-labs/channel/dev/noarch/

(py35)rabi@688penningA:~$ conda info
Current conda install:

             platform : linux-64
        conda version : 3.17.0
  conda-build version : 1.14.1
       python version : 3.4.3.final.0
     requests version : 2.7.0
     root environment : /home/rabi/anaconda3  (writable)
  default environment : /home/rabi/anaconda3/envs/py35
     envs directories : /home/rabi/anaconda3/envs
        package cache : /home/rabi/anaconda3/pkgs
         channel URLs : http://conda.anaconda.org/m-labs/channel/main/linux-64/
                        http://conda.anaconda.org/m-labs/channel/main/noarch/
                        http://conda.anaconda.org/m-labs/channel/dev/linux-64/
                        http://conda.anaconda.org/m-labs/channel/dev/noarch/
                        https://repo.continuum.io/pkgs/free/linux-64/
                        https://repo.continuum.io/pkgs/free/noarch/
                        https://repo.continuum.io/pkgs/pro/linux-64/
                        https://repo.continuum.io/pkgs/pro/noarch/
          config file : /home/rabi/.condarc
    is foreign system : False

(py35)rabi@688penningA:~$ conda update artiq
Fetching package metadata: ....Error: HTTPError: 500 Server Error: INTERNAL SERVER ERROR: http://conda.anaconda.org/m-labs/channel/dev/noarch/

whitequark, 2015-10-22:
That's wonderful. @sbourdeauducq, are you paying for anaconda? I really hope so, because then there's at least a tiny chance anyone at all would care about this.

whitequark, 2015-10-22:
I filed this as https://github.com/ContinuumIO/anaconda-issues/issues/490 but they have three hundred open issues, some dating back to 2013.

whitequark, 2015-10-22:
Nevermind, the entirety of https://conda.anaconda.org/ is down. This is an upstream issue.

jboulder, 2015-10-22:
It work. Thanks for the quick save Peter.

jboulder, 2015-10-22:
But not out of the woods all together. By "it works" I mean the conda install works. But now it can't find the bitfiles. 

```python
(py35)rabi@688penningA:~/gitlab/nistpenning/artiq$ conda update artiq
Fetching package metadata: ........
Solving package specifications: .
Package plan for installation in environment /home/rabi/anaconda3/envs/py35:

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    openssl-1.0.2d             |                0         3.2 MB
    freetype-2.5.5             |                0         2.2 MB
    alabaster-0.7.6            |           py35_0          14 KB
    python-3.5.0               |                1        16.6 MB
    h5py-2.5.0                 |      np110py35_4         2.7 MB
    artiq-0.0                  |py_1650+git30842a7         215 KB
    ------------------------------------------------------------
                                           Total:        24.9 MB

The following packages will be UPDATED:

    alabaster: 0.7.3-py35_0      --> 0.7.6-py35_0          
    artiq:     0.0-1639          --> 0.0-py_1650+git30842a7
    freetype:  2.5.2-2           --> 2.5.5-0               
    h5py:      2.5.0-np110py35_3 --> 2.5.0-np110py35_4     
    openssl:   1.0.1k-1          --> 1.0.2d-0              
    python:    3.5.0-0           --> 3.5.0-1               

Proceed ([y]/n)? y

Fetching packages ...
openssl-1.0.2d 100% |###########################################################| Time: 0:00:00   4.06 MB/s
freetype-2.5.5 100% |###########################################################| Time: 0:00:00   4.04 MB/s
alabaster-0.7. 100% |###########################################################| Time: 0:00:00 324.51 kB/s
python-3.5.0-1 100% |###########################################################| Time: 0:00:00  21.60 MB/s
h5py-2.5.0-np1 100% |###########################################################| Time: 0:00:00   5.66 MB/s
artiq-0.0-py_1 100% |###########################################################| Time: 0:00:00 840.90 kB/s
Extracting packages ...
[      COMPLETE      ]|##############################################################################| 100%
Unlinking packages ...
[      COMPLETE      ]|##############################################################################| 100%
Linking packages ...
[      COMPLETE      ]|##############################################################################| 100%
(py35)rabi@688penningA:~/gitlab/nistpenning/artiq$ ls
artiq_gui_missing_tabs.pyon    dataset_db.pyon~  terminator_688penninga.config
artiq_gui.pyon                 device_db.pyon    terminator_localhost.config
artiq_start_688penninga.tmux~  repository        test.dat
dataset_db.pyon                results
(py35)rabi@688penningA:~/gitlab/nistpenning/artiq$ artiq_flash.sh -t kc705 -m nist_qc1 kc705 flash proxy (bscan_spi_kc705.bit) not found.
Please put it in ~/.migen or /usr/local/share/migen or /usr/share/migen
To get the flash proxy, follow the Install the required flash proxy bitstreams:
bullet point from http://m-labs.hk/artiq/manual/installing.html#preparing-the-core-device-fpga-board

```

whitequark, 2015-10-22:
That's expected after https://github.com/m-labs/artiq/issues/146. I will update the manual shortly.

whitequark, 2015-10-22:
So far I suggest doing `conda install artiq-pipistrello-nist_qc1 artiq-kc705-nist_qc1 artiq-kc705-nist_qc2` to get the old behavior.

jboulder, 2015-10-22:
No go.

```
(py35)rabi@688penningA:~/gitlab/nistpenning/artiq$ conda install artiq-pipistrello-nist_qc1 artiq-kc705-nist_qc1 artiq-kc705-nist_qc2
Fetching package metadata: ........
Solving package specifications: .An unexpected error has occurred, please consider sending the
following traceback to the conda GitHub issue tracker at:

    https://github.com/conda/conda/issues

Include the output of the command 'conda info' in your report.

Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/py35/bin/conda", line 5, in <module>
    sys.exit(main())
  File "/home/rabi/anaconda3/lib/python3.4/site-packages/conda/cli/main.py", line 194, in main
    args_func(args, p)
  File "/home/rabi/anaconda3/lib/python3.4/site-packages/conda/cli/main.py", line 201, in args_func
    args.func(args, p)
  File "/home/rabi/anaconda3/lib/python3.4/site-packages/conda/cli/main_install.py", line 45, in execute
    install.install(args, parser, 'install')
  File "/home/rabi/anaconda3/lib/python3.4/site-packages/conda/cli/install.py", line 334, in install
    minimal_hint=args.alt_hint)
  File "/home/rabi/anaconda3/lib/python3.4/site-packages/conda/plan.py", line 403, in install_actions
    config.track_features, minimal_hint=minimal_hint):
  File "/home/rabi/anaconda3/lib/python3.4/site-packages/conda/resolve.py", line 747, in solve
    minimal_hint=minimal_hint)
  File "/home/rabi/anaconda3/lib/python3.4/site-packages/conda/resolve.py", line 492, in solve2
    clauses = set(self.gen_clauses(v, dists, specs, features))
  File "/home/rabi/anaconda3/lib/python3.4/site-packages/conda/resolve.py", line 350, in gen_clauses
    assert len(clause) > 1, '%s %r' % (fn1, ms)
AssertionError: artiq-0.0-py_1650+git30842a7.tar.bz2 MatchSpec('aiohttp')
```

whitequark, 2015-10-22:
Hm, seems to be a bug in conda's dependency resolver. I can reproduce it locally, though no idea why it happens or what can we do...

jboulder, 2015-10-23:
```bash
$ conda info artiq-kc705-nist_qc1

artiq-kc705-nist_qc1 0.0 py_1650+git30842a7
-------------------------------------------
file name   : artiq-kc705-nist_qc1-0.0-py_1650+git30842a7.tar.bz2
name        : artiq-kc705-nist_qc1
version     : 0.0
build number: 1650
build string: py_1650+git30842a7
channel     : http://conda.anaconda.org/m-labs/channel/dev/noarch/
size        : 601 KB
arch        : None
binstar     : {'package_id': '5628e43ad6f51f0b13332e1b', 'channel': 'dev', 'owner_id': '55cc3c85d2626f681e0ee47f'}
license     : 3-clause BSD
machine     : None
md5         : f34f36da5e7af14895eeac85bc4a3bce
operatingsystem: None
platform    : None
subdir      : noarch
target-triplet: None-any-None
installed environments:
dependencies:
    artiq 0.0 py_1650+git30842a7
```
Conda has a funny way of specifying specific builds. It's not officially documented as far as I could find. The following permits me to choose a specific ARTIQ build.

```
$ conda install artiq=0.0=py_1650+git30842a7
```
And to install a package with a specific dependency override...
```
$ conda install artiq=0.0=py_1650+git30842a7 artiq-kc705-nist_qc1
Fetching package metadata: ........
Solving package specifications: .
Package plan for installation in environment /home/rabi/anaconda3/envs/py35test:

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    artiq-kc705-nist_qc1-0.0   |py_1650+git30842a7         601 KB

The following NEW packages will be INSTALLED:

    artiq-kc705-nist_qc1: 0.0-py_1650+git30842a7

Proceed ([y]/n)? y

Fetching packages ...
artiq-kc705-ni 100% |########################################################| Time: 0:00:01 607.51 kB/s
Extracting packages ...
[      COMPLETE      ]|###########################################################################| 100%
Linking packages ...
[      COMPLETE      ]|###########################################################################| 100%
```

whitequark, 2015-10-26:
The bug is https://github.com/conda/conda/pull/1740. I've tried working around it in f384142 but that didn't work as it just sends conda into infinite recursion. I am looking at other solutions now.

whitequark, 2015-10-27:
My patch also doesn't work because if the artiq package is not already installed, resolution fails. There is also the problem that there is no public conda recipe to build conda, and my quick attempt at it results in bin/conda being deleted for some reason I have not quite figured out.

whitequark, 2015-10-29:
Conda was fixed.

closed 2015-10-29 by whitequark