m-labs/artiq #161: master can ping device; but device responses inaccessible to user program

kgilmo, 2015-10-23:
I can control and receive return values from calls to the novatech_wall device using artiq_rpctool. 
I can control the novatech_wall device using artiq_run. 
However, None is returned from rpc calls made by  artiq_run and  artiq_gui.

```python
from artiq import *
import logging

class NovatechTest(EnvExperiment):
    """novatech test"""
    def build(self):
        self.novatech_wall = self.get_device("novatech_wall")

    def run(self):

        s1 = self.novatech_wall.get_status()
        s2 = self.novatech_wall.ping()
        logging.critical(s1)
        logging.critical(s2)
        logging.critical("this is novatech_test.py")
```

Below is transcript of artiq_master log. It shows that the master can successfully ping the device and receive the correct return value. But the program when run by artiq_run always obtains None as the return value. 

```bash
(py35)rabi@688penningA:~/gitlab/nistpenning/artiq$ DEBUG:controller(novatech_wall):artiq.protocols.pc_rpc:calling get_status()
WARNING:worker(9990):artiq.protocols.pc_rpc:connection failed while attempting RPC to 68810artiq1.bw.nist.gov:50001[novatech409b], re-establishing connection in the background
CRITICAL:worker(9990):root:None
CRITICAL:worker(9990):root:None
CRITICAL:worker(9990):root:this is novatech_test.py
DEBUG:controller(novatech_wall):artiq.devices.novatech409b.driver:got device status: ['003D0900 0000 0000 0000 00000000 00000000 000301', '003D0900 0000 0000 0000 00000000 00000000 000301', '003D0900 0000 0000 0000 00000000 00000000 000301', '003D0900 0000 0000 0000 00000000 00000000 000301', '80 BC0000 0000 0002 21']
DEBUG:ctlmgr(68810artiq1):artiq.frontend.artiq_ctlmgr:pinging controller trans1
DEBUG:ctlmgr(68810artiq1):artiq.frontend.artiq_ctlmgr:pinging controller novatech_wall
DEBUG:controller(novatech_wall):artiq.protocols.pc_rpc:calling ping()
DEBUG:controller(novatech_wall):artiq.devices.novatech409b.driver:got device status: ['003D0900 0000 0000 0000 00000000 00000000 000301', '003D0900 0000 0000 0000 00000000 00000000 000301', '003D0900 0000 0000 0000 00000000 00000000 000301', '003D0900 0000 0000 0000 00000000 00000000 000301', '80 BC0000 0000 0002 21']
DEBUG:controller(novatech_wall):artiq.devices.novatech409b.driver:ping successful
```

device_db.pyon on master 
```python
    "novatech_wall": {
        "type": "controller",
        "best_effort": True,
        "host": "68810artiq1.bw.nist.gov",
        #"comment": "68810artiq1",
        "port": 50001,
        "target_name": "novatech409b",
        "command": "novatech409b_controller  -v -v -p {port} --bind {bind} -d \"hwgrep://USB VID:PID=067b:2303\" "
    },
```

kgilmo, 2015-10-23:
Submitted prematurely... More info: 

This is same problem as #149. But the problem is more clearly defined. If I run the same program but with "best_effort": False in the device_db.pyon then it works fine. It appears there's a bug with best_effort option.

```bash
(py35)rabi@688penningA:~/gitlab/nistpenning/artiq$ artiq_run -v -v repository/novatech_test.py 
CRITICAL:root:['003D0900 0000 0000 0000 00000000 00000000 000301', '003D0900 0000 0000 0000 00000000 00000000 000301', '003D0900 0000 0000 0000 00000000 00000000 000301', '003D0900 0000 0000 0000 00000000 00000000 000301', '80 BC0000 0000 0102 21']
CRITICAL:root:True
CRITICAL:root:this is novatech_test.py

```

sbourdeauducq, 2015-10-24:
Works fine here.

The relevant message is this:
``WARNING:worker(9990):artiq.protocols.pc_rpc:connection failed while attempting RPC to 68810artiq1.bw.nist.gov:50001[novatech409b], re-establishing connection in the background``
Do you confirm that this is printed all the time when you use best_effort=True, and it works with the exact same code when best_effort=False?

kgilmo, 2015-10-30:
Yes, that's right. If I change best_effort to 'True' that is always printed when running the above. Changing only the value of best_effort to 'False' and it works just fine.

commit 5bf25781 (2016-01-15): protocols/pc_rpc: use timeout only for connecting. Closes #161
closed 2016-01-15 by sbourdeauducq