m-labs/artiq #163: GUI DDS & TTL panes can't shrink enough

jboulder, 2015-10-25:
* DDS ... more horizontal shrink desired
* TTL ... more horizontal and vertical shrink desired
* suggestion: font size dynamic, let physicists eyes suffer if pane is so small text becomes unreadable
[image]
[image]

dhslichter, 2015-10-26:
Once we have the new hardware in place this will be even more of an issue because of many more channels.  Is there a simple way in the GUI to configure which DDS or TTL channels are show in the panel, and/or a place where one can set the size of the displayed array (e.g. could make DDS 2 cols, 4 rows instead of 2 rows, 4 cols)?  This is an alternative solution to dynamic font sizes.

sbourdeauducq, 2015-10-27:
What about scrollbars?

jboulder, 2015-10-27:
I prefer a long vertical single-column list.

I don't think scroll bars are a good idea for this. The TTL toggles are so
often used it's helpful to have them all visible at once.

The current GUI's right button pop up menu is nice but slow in use. Easier
would be click to cycle through the three available states. On off dynamic.
That's how the current HFgui works.

dhslichter, 2015-10-27:
I agree that it's nice to avoid scroll bars.  It would be ideal if there were a simply edited configuration that allowed the user to choose the number of rows and columns, and which channels are displayed (if a subset is desired).  Depending on whether people have a vertical or horizontal monitor, for example, they might have a preference.

jboulder, 2015-10-27:
With the perspective that GUI modifications are somewhat painstaking
to get right, I propose we get something that's simple and defer the
problem of bling and user-parameterized layout. Something that looks
like the existing Datasets pane where...

TTL Pane's columns:
* channel no
* status
* comment
** click on row to toggle status: on/off/auto
** right click to bring up existing status toggle menu

DDS Pane's columns:
* channel no
* frequency
* comment

Common to both:
* scrollbars appear if user shrinks window to be too small to display full list

sbourdeauducq, 2015-10-28:
Is cycling through the output states really that good? It causes electrical glitches and I don't see how it is faster than the menu, as you typically need several clicks as well to reach the state you want.

sbourdeauducq, 2015-10-28:
Seems to me that a better option would be to have buttons that select each state on each TTL channel widget. Maybe hide them when the cursor is not on the widget in order to reduce visual clutter.

sbourdeauducq, 2015-10-28:
Maybe this is a good solution for the layout:
http://stackoverflow.com/questions/14619047/qt-layouts-how-to-make-widgets-in-horizontal-layout-move-down-a-row-instead-of
http://doc.qt.io/qt-5/qtwidgets-layouts-flowlayout-example.html

If we use this, probably there is a bit of work to do to handle the comment fields whose size can vary from zero to several lines of text.

dhslichter, 2015-10-28:
I like the proposed flowlayout solution, this would definitely be nicest.  What is the comment field currently being used for?  Naming of channels?  Also, I prefer Sebastien's suggestion of 3 buttons for on/off/auto; I don't like the current cycling through method, and there are times when I'd want to make sure I don't accidentally go through "off" on the way from "on" to "auto", for example.

sbourdeauducq, 2016-01-26:
@jordens Implementing all those changes will take a lot of time so we probably want to release 1.0 before.

jordens, 2016-01-26:
Ack about the feature creep. But the original thing was making these docks smaller and more compact which would be very helpful.

jordens, 2016-03-02:
Let me try to summarize this:

There are different layout preferences. Some like an array of widgets that flows depending on the size of the dock. Others like a single column. But they would probably resize the dock to not be overly large horizontally anyway and thus would be fine with a flowing layout.

The actions on a ttl are as follows and they should be a quickly accessible (preferrably single click, no context menu, no keyboard required): force zero, force one, let the experiment-control it.

But I think people will like to be able to toggle between experiment-controlled and force without moving the mouse. That requires the GUI to show what was forced and use that again when forcing is enabled. I would suspect that the sequence "force 1, experiment control, force 0, ..." is far less common than either "force X, experiment control, force X, ..." or "force 0, force 1, ...". Thus is seems acceptable that the first sequence needs one more click between "experiment control" and "force 0". Then having just two check boxes could be sufficient. One to determine the state when forced and the other to determine whether to force or not.

As another change, the sensitivity area should be maximized so that accurate targeting with the mouse becomes less of an issue. If something looks like a button it should behave like a button if at all possible.

sbourdeauducq, 2016-03-03:
The override value can have potentially three states (0/1/input) for ``TTLInOut`` so maybe 3 check boxes:
  * override on/off
  * drive value 1/0
  * drive on/off

Those checkboxes can be hidden when the cursor is not over the corresponding widget, so as to reduce visual clutter.

Touching the drive value checkbox can set override=on and drive=on.

jordens, 2016-03-04:
Yes. If people really want that direction switch, it would be a third bit.

Wouldn't the layout change and poasibly reflow then on mouseover? That would be ugly.

sbourdeauducq, 2016-03-04:
Of course not, the checkboxes would simply be hidden but use exactly the same amount of space as when they are shown. The idea here is simply to reduce the amount of distracting visual elements on the screen.

sbourdeauducq, 2016-03-05:
Note that on QC2, direction overrides would not work because they won't reprogram the I2C direction switch.

dhslichter, 2016-03-07:
Is the direction of a TTL line something that one really wants to allow the user to change in the moninj system?  I can't think of an instance where this would be necessary, and I can definitely imagine an instance where someone unintentionally changes the direction from output to input, leaving that line in a high impedance state and potentially causing the logic state of whatever was on the receiving end of that TTL output to be undefined (possibly switching to an undesired or unsafe state).  What is the use case for changing direction in moninj?

sbourdeauducq, 2016-03-08:
Then let's not have this feature, it makes the design simpler.

dhslichter, 2016-03-08:
Agreed.

sbourdeauducq, 2016-03-29:
Flow layout done.

commit fc524961 (2016-04-05): dashboard/moninj: fancy level setting and layout improvements. Closes #163
closed 2016-04-05 by sbourdeauducq