m-labs/artiq #168: graceful error handling, more exceptions during exception handling is undesired

jboulder, 2015-10-27:
Errors in the use of the ARTIQ language should result in concise error messages that can guide the user. Here's an example of this going wrong.

I've attempted to get_dataset() for a dataset that doesn't exist. 
```python
2015-10-27 08:16:10,042 ERROR:worker(10107):root:Worker terminating with exception
2015-10-27 08:16:10,043 INFO:worker(10107):print:Traceback (most recent call last):
2015-10-27 08:16:10,044 INFO:worker(10107):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_db.py", line 192, in get
2015-10-27 08:16:10,045 INFO:worker(10107):print:    return self.local[key]
2015-10-27 08:16:10,045 INFO:worker(10107):print:KeyError: 'wall_f'
2015-10-27 08:16:10,046 INFO:worker(10107):print:
2015-10-27 08:16:10,047 INFO:worker(10107):print:During handling of the above exception, another exception occurred:
2015-10-27 08:16:10,047 INFO:worker(10107):print:
2015-10-27 08:16:10,048 INFO:worker(10107):print:Traceback (most recent call last):
2015-10-27 08:16:10,049 INFO:worker(10107):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 198, in main
2015-10-27 08:16:10,049 INFO:worker(10107):print:    exp_inst.run()
2015-10-27 08:16:10,050 INFO:worker(10107):print:  File "/home/rabi/gitlab/nistpenning/artiq/repository/novatech_wall_config.py", line 52, in run
2015-10-27 08:16:10,051 INFO:worker(10107):print:    self.instantaneous_f = p_wall_f = self.get_dataset("wall_f")  # in Hz
2015-10-27 08:16:10,051 INFO:worker(10107):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/language/environment.py", line 239, in get_dataset
2015-10-27 08:16:10,052 INFO:worker(10107):print:    return self.__dataset_mgr.get(key)
2015-10-27 08:16:10,053 INFO:worker(10107):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_db.py", line 194, in get
2015-10-27 08:16:10,053 INFO:worker(10107):print:    return self.ddb.get(key)
2015-10-27 08:16:10,054 INFO:worker(10107):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 45, in parent_action
2015-10-27 08:16:10,055 INFO:worker(10107):print:    raise exception(reply["message"])
2015-10-27 08:16:10,055 INFO:worker(10107):print:KeyError: 'Traceback (most recent call last):\n  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 175, in _handle_worker_requests\n    data = func(**obj)\n  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/databases.py", line 54, in get\n    return self.data.read[key][1]\nKeyError: \'wall_f\'\n'
2015-10-27 08:16:10,083 ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 10107
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 279, in _do
    completed = await run.run()
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 32, in worker_method
    return await m(*args, **kwargs)
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 220, in run
    completed = await self._worker_action({"action": "run"})
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 196, in _worker_action
    completed = await self._handle_worker_requests()
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 153, in _handle_worker_requests
    obj = await self._recv(self.watchdog_time())
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 141, in _recv
    raise WorkerError("Worker ended while attempting to receive data")
artiq.master.worker.WorkerError: Worker ended while attempting to receive data
```
The problem is simple and properly caught by KeyError. The expected result is simple: the worker terminates. OK.

But why does this resolution pass through another exception state? "During handling of the above exception, another exception occurred:" and the ensuing stream of crud. This verbosity makes ARTIQ harder to use.

closed 2015-10-28 by sbourdeauducq

jboulder, 2015-10-28:
The commit does not resolve this issue. I expect that the log output ceases after 

> 2015-10-28 12:20:52,311 INFO:worker(0):print:KeyError: 'common_plot1_ypp'

Running a build including ec328cf...

```python
2015-10-28 12:20:52,305 ERROR:worker(0):root:Worker terminating with exception
2015-10-28 12:20:52,306 INFO:worker(0):print:Traceback (most recent call last):
2015-10-28 12:20:52,308 INFO:worker(0):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_db.py", line 192, in get
2015-10-28 12:20:52,309 INFO:worker(0):print:    return self.local[key]
2015-10-28 12:20:52,311 INFO:worker(0):print:KeyError: 'common_plot1_ypp'
2015-10-28 12:20:52,312 INFO:worker(0):print:
2015-10-28 12:20:52,314 INFO:worker(0):print:During handling of the above exception, another exception occurred:
2015-10-28 12:20:52,315 INFO:worker(0):print:
2015-10-28 12:20:52,317 INFO:worker(0):print:Traceback (most recent call last):
2015-10-28 12:20:52,318 INFO:worker(0):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 192, in main
2015-10-28 12:20:52,320 INFO:worker(0):print:    **expid["arguments"])
2015-10-28 12:20:52,321 INFO:worker(0):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/language/environment.py", line 126, in __init__
2015-10-28 12:20:52,323 INFO:worker(0):print:    self.build()
2015-10-28 12:20:52,324 INFO:worker(0):print:  File "/home/jwbritto/gitlab/nistpenning/artiq/repository/scratch.py", line 13, in build
2015-10-28 12:20:52,326 INFO:worker(0):print:    self.ypp = self.get_dataset("common_plot1_ypp")
2015-10-28 12:20:52,327 INFO:worker(0):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/language/environment.py", line 239, in get_dataset
2015-10-28 12:20:52,328 INFO:worker(0):print:    return self.__dataset_mgr.get(key)
2015-10-28 12:20:52,330 INFO:worker(0):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_db.py", line 194, in get
2015-10-28 12:20:52,331 INFO:worker(0):print:    return self.ddb.get(key)
2015-10-28 12:20:52,333 INFO:worker(0):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 45, in parent_action
2015-10-28 12:20:52,334 INFO:worker(0):print:    raise exception(reply["message"])
2015-10-28 12:20:52,336 INFO:worker(0):print:KeyError: 'Traceback (most recent call last):\n  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 175, in _handle_worker_requests\n    data = func(**obj)\n  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/databases.py", line 54, in get\n    return self.data.read[key][1]\nKeyError: \'common_plot1_ypp\'\n'
2015-10-28 12:20:52,378 ERROR:master:artiq.master.scheduler:got worker exception in prepare stage, deleting RID 0
Traceback (most recent call last):
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 228, in _do
    await run.build()
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 110, in build
    self.priority)
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 32, in worker_method
    return await m(*args, **kwargs)
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 214, in build
    timeout)
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 196, in _worker_action
    completed = await self._handle_worker_requests()
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 153, in _handle_worker_requests
    obj = await self._recv(self.watchdog_time())
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 141, in _recv
    raise WorkerError("Worker ended while attempting to receive data")
artiq.master.worker.WorkerError: Worker ended while attempting to receive data
```

sbourdeauducq, 2015-10-29:
If you were running a version that included that commit, the message would be something like:
```
ERROR:worker(207):root:Worker terminating with exception
INFO:worker(207):print:Traceback (most recent call last):
INFO:worker(207):print:  File "/home/sb/NIST/artiq/artiq/master/worker_impl.py", line 193, in main
INFO:worker(207):print:    exp_inst.run()
INFO:worker(207):print:  File "/home/sb/NIST/artiq/examples/master/repository/arguments_demo.py", line 70, in run
INFO:worker(207):print:    self.get_dataset("foobar")
INFO:worker(207):print:  File "/home/sb/NIST/artiq/artiq/language/environment.py", line 245, in get_dataset
INFO:worker(207):print:    return self.__dataset_mgr.get(key)
INFO:worker(207):print:  File "/home/sb/NIST/artiq/artiq/master/worker_db.py", line 194, in get
INFO:worker(207):print:    return self.ddb.get(key)
INFO:worker(207):print:  File "/home/sb/NIST/artiq/artiq/master/worker_impl.py", line 42, in parent_action
INFO:worker(207):print:    raise exception(reply["message"])
INFO:worker(207):print:KeyError: 'Traceback (most recent call last):\n  File "/home/sb/NIST/artiq/artiq/master/worker.py", line 181, in _handle_worker_requests\n    data = func(*obj["args"], **obj["kwargs"])\n  File "/home/sb/NIST/artiq/artiq/master/databases.py", line 54, in get\n    return self.data.read[key][1]\nKeyError: \'foobar\'\n'
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 207
```

Still not perfect, but it does not contain the double exception you were talking about and cuts some of the tracebacks.

commit ec328cf5 (2015-10-28): master/worker_db: reduce exception backtrace clutter of get_dataset. Closes #168