m-labs/artiq #169: tdc001 to prm1z8 rotation stage unresponsive after initial move command

kgilmo, 2015-10-27:
From terminal, commands to thorlabs tdc001 connected to prm1z8 rotation stage are executed but do not terminate. I can ping the device and get its position initially. Upon calling move_() it moves, but the terminal never registers that it has completed its move. The device becomes unresponsive after this. I do not have this problem with a linear translation stage, so far only the rotation stage.

sbourdeauducq, 2015-10-28:
I remember testing this successfully with a rotation stage in the assembly room, though I'm not sure if it was a prm1z8.

Can you see if this this is not a hardware glitch:
  * does the controller also fail with another prm1z8?
  * does the initial prm1z8 work when using another driver?

kgilmo, 2015-10-28:
Yes, it fails with another prm1zd. It works with the other driver we've be
using.

sbourdeauducq, 2015-10-28:
When do you need this to work?

sbourdeauducq, 2015-10-28:
And are you running the controller standalone, or via the controller manager? How long does it take for the rotation stage to complete the move?

kgilmo, 2015-10-28:
I was running just the thor_labs_tcube_controller. Depending on the length of the move, a few seconds. I could reproduce the behavior with any movement command, though. Even if the device is 'home', telling it to 'go home' - which doesn't cause it to move at all - results in it freezing up like this.

kgilmo, 2015-10-28:
It's not urgent, I'm just working on getting the various devices up and running with artiq. @jboulder is there a timeline for getting the tcube stuff working with artiq?

sbourdeauducq, 2015-10-29:
I guess I'll have a look at this during the next visit then. Or you can add some ``print``s in the driver and see where it blocks...

jboulder, 2015-10-30:
The full transition of the Penning experiment to ARTIQ requires use of the
Thorlabs tcubes to control wave-plate angles using the rotation stages. It
would be good to get this to work next week.

kgilmo, 2015-11-02:
The commit on June 18th

https://github.com/m-labs/artiq/commit/4d077f50c6621ad771818f489a5dd611e320ea6d

seems to be the source of the issue. In reverting to prior to these changes, I'm able to get the rotation stages working just fine. Except for 'move_home', which has the same behavior as before. That is, the rotation stage moves 'home' and then hangs and becomes unresponsive. Otherwise, I seem to be able to move the device. 

Also, the model of this rotation stage should be the same as the one that you tested in the assembly room - according to others here.

kgilmo, 2015-11-09:
Just checking in on the status of this issue. Again, it seems the addition of the 'wait_for_msgs' parameter in https://github.com/m-labs/artiq/commit/4d077f50c6621ad771818f489a5dd611e320ea6d is what is in some way causing the hanging problem.

sbourdeauducq, 2015-11-10:
If you want me to fix this, I need the devices.

dhslichter, 2015-11-10:
@fallen has some TDC001 devices currently, can he reproduce the effect?

kgilmo, 2015-11-10:
I'm having a TDC001 and rotation stage ordered and sent out to you

sbourdeauducq, 2015-11-23:
Received

sbourdeauducq, 2015-11-24:
Unfortunately, this works fine with the device I have received. Let me improve the logging so I can guess what is wrong with yours.

kgilmo, 2015-11-24:
Ah.. Ok. Let me know what you need from me.

sbourdeauducq, 2015-11-25:
Please replace your ``artiq/devices/thorlabs_tcube/driver.py`` with the current one from git, and run ``thorlabs_tcube_controller -P tdc001 -d hwgrep://0403:faf0 -v -v`` in one terminal, and ``artiq_rpctool.py -p 3255 call move_home`` in another.

When you issue the move home command, the controller should print the messages below, and the ``move_home`` call should finish afterwards:

```
DEBUG:artiq.protocols.pc_rpc:calling move_home()
DEBUG:artiq.devices.thorlabs_tcube.driver:sending: <Message MGMSG.MOT_MOVE_HOME p1=0x01 p2=0x00 dest=0x50 src=0x01>
DEBUG:artiq.devices.thorlabs_tcube.driver:received header: b'D\x04\x01\x00\x01P'
DEBUG:artiq.devices.thorlabs_tcube.driver:receiving: <Message MGMSG.MOT_MOVE_HOMED p1=0x01 p2=0x00 dest=0x01 src=0x50>
```

You can locate your ``driver.py`` this way:
```
>>> from artiq.devices.thorlabs_tcube import driver
>>> driver
<module 'artiq.devices.thorlabs_tcube.driver' from '/home/sb/artiq/artiq/devices/thorlabs_tcube/driver.py'>
```

(The new debug feature will be part of the new conda packages after someones figures out what is wrong with conda again)

kgilmo, 2015-11-25:
I've updated the t_cube driver to e1e082e2ec2a7a5dec155508a4430248524815ad.

Using either the working translation stage or the rotation stage, which still freezes up, I don't see any messages printed.

```bash

(py35)rabi@68810artiq1:~/gitlab/nistpenning/artiq$ thorlabs_tcube_controller -P tdc001 -d hwgrep://0403:faf0 -v -v
DEBUG:artiq.devices.thorlabs_tcube.driver:baud rate set to 115200
DEBUG:asyncio:Using selector: EpollSelector

```

I used the following as well, in case I need to specify the exact device (since there are several connected). Didn't seem to make a difference. 

```bash

(py35)rabi@68810artiq1:~/gitlab/nistpenning/artiq$ thorlabs_tcube_controller -P tdc001 -d /dev/ttyUSB3 -v -v
DEBUG:artiq.devices.thorlabs_tcube.driver:baud rate set to 115200
DEBUG:asyncio:Using selector: EpollSelector
```

Any idea why I'm not seeing the debug messages?

sbourdeauducq, 2015-11-26:
Have you sent any move commands?

kgilmo, 2015-11-26:
Yes. Using ```bash artiq_rpctool ::1 3255 call ``` for various move commands resulted in movement with no messages for the translation stage, and no movement with no messages for the rotation.

sbourdeauducq, 2015-11-26:
That does not make sense, if there is a movement it means the RPC has been received by the controller and you should see at least ``DEBUG:artiq.protocols.pc_rpc:calling move_home()`` which ARTIQ has been printing for many months when in debug mode.

sbourdeauducq, 2015-11-26:
All the messages I'm talking about are printed by the controller, not rpctool.

kgilmo, 2015-11-26:
I think there was another instance of the controller running. I'll check
this on Monday and get back to you.

sbourdeauducq, 2015-11-26:
Running two controllers on the same device may also cause the initial issue.

kgilmo, 2015-11-26:
There's only one running normally. I think I unintentionally ran a script
yesterday that starts up the controller, and then started it again
manually. I have seen the debug messages previously, so it has definitely
worked recently.

kgilmo, 2015-11-30:
Alright, with only one instance of the controller running I get the debug messages. Here's what this looks like for the translation stage that's been working:

```bash

(py35)rabi@68810artiq1:~/gitlab/nistpenning/artiq$ thorlabs_tcube_controller  -P tdc001 -p 50100 --bind 132.163.182.75 -d "hwgrep://USB VID:PID=0403:faf0 SNR=83833889" -v -v
DEBUG:artiq.devices.thorlabs_tcube.driver:baud rate set to 115200
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:artiq.protocols.pc_rpc:calling move_absolute(1000)
DEBUG:artiq.devices.thorlabs_tcube.driver:sending: <Message MGMSG.MOT_MOVE_ABSOLUTE p1=0x00 p2=0x00 dest=0xd0 src=0x01>
DEBUG:artiq.devices.thorlabs_tcube.driver:received header: b'd\x04\x0e\x00\x80\x00'
DEBUG:artiq.devices.thorlabs_tcube.driver:receiving: <Message MGMSG.MOT_MOVE_COMPLETED p1=0x0e p2=0x00 dest=0x80 src=0x00>

```

when I call

```bash

(py35)rabi@68810artiq1:~/gitlab/nistpenning/artiq$ artiq_rpctool 68810artiq1.bw.nist.gov 50100 call move_absolute 1000

```
and

```bash
DEBUG:artiq.protocols.pc_rpc:calling move_home()
DEBUG:artiq.devices.thorlabs_tcube.driver:sending: <Message MGMSG.MOT_MOVE_HOME p1=0x01 p2=0x00 dest=0x50 src=0x01>
DEBUG:artiq.devices.thorlabs_tcube.driver:received header: b'D\x04\x01\x00\x00\x00'
DEBUG:artiq.devices.thorlabs_tcube.driver:receiving: <Message MGMSG.MOT_MOVE_HOMED p1=0x01 p2=0x00 dest=0x00 src=0x00>
```

when I call

```bash
(py35)rabi@68810artiq1:~/gitlab/nistpenning/artiq$ artiq_rpctool 68810artiq1.bw.nist.gov 50100 call move_home
```

Contrasted with the rotation stage that is not working properly:

```bash
(py35)rabi@68810artiq1:~/gitlab/nistpenning/artiq$ thorlabs_tcube_controller  -P tdc001 -p 50101 --bind 132.163.182.75 -d "hwgrep://USB VID:PID=0403:faf0 SNR=83823340" -v -v
DEBUG:artiq.devices.thorlabs_tcube.driver:baud rate set to 115200
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:artiq.protocols.pc_rpc:calling move_absolute(1000)
DEBUG:artiq.devices.thorlabs_tcube.driver:sending: <Message MGMSG.MOT_MOVE_ABSOLUTE p1=0x00 p2=0x00 dest=0xd0 src=0x01>
```

at which point it hangs. This was using

```bash
(py35)rabi@68810artiq1:~/gitlab/nistpenning/artiq$ artiq_rpctool 68810artiq1.bw.nist.gov 50101 call move_absolute 1000
```

Same thing happens with ```bash move_home ```:

``` bash
(py35)rabi@68810artiq1:~/gitlab/nistpenning/artiq$ thorlabs_tcube_controller  -P tdc001 -p 50101 --bind 132.163.182.75 -d "hwgrep://USB VID:PID=0403:faf0 SNR=83823340" -v -v
DEBUG:artiq.devices.thorlabs_tcube.driver:baud rate set to 115200
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:artiq.protocols.pc_rpc:calling move_home()
DEBUG:artiq.devices.thorlabs_tcube.driver:sending: <Message MGMSG.MOT_MOVE_HOME p1=0x01 p2=0x00 dest=0x50 src=0x01>
```

Previously, I was able to send some move commands to the rotation stage before it would freeze. I'm not sure what's changed, but I seem to not be able to do that anymore. It's now completely unresponsive.

sbourdeauducq, 2015-12-01:
Can you use Wireshark to obtain USB dumps of these two scenarios (with the same device) and post them:

  1. homing with the ARTIQ driver until the motor stops moving and completion should have been detected
  2. homing with the other driver until it detects the completion correctly

Do you have the source code of the other driver, or an idea of the protocol it uses to detect completion?

sbourdeauducq, 2015-12-01:
Are Thorlabs offering support on this sort of thing? p. 41 of http://www.sal.wisc.edu/manga/archive/public/datasheets/thorlabs/APT_Communications_Protocol_Rev%203.pdf is pretty clear about what should happen, but the controller in this case does not send anything back (or sends fewer than 6 bytes).

kgilmo, 2015-12-01:
The driver that was working with the rotation stage is from prior to this commit: https://github.com/m-labs/artiq/commit/4d077f50c6621ad771818f489a5dd611e320ea6d.

I'm not sure about Thorlabs support. If we continue having problems, I'll contact them to see.

jboulder, 2015-12-01:
> commit: 4d077f5

I'd get on the horn with Thorlabs as a parallel track. If the devices are
not returning the expected byte payload this sounds like a bug.

kgilmo, 2015-12-01:
The problem seems to be with the tcubes. We have a few different 'issues' of these things, and some don't work properly with the controller. Issue 2 MOD A and Issue 4 work fine, but Issue 2 MOD 0 (which the rotation stages I was trying to use are connected to) and Issue 4 MOD B do not work properly - ie they don't return anything and hang.

I'm contacting thorlabs about this.

sbourdeauducq, 2015-12-01:
The previous driver did not check for completion of the commands, so of course it was returning...

kgilmo, 2016-01-13:
This issue has seemingly been tracked down:

The older tcube models must first be conncted via usb to the computer before they are powered on. If power is supplied first, or if the usb is disconnected and reconnected without power cycling, the tcubes fail to respond appropriately. 

If the tcube is in this state, power cycling seems to bring it back to normal working order.

We ran into this issue because the tcubes had been disconnected and reconnected to the computer without power cycling. Upon power cycling they so far work fine.

closed 2016-01-13 by kgilmo