m-labs/artiq #170: artiq_rpctool targets don't accept string arguments

kgilmo, 2015-10-27:
Initialized pdq2 with:
```bash
(py35)rabi2@vboxartiq:~$ pdq2_controller --bind ::1 -p 3252 -d /dev/ttyUSB2 -v -v
DEBUG:asyncio:Using selector: EpollSelector
```

When attempting to use command "cmd", get the following error message:

```bash
(py35)rabi2@vboxartiq:~$ artiq_rpctool ::1 3252 call cmd "RESET" True
Traceback (most recent call last):
  File "/home/rabi2/anaconda3/envs/py35/bin/artiq_rpctool", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/frontend/artiq_rpctool.py", line 97, in main
    call_method(remote, args.method, args.args)
  File "/home/rabi2/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/frontend/artiq_rpctool.py", line 79, in call_method
    ret = method(*[eval(arg) for arg in args])
  File "/home/rabi2/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/frontend/artiq_rpctool.py", line 79, in <listcomp>
    ret = method(*[eval(arg) for arg in args])
  File "<string>", line 1, in <module>
NameError: name 'RESET' is not defined
(py35)rabi2@vboxartiq:~$ 
```

This error occurs in general when passing a string to the command.

```bash
(py35)rabi2@vboxartiq:~$ artiq_rpctool ::1 3252 call funny 'TRUE'
Traceback (most recent call last):
  File "/home/rabi2/anaconda3/envs/py35/bin/artiq_rpctool", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/frontend/artiq_rpctool.py", line 97, in main
    call_method(remote, args.method, args.args)
  File "/home/rabi2/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/frontend/artiq_rpctool.py", line 79, in call_method
    ret = method(*[eval(arg) for arg in args])
  File "/home/rabi2/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/frontend/artiq_rpctool.py", line 79, in <listcomp>
    ret = method(*[eval(arg) for arg in args])
  File "<string>", line 1, in <module>
NameError: name 'TRUE' is not defined
```
The relevant definitions in the driver, including a definition that was added to Robert's driver for debugging:
```python
    def funny(self, cmd):
        print(cmd)
```

```python

    def cmd(self, cmd, enable):
        cmd = self._commands.index(cmd) << 1
        if not enable:
            cmd |= 1
        self.write(struct.pack("cb", self._escape, cmd))
```

Pdq2 driver at artiq/devices/pdq2/driver.py:

```python
# Copyright (C) 2012-2015 Robert Jordens <jordens@gmail.com>

from math import log, sqrt
import logging
import struct

import serial

from artiq.wavesynth.coefficients import discrete_compensate

logger = logging.getLogger(__name__)

[... 184 lines cut ...]
            duration = line["duration"]
            trigger = line.get("trigger", False)
            for segment, data in zip(segments, line["channel_data"]):
                if len(data) != 1:
                    raise ValueError("only one target per channel and line "
                                     "supported")
                for target, target_data in data.items():
                    getattr(segment, target)(
                        shift=shift, duration=duration, trigger=trigger,
                        **target_data)
        # append an empty line to stall the memory reader before jumping
        # through the frame table
        for segment in segments:
            segment.line(typ=3, data=b"", trigger=True, duration=10)
        return segments

    def program(self, program):
        self.clear_all()
        for frame_data in program:
            self.program_frame(frame_data)
        self.write_all()

    def ping(self):
        return True
```

kgilmo, 2015-10-27:
Not sure if this is a related or not, but additionally when I run pdq2_client I get the following error:

```bash
(py35)rabi2@vboxartiq:~/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/frontend$ pdq2_client -s ::1 --port 3252 -b -v -v
Traceback (most recent call last):
  File "/home/rabi2/anaconda3/envs/py35/bin/pdq2_client", line 6, in <module>
    sys.exit(main())
  File "/home/rabi2/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/frontend/pdq2_client.py", line 58, in main
    dev.init()
  File "/home/rabi2/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pc_rpc.py", line 163, in proxy
    return self.__do_rpc(name, args, kwargs)
  File "/home/rabi2/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pc_rpc.py", line 155, in __do_rpc
    return self.__do_action(obj)
  File "/home/rabi2/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pc_rpc.py", line 149, in __do_action
    raise RemoteError(obj["message"])
artiq.protocols.pc_rpc.RemoteError: Traceback (most recent call last):
  File "/home/rabi2/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pc_rpc.py", line 491, in _handle_connection_cr
    method = getattr(target, obj["name"])
AttributeError: 'Pdq2' object has no attribute 'init'
```
with pdq2_controller running

```bash
(py35)rabi2@vboxartiq:~$ pdq2_controller --bind ::1 -p 3252 -d /dev/ttyUSB2 -v -v
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:artiq.protocols.pc_rpc:calling init()
```

sbourdeauducq, 2015-10-28:
" characters are interpreted by your shell and need to be escaped, e.g. \"RESET\"

sbourdeauducq, 2015-10-28:
The other problem is unrelated and comes from the fact that pdq2_client is somewhat out of sync with the latest changes in the driver. @jordens Does the new pdq2 driver still need the equivalent of that init() call?

jordens, 2015-10-28:
I don't remember why `init()` was added. According to blame, you did that.

sbourdeauducq, 2015-10-28:
There is the same problem with ``set_freq``.

jordens, 2015-12-17:
Closing in favor of #178
Please provide comments there if anything from this bug here gets lost there.

closed 2015-12-17 by jordens