m-labs/artiq #171: Scannable GUI does not always show ndecimals

jboulder, 2015-10-28:
>        self.scanf = self.get_argument("frequency_scan", Scannable(
            default=LinearScan(1e6, 2e6, 100), scale=1e6, unit="MHz", ndecimals=4))

Layout of GUI should always show ndecimals in text box. Consider using reflow to new line to get GUI elements to fit while always showing ndecimals.

jboulder, 2015-10-28:
https://github.com/m-labs/artiq/issues/163#issuecomment-151708097

sbourdeauducq, 2015-10-29:
The problem goes beyond ndecimals: if a spinbox does not have min/max boundaries it can need an arbitrarily large size.

The proper way to fix this is to have Qt do whatever layout adjustments are necessary, including expanding windows, so that all spinboxes can display the full contents that they have at the current time. As far as I can tell Qt does not already support this, and it is difficult to implement.

sbourdeauducq, 2015-10-29:
Actually, Qt is capable of resizing windows, as demonstrated by the following program:
```python
import sys
from PyQt4 import QtGui, QtCore

def main():
    app = QtGui.QApplication(sys.argv)

    w = QtGui.QMainWindow()
    label = QtGui.QLabel("This is the initial text")
    w.setCentralWidget(label)
    w.show()

    timer = QtCore.QTimer()
    timer.timeout.connect(lambda: label.setText(label.text() + " tick"))
    timer.start(1000)
    
    sys.exit(app.exec_())

if __name__ == '__main__':
    main()
```

Let me see if that can be done for QSpinBox too ...

sbourdeauducq, 2015-10-29:
Qt actually often sets a sensible size hint for the spinboxes (and takes into account ``global_min`` and ``global_max``) and the scan widget. However, the size hint and the size policy are completely ignored by the ``QTreeWidget`` that puts them in the list...

jboulder, 2015-11-20:
Any developments on how to fix this? Suggestions follow:

Replacement of radio buttons with drop-down box for {No scan, Linear, Random, Explicit} would save space. # Point could be to the left of the drop-down on the same line. 

Consider removing "Min" and "Max:" labels and replacing with min_box " to " max_box. And, remove the up-down arrow buttons since keyboard up/down pgup/pgdn shortcuts are available. 

[image]

sbourdeauducq, 2015-11-20:
This is difficult to fix (I suspect the root of the problem is deep in the C++ code of Qt) and the consequences are minor (just make the window large enough manually). Low priority imo.

The other points should go into another issue. Anyway I like the radio buttons as they visually distinguish the scan widget from enumerations.

jboulder, 2015-11-20:
SB, We do spectroscopy here. Being unable to display long numbers is a problem. I'm not suggesting a repair of QT, rather a widget rejigging that is readable. How about Min and Max on separate lines?

sbourdeauducq, 2015-11-20:
9826b19 does this. The Qt bug is still present though (it is also shown on the radio buttons on your screenshot, e.g. the "n" of "scan" is missing).

sbourdeauducq, 2015-11-20:
Conda is acting up again, so please patch your local file until it is fixed.

sbourdeauducq, 2015-12-05:
New packages are available now. Also, the Qt problem only appears with ``stretchLastSection=True`` and the new layout (with argument recompute buttons) will not use this feature.

commit e8bd24d5 (2015-12-05): gui/experiments: add argument recompute buttons. Closes #171
closed 2015-12-05 by sbourdeauducq