m-labs/artiq #172: when one program ends, it also terminates adjacent scheduled program

jboulder, 2015-10-29:
Observation: when one program ends, it also terminates adjacent scheduled program

I don't understand the logging output. 

```python
rabi@688penningA:~/gitlab/nistpenning/artiq$ tail -n 100 master_log.txt
2015-10-28 20:15:25,836 INFO:controller(trans1):print:    raise serial.SerialException('no ports found matching regexp %r' % (url,))
2015-10-28 20:15:25,837 INFO:controller(trans1):print:serial.serialutil.SerialException: no ports found matching regexp 'USB VID:PID=0403:faf0 SNR=83833889'
2015-10-28 20:15:25,865 WARNING:ctlmgr(68810artiq1):artiq.frontend.artiq_ctlmgr:Controller trans1 exited
2015-10-28 20:15:25,865 WARNING:ctlmgr(68810artiq1):artiq.frontend.artiq_ctlmgr:Restarting in 1039.8 seconds
2015-10-28 20:15:48,997 ERROR:worker(10351):root:Worker terminating with exception
2015-10-28 20:15:48,998 INFO:worker(10351):print:Traceback (most recent call last):
2015-10-28 20:15:48,999 INFO:worker(10351):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 198, in main
2015-10-28 20:15:48,999 INFO:worker(10351):print:    exp_inst.run()
2015-10-28 20:15:49,000 INFO:worker(10351):print:  File "/home/rabi/gitlab/nistpenning/artiq/repository/default_std.py", line 56, in run
2015-10-28 20:15:49,001 INFO:worker(10351):print:    self.run_kernel()
2015-10-28 20:15:49,001 INFO:worker(10351):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/language/core.py", line 121, in run_on_core
2015-10-28 20:15:49,002 INFO:worker(10351):print:    return exp.core.run(arg, ((exp,) + k_args), k_kwargs)
2015-10-28 20:15:49,003 INFO:worker(10351):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/coredevice/core.py", line 128, in run
2015-10-28 20:15:49,004 INFO:worker(10351):print:    self.comm.load(binary)
2015-10-28 20:15:49,004 INFO:worker(10351):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 129, in load
2015-10-28 20:15:49,005 INFO:worker(10351):print:    _, ty = self._read_header()
2015-10-28 20:15:49,006 INFO:worker(10351):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 82, in _read_header
2015-10-28 20:15:49,007 INFO:worker(10351):print:    (c, ) = struct.unpack("B", self.read(1))
2015-10-28 20:15:49,007 INFO:worker(10351):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/coredevice/comm_tcp.py", line 53, in read
2015-10-28 20:15:49,008 INFO:worker(10351):print:    raise IOError("Connection closed")
2015-10-28 20:15:49,009 INFO:worker(10351):print:OSError: Connection closed
2015-10-28 20:15:49,051 ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 10351
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 276, in _do
    completed = await run.resume(request_termination)
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 32, in worker_method
    return await m(*args, **kwargs)
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 230, in resume
    "data": request_termination})
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 196, in _worker_action
    completed = await self._handle_worker_requests()
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 153, in _handle_worker_requests
    obj = await self._recv(self.watchdog_time())
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 141, in _recv
    raise WorkerError("Worker ended while attempting to receive data")
artiq.master.worker.WorkerError: Worker ended while attempting to receive data
2015-10-28 20:17:22,570 ERROR:worker(10354):root:Worker terminating with exception
2015-10-28 20:17:22,571 INFO:worker(10354):print:Traceback (most recent call last):
2015-10-28 20:17:22,572 INFO:worker(10354):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 198, in main
2015-10-28 20:17:22,573 INFO:worker(10354):print:    exp_inst.run()
2015-10-28 20:17:22,574 INFO:worker(10354):print:  File "/home/rabi/gitlab/nistpenning/artiq/repository/default_std.py", line 56, in run
2015-10-28 20:17:22,574 INFO:worker(10354):print:    self.run_kernel()
2015-10-28 20:17:22,575 INFO:worker(10354):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/language/core.py", line 121, in run_on_core
2015-10-28 20:17:22,576 INFO:worker(10354):print:    return exp.core.run(arg, ((exp,) + k_args), k_kwargs)
2015-10-28 20:17:22,577 INFO:worker(10354):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/coredevice/core.py", line 128, in run
2015-10-28 20:17:22,577 INFO:worker(10354):print:    self.comm.load(binary)
2015-10-28 20:17:22,578 INFO:worker(10354):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 129, in load
2015-10-28 20:17:22,579 INFO:worker(10354):print:    _, ty = self._read_header()
2015-10-28 20:17:22,580 INFO:worker(10354):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 82, in _read_header
2015-10-28 20:17:22,581 INFO:worker(10354):print:    (c, ) = struct.unpack("B", self.read(1))
2015-10-28 20:17:22,582 INFO:worker(10354):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/coredevice/comm_tcp.py", line 53, in read
2015-10-28 20:17:22,582 INFO:worker(10354):print:    raise IOError("Connection closed")
2015-10-28 20:17:22,583 INFO:worker(10354):print:OSError: Connection closed
2015-10-28 20:17:22,625 ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 10354
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 276, in _do
    completed = await run.resume(request_termination)
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 32, in worker_method
    return await m(*args, **kwargs)
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 230, in resume
    "data": request_termination})
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 196, in _worker_action
    completed = await self._handle_worker_requests()
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 153, in _handle_worker_requests
[... 13 lines cut ...]
2015-10-28 20:17:34,662 INFO:worker(10356):print:    self.comm.load(binary)
2015-10-28 20:17:34,663 INFO:worker(10356):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 129, in load
2015-10-28 20:17:34,664 INFO:worker(10356):print:    _, ty = self._read_header()
2015-10-28 20:17:34,664 INFO:worker(10356):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 82, in _read_header
2015-10-28 20:17:34,665 INFO:worker(10356):print:    (c, ) = struct.unpack("B", self.read(1))
2015-10-28 20:17:34,666 INFO:worker(10356):print:  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/coredevice/comm_tcp.py", line 53, in read
2015-10-28 20:17:34,667 INFO:worker(10356):print:    raise IOError("Connection closed")
2015-10-28 20:17:34,667 INFO:worker(10356):print:OSError: Connection closed
2015-10-28 20:17:34,707 ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 10356
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 276, in _do
    completed = await run.resume(request_termination)
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 32, in worker_method
    return await m(*args, **kwargs)
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 230, in resume
    "data": request_termination})
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 196, in _worker_action
    completed = await self._handle_worker_requests()
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 153, in _handle_worker_requests
    obj = await self._recv(self.watchdog_time())
  File "/home/rabi/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 141, in _recv
    raise WorkerError("Worker ended while attempting to receive data")
artiq.master.worker.WorkerError: Worker ended while attempting to receive data

```

sbourdeauducq, 2015-10-29:
Looks like you are just running a series of experiments that fail one by one to do something related to the core device.

jboulder, 2015-10-30:
Suggestions on how to debug this? What can cause the following exception?
> artiq.master.worker.WorkerError: Worker ended while attempting to receive data

sbourdeauducq, 2015-10-30:
The relevant part is before. The core device closes the connection for some reason. Can you run the problematic experiment again, and then use ``artiq_coretool log`` to get the core device low-level log?

sbourdeauducq, 2015-10-31:
Or could it be that you are pausing from a kernel RPC? Does the problem also manifest itself when the experiment is run alone, or through ``artiq_run``?

The proper way to use pause with kernels is to disconnect from the core device (``self.core.comm.close()``) before pausing, which precludes RPCing ``pause`` calls as well.

Otherwise, the core device will disconnect the paused experiment (which will cause this backtrace) when the preempting experiment connects.

sbourdeauducq, 2016-01-14:
@whitequark could you add support for non-RPCable functions (a flag on a function that causes a compile-time error when attempting to call it) and mark the pause function as such?

whitequark, 2016-01-15:
Will do

whitequark, 2016-01-15:
Done

closed 2016-01-15 by sbourdeauducq