m-labs/artiq #173: exception reporting verbosity control

jboulder, 2015-10-29:
This topic was initially discussed here https://github.com/m-labs/artiq/issues/145#issuecomment-151944564 . It should be it's own topic. 

The present log output is distressingly long in the event of routine ARTIQ exceptions. Below is an example of attempting to use an unsupported type with set_dataset().

```python 
2015-10-29 11:02:20,990 ERROR:worker(27):root:Worker terminating with exception
2015-10-29 11:02:20,993 INFO:worker(27):print:Traceback (most recent call last):
2015-10-29 11:02:20,995 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_impl.py
", line 198, in main
2015-10-29 11:02:20,997 INFO:worker(27):print:    exp_inst.run()
2015-10-29 11:02:21,000 INFO:worker(27):print:  File "/home/jwbritto/gitlab/nistpenning/artiq/repository/diagnostic/test_hdf5.py", line 36, in r
un
2015-10-29 11:02:21,003 INFO:worker(27):print:    persist=False, save=False, broadcast=True)
2015-10-29 11:02:21,006 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/language/environment.
py", line 222, in set_dataset
2015-10-29 11:02:21,008 INFO:worker(27):print:    return self.__dataset_mgr.set(key, value, broadcast, persist, save)
2015-10-29 11:02:21,009 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_db.py",
 line 184, in set
2015-10-29 11:02:21,011 INFO:worker(27):print:    self.broadcast[key] = (persist, value)
2015-10-29 11:02:21,013 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/sync_struct
.py", line 179, in __setitem__
2015-10-29 11:02:21,015 INFO:worker(27):print:    "value": value})
2015-10-29 11:02:21,017 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_impl.py
", line 35, in parent_action
2015-10-29 11:02:21,018 INFO:worker(27):print:    put_object(request)
2015-10-29 11:02:21,020 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker_impl.py
", line 19, in put_object
2015-10-29 11:02:21,021 INFO:worker(27):print:    ds = pyon.encode(obj)
2015-10-29 11:02:21,022 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pyon.py", l
ine 143, in encode
2015-10-29 11:02:21,024 INFO:worker(27):print:    return _Encoder(pretty).encode(x)
2015-10-29 11:02:21,026 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pyon.py", l
ine 135, in encode
2015-10-29 11:02:21,028 INFO:worker(27):print:    return getattr(self, "encode_" + _encode_map[type(x)])(x)
2015-10-29 11:02:21,030 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pyon.py", l
ine 99, in encode_dict
2015-10-29 11:02:21,031 INFO:worker(27):print:    for k, v in x.items()])
2015-10-29 11:02:21,033 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pyon.py", l
ine 99, in <listcomp>
2015-10-29 11:02:21,034 INFO:worker(27):print:    for k, v in x.items()])
2015-10-29 11:02:21,035 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pyon.py", l
ine 135, in encode
2015-10-29 11:02:21,037 INFO:worker(27):print:    return getattr(self, "encode_" + _encode_map[type(x)])(x)
2015-10-29 11:02:21,038 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pyon.py", l
ine 99, in encode_dict
2015-10-29 11:02:21,040 INFO:worker(27):print:    for k, v in x.items()])
2015-10-29 11:02:21,043 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pyon.py", l
ine 99, in <listcomp>
2015-10-29 11:02:21,044 INFO:worker(27):print:    for k, v in x.items()])
2015-10-29 11:02:21,046 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pyon.py", l
ine 135, in encode
2015-10-29 11:02:21,047 INFO:worker(27):print:    return getattr(self, "encode_" + _encode_map[type(x)])(x)
2015-10-29 11:02:21,049 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pyon.py", l
ine 85, in encode_tuple
2015-10-29 11:02:21,050 INFO:worker(27):print:    r += ", ".join([self.encode(item) for item in x])
2015-10-29 11:02:21,052 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pyon.py", l
ine 85, in <listcomp>
2015-10-29 11:02:21,053 INFO:worker(27):print:    r += ", ".join([self.encode(item) for item in x])
2015-10-29 11:02:21,055 INFO:worker(27):print:  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/protocols/pyon.py", l
ine 135, in encode
2015-10-29 11:02:21,056 INFO:worker(27):print:    return getattr(self, "encode_" + _encode_map[type(x)])(x)
2015-10-29 11:02:21,057 INFO:worker(27):print:KeyError: <class 'complex'>
2015-10-29 11:02:21,098 ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 27
Traceback (most recent call last):
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 279, in _do
    completed = await run.run()
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 32, in worker_method
    return await m(*args, **kwargs)
  File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/worker.py", line 220, in run
    completed = await self._worker_action({"action": "run"})
[... 42 lines cut ...]
    File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 279, in _docompleted = await run.run()
    File "/home/jwbritto/anaconda3/envs/py35/lib/python3.5/site-packages/artiq/master/scheduler.py", line 32, in worker_method return await m(*args, **kwargs)
    ....
    artiq.master.worker.WorkerError: Worker ended while attempting to receive data

2015-10-29 11:02:21,098 ERROR:master: Worker ended while attempting to receive data
```

Note that the last line of the trackeback appears as both part of the INFO logged Traceback() and logged again at the ERROR level. This permits easy filtering by the user. 

Here's how the same looks with the Log filter set to level ERROR.
```python
2015-10-29 11:02:20,990 ERROR:worker(27):root: Worker terminating with exception
2015-10-29 11:02:21,057 ERROR:worker(27):print:KeyError: <class 'complex'>
2015-10-29 11:02:21,098 ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 27
2015-10-29 11:02:21,098 ERROR:master: Worker ended while attempting to receive data
```
If the filter is set to CRITICAL the output is as concise as I'd like it to be. 
```python
(* EVENT LEVEL TABLE ENTRY *) KeyError: <class 'complex'>
(* ALL BELOW LIES BELOW THIS ENTRY *)
2015-10-29 11:02:21,057 CRITICAL:worker(27):print:KeyError: <class 'complex'>
```

I'm certain that my understanding of the innards of python exception handling is naive.  Let's figure out how to use the nice Logger capability to present concise, actionable information to the ARTIQ user.

sbourdeauducq, 2015-11-22:
see "173a/173b" in https://ssl.serverraum.org/lists-archive/artiq/2015-November/000595.html

sbourdeauducq, 2016-01-21:
Keeping human readability, a multi-line log entry could be formatted by the program emitting it as:
``LEVEL<nlines>:module:message``
The parser would then take the following nlines-1 lines and combine them with the same log entry. If nlines=1, the <nlines> element can be omitted.

sbourdeauducq, 2016-01-26:
In the worker_pipeipc branch (not Windows compatible yet) multiline log entries can now be passed around (worker -> master, controller -> manager -> master). The full log entries (all lines) have the correct level in the GUI now. Folding multiline entries using QTreeView is more complicated (and it would not surprise me that attempting this will wake up more Qt bugs).

sbourdeauducq, 2016-02-11:
There is no folding, but multiline log entries are displayed in a single cell in the GUI now:
[image]

I plan to release this as it is for 1.0. Open a new issue if you really need folding.

closed 2016-02-11 by sbourdeauducq