m-labs/artiq #174: build system exports failing builds via conda

jboulder, 2015-11-04:
Conda offers builds even if they fail build tests. Presently, conda offers 976e136 but this build fails on Travis. I propose that only builds that pass should be made available via Conda. 

>  573.2  Python: 3.5  BUILD_SOC=pipistrello-nist_qc1  30 min 19 sec

sbourdeauducq, 2015-11-05:
What is happening here is the build of some gateware packages fail (in this particular case due to ISE non-determinism) and they are not uploaded, while the build of other packages succeeds and they are uploaded.
I'm not sure if Travis has any option to permit uploading all packages once at the end if all sub-builds have succeeded...

whitequark, 2015-11-05:
No, it does not. I am not sure if that's possible to easily implement. Let me think about it for a while.

whitequark, 2015-11-05:
So, I thought about deleting all other artifacts during a failed build, but there is a race condition: if the failed build is the first one, the others will not get deleted.

A possibility is a script running in after_success detecting if all three artifacts have been uploaded and then promoting them from e.g. `in-progress` channel to `dev` channel. Only issue is, anaconda CLI tool does not allow to change the channels, like the web UI, only to reupload the artifacts...

sbourdeauducq, 2015-11-05:
Sounds hairy to me and can cause more funny bugs if the anaconda upload is not atomic.

sbourdeauducq, 2015-11-05:
And the consequences of this problem are minor and the same as not flashing the board (and this situation can be easily detected using a runtime/artiq git version check as discussed in #97). If there is no clean solution, we should probably not attempt to fix this.

whitequark, 2015-11-05:
Yeah, I agree that #97 is the way to go. Such failures are to be expected on the dev channel anyway.

closed 2015-11-05 by whitequark