m-labs/artiq #178: PDQ2 Driver Inconsistencies

kgilmo, 2015-11-24:
The versions of the PDQ2 driver in Artiq and in the ions gitweb are different. In the gitweb history, commit 3d93519b33f81b2be79934019f0cf15447374bfd shows a merge with the artiq version, though they are no longer the same. The issue is with the program_frame definition, which seems buggy in the artiq version. Here's the difference:

```python

--- a/qc/hardware/pdq2.py
+++ b/qc/hardware/pdq2.py
@@ -104,7 +104,6 @@ class Channel:
         del self.segments[:]
 
     def new_segment(self):
-        # assert len(self.segments) < self.num_frames
         segment = Segment()
         self.segments.append(segment)
         return segment
@@ -200,24 +199,24 @@ class Pdq2:
         for i, line in enumerate(frame_data):  # segments are concatenated
             dac_divider = line.get("dac_divider", 1)
             shift = int(log(dac_divider, 2))
-            assert 2**shift == dac_divider
+            if 2**shift != dac_divider:
+                raise ValueError("only power-of-two dac_dividers supported")
             duration = line["duration"]
             trigger = line.get("trigger", False)
-            if i == 0:
-                assert trigger
-                #trigger = False  # use wait on the last line
-                # jumping through frame table takes 3-4 cycles
-                # just ensure that time betweeen triggers is always longer than this
-                # and that the first trigger covers both, the end on the last and the
-                # trigger on the first
-            eof = i == len(frame_data) - 1
             for segment, data in zip(segments, line["channel_data"]):
-                assert len(data) == 1
+                if len(data) != 1:
+                    raise ValueError("only one target per channel and line "
+                                     "supported")
                 for target, target_data in data.items():
                     getattr(segment, target)(
-                        shift=shift, duration=duration,
-                        trigger=trigger, wait=eof, jump=eof,
+                        shift=shift, duration=duration, trigger=trigger,
                         **target_data)
+        # append an empty line to stall the memory reader before jumping
+        # through the frame table (`wait` does not prevent reading
+        # the next line)
+        for segment in segments:
+            segment.line(typ=3, data=b"", trigger=True, duration=10,
+                         jump=True)
         return segments
 
     def program(self, program):

```

sbourdeauducq, 2015-11-25:
@jordens what is your take on this?

jordens, 2015-11-25:
This is one of the things that I had changed in my code in the lab. The `pdq2` driver and `wavesynth` need a bit of work to integrate and finish. I am uncertain who can and will do that.

jboulder, 2015-11-30:
>This is one of the things that I had changed in my code in the lab.

It's a huge benefit to expose the code that you've (@jordens
<https://github.com/jordens>) refined in the lab while working on ion
transport. What other things differ between your code in the lab and the
code on github?

Is it possible to simply commit your updated code? As @kgilmo points out,
what's on github is broken anyway. Perhaps m-labs has the bandwidth to
streamline integration.

>The pdq2 driver and wavesynth need a bit of work to integrate
>and finish. I am uncertain who can and will do that.

Can you outline the steps needed to complete the integration?

-Joe

jordens, 2015-12-01:
Joe, good that you want to work on this. Kevin has attached a full diff to the first post in this issue.

Committing this is certainly possible but does not help much.

To get somewhere, you have to explore the wavesynth topic and determine convince yourself that the data format and the API are viable, useful, and to what extent and how it can be mapped to the dumb DACs and the smart ones. Then the different ways of synchronizing and steering the DACs need to be designed and matched with the capabilities of the hard- and gateware. Much of all this is already done but it should be revisited. Then all the things from wavesynth over the different software and gateware simulation methods to the hardware backends as well as the unittests have to be pieced together and linked.

jordens, 2015-12-17:
I'll redesign significant parts of the software and gateware stack. This should cover the upper layers of the gateware (communication, protocol, parsing), the triggering/frame selection logic, and the binary pdq2 data assembly. The goal is to get the most out of this annoying bit of hardware while still producing code that has maximum reusability for the future artiq hardware. It seems wise to actually implement SPI on the old frame-select/trigger lines (that will get rid of many problems) and implement sufficient interlocking for the data transfer process.

dhslichter, 2015-12-17:
Would the notion be to replace the USB communications with SPI communications, or to have them be supplementary?  In other words, will the SPI be used just to communicate frame selection information or would it be more generally for waveform data upload/modification?  If the latter, we need to ensure that the core device is not unduly bogged down by communicating waveform data to the PDQs.  In addition, it seems it would make sense to implement the SPI in such a manner that multiple PDQs can share the same serial bus, lest we run out of TTL lines.  I assume this would be a 3-wire SPI, leaving one remaining input on the PDQ cards to provide a TTL trigger.

jordens, 2016-02-22:
This should work now. I need a guinea pig for testing. @raghavendrasrinivas @dhslichter 
Pending improvements through #276.

raghavendrasrinivas, 2016-02-23:
I'll try to get to this this week but probably not before Wednesday.

kgilmo, 2016-03-02:
Updated: pdq_client, pdq_controller, coefficients, mediator, and driver. 

Were able to use rpc_tool to find target pdq2 and ping and list methods. However, when calling commands run into trouble:

``
 artiq_rpctool ::1 50101 call cmd "START" True
``

``
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/newartiq/bin/artiq_rpctool", line 6, in <module>
    sys.exit(main())
  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/artiq/frontend/artiq_rpctool.py", line 97, in main
    call_method(remote, args.method, args.args)
  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/artiq/frontend/artiq_rpctool.py", line 79, in call_method
    ret = method(*[eval(arg) for arg in args])
  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/artiq/frontend/artiq_rpctool.py", line 79, in <listcomp>
    ret = method(*[eval(arg) for arg in args])
  File "<string>", line 1, in <module>
NameError: name 'START' is not defined
``

kgilmo, 2016-03-02:
So, we've tested the pdq2_controller with rpc_tool and can send it commands. Is there anything in particular to test?

sbourdeauducq, 2016-03-02:
Yes. Try https://github.com/m-labs/artiq/blob/master/examples/master/repository/coredevice_examples/transport.py (or some variation thereof).

sbourdeauducq, 2016-03-02:
To avoid the recurrent mistake with Python string escapes in shell commands I have added an interactive mode to ``rpctool`` ...

```
$ artiq_rpctool.py localhost 3253
(lda) set_attenuation("lskjd")
Remote exception:
Traceback (most recent call last):
  File "/home/sb/artiq/artiq/protocols/pc_rpc.py", line 514, in _handle_connection_cr
    ret = method(*obj["args"], **obj["kwargs"])
  File "/home/sb/artiq/artiq/devices/lda/driver.py", line 50, in set_attenuation
    att = round(attenuation/step)*step
TypeError: unsupported operand type(s) for /: 'str' and 'float'

(lda) set_attenuation(3.4)
(lda) get_attenuation()
3.5
(lda)
```

raghavendrasrinivas, 2016-03-02:
We're currently testing this in the Penning lab with just one pdq2 connected, and could potentially have 3. How should we define electrodes in the device_database? It seems like it's calling functions from the mediator which isn't imported into the pdq2 driver.

jordens, 2016-03-02:
https://github.com/m-labs/artiq/blob/master/examples/master/device_db.pyon#L137
and the following entries.

jordens, 2016-04-08:
@kgilmo , @r-srinivas: how is testing going? Can we close this?

r-srinivas, 2016-04-09:
No not yet. We had some software issues on the computer in the penning lab so we were going to test it on the new pdqs we were setting up for the magtrap but that hasn't finished yet. Maybe next week.

sbourdeauducq, 2016-04-15:
@kgilmo @r-srinivas @dhslichter This is the last hardware thing that you need to test before 1.0.

r-srinivas, 2016-05-03:
I tried running transport.py on 1.0rc3 on linux. With the default values I get an underflow error,

```
ERROR:worker(3496,transport.py):root:Terminating with exception (RTIOUnderflow: RTIO underflow at 951745453455 mu, channel 0, slack -3785 mu)
Core Device Traceback (most recent call last):
  File "/home/rabi/artiq-master/repository/examples/transport.py", line 88, in artiq_worker_transport.Transport.repeat (RA=+0x16d8)
    n = self.one()
  File "/home/rabi/artiq-master/repository/examples/transport.py", line 81, in artiq_worker_transport.Transport.one (inlined)
    return self.detect()
  File "/home/rabi/artiq-master/repository/examples/transport.py", line 72, in artiq_worker_transport.Transport.detect (inlined)
    self.bd_sw.off()
  File "<artiq>/coredevice/ttl.py", line 141, in artiq.coredevice.ttl.TTLInOut.off (inlined)
    self.set_o(False)
  File "<artiq>/coredevice/ttl.py", line 119, in __modinit__ (inlined)
    rtio_output(now_mu(), self.channel, 0, 1 if o else 0)
  File "/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/work/artiq/runtime/rtio.c", line 28, in rtio_process_exceptional_status
    <unknown>
artiq.coredevice.exceptions.RTIOUnderflow(0): RTIO underflow at 951745453455 mu, channel 0, slack -3785 mu
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/artiq-2016-04-25/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 220, in main
    exp_inst.run()
  File "/home/rabi/artiq-master/repository/examples/transport.py", line 105, in run
    self.scan(stops)
  File "/home/rabi/artiq-master/repository/examples/transport.py", line 100, in scan
    self.repeat()
  File "/home/rabi/anaconda3/envs/artiq-2016-04-25/lib/python3.5/site-packages/artiq/language/core.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-25/lib/python3.5/site-packages/artiq/coredevice/core.py", line 114, in run
    self.comm.serve(object_map, symbolizer)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-25/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 527, in serve
    self._serve_exception(object_map, symbolizer)
  File "/home/rabi/anaconda3/envs/artiq-2016-04-25/lib/python3.5/site-packages/artiq/coredevice/comm_generic.py", line 519, in _serve_exception
    raise python_exn
artiq.coredevice.exceptions.RTIOUnderflow: RTIO underflow at 951745453455 mu, channel 0, slack -3785 mu
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 3496
```

That seems to be fixed by adding a delay between self.pmt.count() and turning bd_sw.off()

```diff
    @kernel
    def detect(self):
        self.bd_sw.on()
        self.pmt.gate_rising(100*us)
        count = self.pmt.count()
+       delay(1*ms)
        self.bd_sw.off()
        return 3000
```

It's a little hard to test the waveforms now because our artiq_crate and pdqs are in different rooms. It seems like connecting to the pdq_controllers works fine. I'd suggest maybe closing the issue for now and I'll do a more thorough test about timing issues once we are better set up to do this. I'll raise any issues then if I run into them.

jordens, 2016-05-03:
That code snippet is not what's in `transport.py`.

closed 2016-05-03 by jordens

sbourdeauducq, 2016-05-03:
Yes, and the underflow is normal and expected (``pmt.count`` makes the CPU wait until the gate closes).