m-labs/artiq #180: Core device RPC should support dynamic typing

sbourdeauducq, 2015-11-24:
This should be valid code for the compiler:

```python
self.set_dataset("a", 1)
self.set_dataset("b", [1, 2])
```

Instead it results in a type unification error.

whitequark, 2015-11-24:
Well, there's not really any reasonable way to support this, so a workaround will be needed.

whitequark, 2015-11-24:
To elaborate, the best solution I can envision right now is something along the lines of...
```
self.set_dataset("a", dynamic([1, 2]))
```

sbourdeauducq, 2015-11-24:
If the compiler can itself insert ``dynamic`` automatically on all RPCs, LGTM.

whitequark, 2015-11-24:
No, it cannot, which is the exact issue here.

whitequark, 2015-11-24:
Specifically, the problem is that the compiler does not allow automatic coercion during argument passing. If it did, it would require mandatory annotations on every function and every variable, because then it would be impossible to infer the type of variable from its usage or, conversely, the type of formal argument from the passed formal arguments.

Since automatic coercion is not supported, the only option I see is manual coercion.

sbourdeauducq, 2015-11-24:
Can't you detect RPCs early, mark them as such and disable some of the type system for them?

whitequark, 2015-11-24:
Unfortunately not, because any function can be an RPC, even a deeply nested method, and unification-based inference is not directional: we may well infer the types of arguments before the type of callee, or vice versa.

There is a further problem; let's say I hypothetically add a `Dynamic` type for these. This type would behave just like any other during unification, which means it would propagate out from its usage, i.e. a variable passed to an RPC will gain this `Dynamic` type. But we cannot represent dynamically typed values at runtime because at runtime, types do not exist--we don't have an RTTI or equivalent facility! So it is not possible to actually have (allocate storage for, etc) a variable with the `Dynamic` type.

whitequark, 2015-11-24:
Ok, I think I know how to solve this. Every time a type is instantiated for an RPC function, every `Dynamic` in the signature is replaced with a fresh type variable. This way, we will have the following types:
```python
self.set_dataset("a", a:'a)
self.set_dataset("b", b:'b)
```

After that, once the types for `a` and `b` are know, this is refined to e.g.:
```python
self.set_dataset("a", a:str)
self.set_dataset("b", b:int(width=32))
``` 

The LLVM IR generator only really cares about argument types, and the RPC number itself is encoded in the RPC type, so this should not cause any problems during code generation.

jboulder, 2015-11-24:
This sounds cumbersome... Here's my understanding of the intended set of
types supported by set_dataset().

```python
from artiq import *
import numpy as np
import time

class SetDatasetTets(EnvExperiment):
    """Set Dataset Test"""
    def build(self):
        pass

    def run(self):

        tests = [
            True,
            [True, False],
            int(1234),
            [1, 2, 3]*3,
            float(1234),
            [1.1, 1.2, 1.3]*3,
            "asdf",
            ["asdf", "lkj"]*3,
            np.array(np.random.random(10), dtype=np.float32),
            np.array(np.random.random(10), dtype=np.float64),
            np.array(np.random.random(10), dtype=np.int32),
            np.array(np.random.random(10), dtype=np.int64),
            np.array(np.random.random(10), dtype=np.complex64),
            np.array(np.random.rand(10, 10), dtype=np.float32),
            np.array(np.random.rand(10, 10, 10), dtype=np.float32)
        ]

        for t in tests:
            self.set_dataset("test", t,
                              persist=False, save=False, broadcast=False)
            self.set_dataset("test", t,
                              persist=False, save=False, broadcast=True)
            self.set_dataset("test", t,
                              persist=False, save=True, broadcast=False)
            self.set_dataset("test", t,
                              persist=True, save=False, broadcast=False)
            self.set_dataset("test", t,
                              persist=True, save=True, broadcast=True)
```

whitequark, 2015-11-24:
@jboulder the precise set of types doesn't matter as long as it is larger than one.

jordens, 2015-11-25:
Alternatively, couldn't you do this `varargs`-style?

whitequark, 2015-11-25:
@jordens That's more complex than implementing polymorphism for methods.

commit 16ae0fb6 (2015-11-27): compiler.embedding: instantiate RPC method types (fixes #180).
closed 2015-11-27 by whitequark