m-labs/artiq #181: Reduce import of spurious dependencies/modules

sbourdeauducq, 2015-11-24:
With the new compiler, ``from artiq.protocols import pyon`` results in a lot of unnecessary modules (regex, levenshtein, ...) being pulled in from the compiler through ``artiq.language``. There are many cases where one would use pyon (or other ARTIQ library) without needing the compiler: controllers, GUI...

Loading those modules anyway increases memory consumption and dependency hell.

whitequark, 2015-11-24:
Is there any convenient way to see what modules are loaded?

sbourdeauducq, 2015-11-24:
```
>>> import sys
>>> a = set(sys.modules.keys())
>>> from artiq.protocols import pyon
>>> b = set(sys.modules.keys())
>>> b - a
```

whitequark, 2015-11-24:
Note that this does nothing to alleviate "dependency hell", although it does slightly reduce memory consumption and much more importantly startup times.

closed 2015-11-24 by whitequark

sbourdeauducq, 2015-11-24:
What I mean with alleviate dependency hell is things like: if you install ARTIQ manually to work on a specific part of it on a new machine, you don't have to install every dependency or touch conda.

commit e1cd2ccd (2015-11-24): compiler: pull in dependencies in more finely grained way (fixes #181).