m-labs/artiq #184: Master quits after ~1 hr without interaction

justboh, 2015-12-02:
I will start the master pc and GUI at the start of the day.  Then I leave it alone for ~1-4 hours typically.  I don't notice the GUI close, but when I come back to use the GUI again, the master and GUI have quit and I have to restart them.

Sorry I can't provide more details.

sbourdeauducq, 2015-12-03:
Any error log on the terminal where the master was running? Did you have a master log file and what are its last entries? Can you reproduce the crash (I've run the master for days here without problem) or was it a one-off issue?

sbourdeauducq, 2015-12-03:
And were you running something in particular? It also seems surprising that both the master and GUI quit at the same time. Are you certain your computer was not simply rebooted?

sbourdeauducq, 2015-12-05:
Without more details, we cannot do anything about this. If the problem occurs again, please pay attention to what exactly happened and submit a useful report.

closed 2015-12-05 by sbourdeauducq