m-labs/artiq #185: remove flterm from conda packages

sbourdeauducq, 2015-12-03:
The C flterm is currently installed as a ARTIQ dependency. We should switch to the Python version, which should be part of misoc and not automatically installed with ARTIQ.

I am not sure about the quality of the Python flterm though, and we may want to postpone the removal of the C flterm from the packages until the Python version uses asyncio.

jboulder, 2015-12-03:
>The C flterm is currently installed as a ARTIQ dependency. We should
>switch to the Python version, which should be part of misoc and not
>automatically installed with ARTIQ.

flterm is the debug tool of last resort. And it permits bare-bones
operation of the soc. For me it's be a reliable, helpful tool. It
should be part of ARTIQ's conda distribution.

>I am not sure about the quality of the Python flterm though, and
>we may want to postpone the removal of the C flterm from the
>packages until the Python version uses asyncio.

The C version works well. What's the motivation to change to a new,
potentially unstable version?

sbourdeauducq, 2015-12-04:
It would still be available, but with the misoc package and not automatically installed with "conda install artiq".

Python is generally a better language for this sort of non-performance-sensitive software, the Python version is portable and allows the misoc package to be noarch, and installing a Python script is easier than installing compiled software when you are using setuptools. Maybe @enjoy-digital who wrote the Python flterm can offer other arguments.

enjoy-digital, 2015-12-04:
The main argument for writing it in Python was for portability.

As @sbourdeauducq said, it could be improved in the future to use asyncio but I'm using it regularly without problems.

dhslichter, 2016-01-05:
The presence of flterm as a dependency in artiq currently breaks the conda install on Windows, since flterm will not install on Windows.  Can we have a version of the Win32 conda package that removes the dependency on flterm (recognizing that this will break some current hardware debug capabilities), to enable further artiq testing on Windows using conda installation?  

In general, I support the notion of moving flterm's functionalities to python for portability.

whitequark, 2016-01-05:
flterm was already rewritten in Python (and included in the misoc conda package), so this should be a simple matter of changing dependencies.

sbourdeauducq, 2016-01-05:
Done, python version actually works fine.

closed 2016-01-05 by sbourdeauducq