m-labs/artiq #186: standardized subprocess communication

sbourdeauducq, 2015-12-03:
* To be used in master/worker, controller manager/controller, GUI/display process
* stdout and/or stderr is used for logging
* Logging messages are output in human-readable format and parsed, with the following format: severity:name:message. Severity is one of DEBUG, INFO, WARNING, ERROR, CRITICAL.
* Messages that do not fit this format are treated as "prints"
* One log entry per line. At a later stage, we should support multi-line log entries, e.g. by adding a line count to the first entry, or using escape characters.
* Optional IPC channels are implemented using anonymous pipes. 
* The subprocess library should support a mechanism for the parent process to request IPC channels, pass the required information to the child process (e.g. file descriptors as command line arguments) and for the child process to get access to the other ends of the pipes.
* The parent process side is based on asyncio and this requires implementing proper pipe support for asyncio (it is currently limited to stdout/stdin/stderr).

jordens, 2015-12-16:
If there are multiple channels, the logging should get its own channel and not be mixed with the regular stdout stuff. But using stderr exclusively for logging would be fine by me as it makes those processes  that don't need extra pipes easy to debug. Then again most processes will need an extra pair just for basic request-reply: that will clash with print() if req-rep doesn't get its own pipe pair.

whitequark, 2015-12-17:
We've already decided to do this via TCP over localhost, as this is the most portable and least intrusive IPC method.

jordens, 2015-12-17:
Pretty intrusive as it kill interaction and debugging through stdio, right? Details please. Is this muxing the channels over one TCP connection? Anyway, what I said still applies mutatis mutandis.

whitequark, 2015-12-17:
No, no muxing, several TCP connections. You could use wireshark for debugging, which would be vastly more convenient than with stdio.

As for interaction, you could even use netcat.

sbourdeauducq, 2016-01-20:
I don't like TCP for IPC. Keeping the connections closely coupled to their processes as pipes do is better for e.g. handling graceful subprocess termination.

Keeping the logging on stdin/stderr makes processes that don't use pipes (e.g. controllers) easier to debug and runnable standalone.

dhslichter, 2016-01-20:
Does the use of pipes complicate Windows compatibility vs using TCP?

sbourdeauducq, 2016-01-20:
Yes. Python does not officially support pipes on Windows with asyncio, so there will be some hacks involved.

sbourdeauducq, 2016-01-27:
Done and works on Windows, still needs to convert applets to it.

closed 2016-02-09 by sbourdeauducq