m-labs/artiq #187: Qt fails to properly lay out log table view when long words are present

sbourdeauducq, 2015-12-06:
This can be demonstrated by the following code:
```python
import logging

from artiq import *

logger = logging.getLogger("qtfail")

class QtFail(EnvExperiment):
    def build(self):
        pass

    def run(self):
        logger.warning("_".join(str(i) for i in range(200)))
```

Here is the result. Note that WARNING, the source and the date are also cut.
[image]

sbourdeauducq, 2016-01-15:
I just saw the columns break again, after your patch. (Qt4 anaconda)

whitequark, 2016-01-15:
That one does look like a layout bug, yes.

jboulder, 2016-01-16:
Column break bug with the latest conda version.

[image]

whitequark, 2016-01-16:
@jboulder can you try the new conda package?

sbourdeauducq, 2016-01-16:
:+1: Works here. We'll probably want to get Qt properly fixed and remove all those workarounds at some point...

sbourdeauducq, 2016-01-27:
Columns are still breaking.
[image]

commit 00b688ca (2016-01-14): gui: wrap very long words too (fixes #187).
commit bebd89c9 (2016-03-25): gui: redesign table/trees to avoid slow and buggy qt/pyqt autosize. Closes #182. Closes #187.
commit 5d5a4433 (2016-03-25): gui: redesign table/trees to avoid slow and buggy qt/pyqt autosize. Closes #182. Closes #187.
closed 2016-03-25 by sbourdeauducq