m-labs/artiq #189: Make sure conda packages for gateware don't depend on main package

whitequark, 2015-12-16:
There's no reason they should, and this avoids dependencies. Currently the build fails because gateware requires numpy (?)

sbourdeauducq, 2015-12-16:
Note that installation instructions assume that they do.

jordens, 2015-12-16:
What does "avoids dependencies" mean? Build dependencies or runtime dependencies? The bitstreams are of little use without artiq (only in some hypothetical "comissioning" installation but even that would usually want to run tests).
We should probably not try to have artiq runtime-depend on bitstreams. "All bitstreams" for mere installation convenience would be problematic because of failing builds even if the bitstreams don't use much space. "Any bitstream" will likely break with conda. The reverse should be done however. Each bitstream should depend on the matching artiq version (if that does not kill conda's dependency resolver) and it should be obvious that a bitstream needs to be installed when installing artiq.

whitequark, 2015-12-18:
Build dependencies. Runtime dependencies are already handled as you say.

sbourdeauducq, 2015-12-19:
The gateware "requires" numpy because importing things in ``artiq.gateware`` executes ``artiq/__init__.py``, which in turns imports things that require numpy. This is the same problem that I mentioned in #181.

whitequark, 2016-01-17:
I've revisited this to triage. The crux of the issue is:
  * `artiq/__init__.py` does `from artiq.language import *`
  * `artiq/language/__init__.py` does `from artiq.language.environment import *`
  * `artiq/language/environment.py` does `from artiq.protocols import pyon` for `PYONValue`
  * `artiq/protocols/pyon.py` needs conversions from/to numpy.

I'm not sure where to break the chain, or if it makes sense at all. Do we still want to address this? If yes how?

jordens, 2016-01-17:
We might want to have experiments do `from artiq.language import *` and not have `artiq/__init__.py` do that. `artiq` encompasses so much more than the language.

sbourdeauducq, 2016-01-25:
Agreed. But this breaks a user API (even if the change is simple), so we should probably do that in version 2.0 and keep version 1.0 compatible with the current code.

jordens, 2016-01-25:
I would argue the opposite. This is a tightly contained change where we know very well what the fallout is. Regressions or follow-up changes are unlikely. The earlier we push this change, the less work it will be.

commit 5444cd3c (2016-01-26): Merge branch 'namespace-experiment'

closes #189

* namespace-experiment:
  artiq_dir: move out of tools to unlink dependencies
  refactor Analyzer constants to unlink dependencies
  CoreException: store at 'py_exn.artiq_core_exception'
  doc: update Underflow catching example
  coredevice: remove some print()s
  language...ARTIQException -> coredevice...CoreException
  artiq.experiment: update examples
  artiq.experiment: merge language and coredevice namespaces
  artiq: move namespace artiq.* -> artiq.language.*
closed 2016-01-26 by jordens