m-labs/artiq #193: Following data dependencies while unrolling loops

whitequark, 2015-12-17:
A feature has been requested (e.g. in https://github.com/m-labs/artiq/issues/42) to unroll loops where the trip count or I/O delay caused by the loop body are dependent on variables outside of the loop. Such as in:
```python
@kernel
def run(self):
    for t in range(10):        
        t2 = t*100
        with parallel:
            self.ttl0.pulse(t2)
```

There are several language design issues with this proposal. To recap, my goal as a language designer is to create a system that is easy to reason about, and easy to form a mental model of. Specifically, someone implementing a process in ARTIQ must be able to follow a small set of straightforward rules that tell whether a given ARTIQ Python construct is legal (accepted by compiler) or not. Conversely, when the ARTIQ compiler encounters an illegal construct, it has to explain to the programmer why the entire class of these constructs is illegal.

To handle the code in the example above 'as one could expect', the compiler has to rely on many assumptions:
  * It must be able to look through assignments and variable references. This is not in general possible, e.g. when closures or control flow is involved.
  * It must be able to look through expressions. This is not in general possible, e.g. when functions are involved.
  * It must be able to unroll loops. This is not in general possible, e.g. when control flow is involved.
  * It must be able to look outside `with parallel:` blocks.

Right now, I believe we have the aforementioned small set of straightforward rules for the interleaving transformation:
  * the only recognized expressions are binary expressions with `+`, `-`, `*`, `/` and `//`, and arguments which are never assigned to;
  * the only recognized statements are statements without I/O delay, or statements without control flow, or loops iterating over an expression which is syntactically a `range(...)` with the arguments specified above;
  * all of these restrictions are only applied within a `with parallel:` block.

It is easy to design code so that it is conforming, in particular, you do not have to guess (by running the compiler) whether given code is valid. The rules above also allow the compiler to express the I/O delay caused by a function as an algebraic expression whose free variables only rely on the function arguments, which means that, no matter how deep the `with parallel:` blocks are composed, it is never necessary to look at more than one function at once.

If we begin handling assignments, things become complicated. If there's only one assignment to a given variable in the entire function, sure, it's just an alias. If there are two consecutive assignments, we can also handle this. What if there's control flow? Arguably `if True:` is as good as no control flow at all. Is `if 1 == 1:`? Is `if 1 == x` when the value of x is known? What if x is assigned in a loop? etc. Handling control flow would make it impossible to express the I/O delay incurred by a function as an algebraic expression, which means that we have to look arbitrarily deep down the call stack to understand an error.

If, apart from the above, we allow `with parallel:` blocks to affect code outside them, things become even more complicated, now we may have to look arbitrarily deep *up* call stack to understand an error.

In other words, the specific set of rules for abstract evaluation of code is complex, requires considering a potentially unlimited scope, prone to change when some corner case has to be addressed, and will inevitably contain omissions simply because some particular cases are not handled due to lack of time to implement or perceived unimportance.

As a result, to answer to the question "will this code be accepted by the compiler?", the programmer essentially has to write the code and try to compile it. Moreover, when it is inevitably rejected, there is no easy way to know what parts to refactor to make it acceptable, and no easy way to display a sensible diagnostic. Consider the case if `if 1 == x:`: a purported diagnostic would have to go something like this:
  * Cannot interleave the code, because...
  * The code has control flow, because...
  * We cannot prove that `1 == x:` always has the same value, because...
  * We cannot prove that `x` always has the same value, because...
  * (for example) `x` could have been assigned a value in a branch we do not always take, because...
  * repeat all of the above for that branch.
  * (another example) `x` is assigned a value from an attribute, which is out of scope for our analysis.

With inlining, the diagnostic would gain context from an unlimited number of functions, which are implicitly called with some arguments that are not immediately visible from the source (because they come from abstractly evaluated expressions), including several invocations of the same function.

While it is technically possible, if somewhat laborous, to implement this scheme as I describe it above, I do not believe that it comprises good language design or makes writing complex programs in ARTIQ Python easier.

To concentrate on the example above again, what's wrong with reformulating it as...
```python
@kernel
def run(self):
    with parallel:
        for t in range(10):
            self.ttl0.pulse(t*100)
```
?

I am assuming that the idea behind allowing that example is to allow `with parallel:` statements to be embedded in some more complex code. The point I am making is that naive implementation of such embedding, while possible, is not actually desirable, and whatever goals it might be trying to achieve, are worth achieving by different means.

I am now asking to clarify what those goals are, so that I can implement them in a way that does not make writing ARTIQ Python programs meaninglessly painful.

jordens, 2015-12-17:
Is that `with parallel:` in the original example at the correct scope? The way it's written I don't understand its intent. A `parallel` block containing a single (itself internally sequential) statement should be the same as within a `sequential` block, no?

Help me understand the two restrictions for parallel blocks that you mention a bit better:

> * the only recognized expressions are binary expressions with +, -, *, / and //, and arguments which are never assigned to;
> * the only recognized statements are statements without I/O delay, or [...]

"This only applies in `parallel`": in the parallel time flow scope or in the `with parallel:` program control flow scope? More specifically, `pulse()` is `sequential` internally. `pulse()` furthermore contains a delay which seems to contradict the second requirement.

I don't understand the `statements without I/O delay`. Without delays inside `parallel` this is seems to be entirely void.

I suspect I roughly get where the data dependency problem comes from. But, complicated expressions will likely arise and being able to name them is something very useful. If we can't have variables, can we get something that can solve a similar problem, something like "macros" or "named expressions"?

I'll try to come up with a few more use cases, but I think a double parallel pulse with latency compensation might already be tricky:

```python

def pulse(self, t):
    delay_mu(-self.latency_rise)
    self.on()
    delay_mu(t + self.latency_rise - self.latency_fall)
    self.off()
    delay_mu(self.latency_fall)

@kernel
def run(self):
    with parallel:
        for i in range(self.repeats):
            self.ttl0.pulse(i*self.t0)
        for i in range(self.repeats):
            self.ttl1.pulse(i*self.t1)
```

By the way, I think it might be useful to clarify again to what depth a `parallel` block interleaves things. In the example above I would consider the loop bodies to be each sequential and the interleaving to only happen on the "top level" in parallel context.

whitequark, 2015-12-18:
> Is that with parallel: in the original example at the correct scope? The way it's written I don't understand its intent. A parallel block containing a single (itself internally sequential) statement should be the same as within a sequential block, no?

Correct. I unintentionally made the example confusing; let's instead look at the following one:

```python
@kernel
def run(self):
    for t in range(10):        
        t2 = t*100
        with parallel:
            self.ttl0.pulse(t2)
            self.ttl1.pulse(t2)
```

(Since currently `with parallel:` blocks with only one statement inside are not treated specially, everything I wrote applies equally to both examples right now.)

> "This only applies in parallel": in the parallel time flow scope or in the with parallel: program control flow scope? More specifically, pulse() is sequential internally. pulse() furthermore contains a delay which seems to contradict the second requirement.

In the `with parallel:` control flow scope, which encompasses any inlined functions.

> I don't understand the statements without I/O delay. Without delays inside parallel this is seems to be entirely void.

For example, this code is valid, despite including complex control flow inside `with parallel:`:

```python
@kernel
def run(self):
    with parallel:
        with sequential:
            self.ttl1.pulse(0.5)
            for x in range(random.randint(10)):
                if x > 5:
                    break
                print(x)
        self.ttl0.pulse(1)
```

The trick is that the interleaving code doesn't care about entire statements no matter how complex if there are no delays inside them. You couldn't replace `break` with `return` though.

> But, complicated expressions will likely arise and being able to name them is something very useful. 

I entirely agree. By placing explicit restrictions on the semantics of language constructs, it would be possible to keep analysis tractable for both programmers and the compiler.

> If we can't have variables, can we get something that can solve a similar problem, something like "macros" or "named expressions"?

Yes. The only issue I see is the syntax. I can see something like this being accepted...

```python
@kernel
def foo(x):
    const_L = x * 10
    delay(const_L)
```

with the implication that any variable named `const_*` can only be assigned once. But I think this is a bad choice for syntax, not to mention unenforceable when running within host Python.

> def pulse(self, t):
>     delay_mu(-self.latency_rise)

Note that this construct is not considered valid because there is no guarantee that `self.latency_rise` is not mutated at runtime and therefore it cannot be used to guide interleaving.

> By the way, I think it might be useful to clarify again to what depth a parallel block interleaves things. In the example above I would consider the loop bodies to be each sequential and the interleaving to only happen on the "top level" in parallel context.

Correct. Not only loop bodies but the entire loops themselves are laid out sequentially in the timeline.

jordens, 2015-12-18:
> The trick is that the interleaving code doesn't care about entire statements no matter how complex if there are no delays or nonlocal control flow inside them. You couldn't replace `break` with `return` though. A rule of thumb is, if you can extract a statement into a closure that is immediately invoked and incurs no I/O delay, it can be used in a `with parallel:` block.

And are you saying this ability can stay? Or is it a challenge?

Is it the potential side-effects of these statements that are troublesome?

> But, complicated expressions will likely arise and being able to name them is something very useful. 

I entirely agree. By placing explicit restrictions on the semantics of language constructs, it would be possible to keep analysis tractable for both programmers and the compiler.

> with the implication that any variable named `const_*` can only be assigned once. But I think this is a bad choice for syntax, not to mention unenforceable when running within host Python.

I was thinking of something like `l = freeze(x*10)` being a special (frozen) expression type or object.

> Note that this construct is not considered valid because there is no guarantee that `self.latency_rise` is not mutated at runtime and therefore it cannot be used to guide interleaving.

`freeze(-self.latency_rise)`?

Or would implicitly making all variables (except the special handling of `now`) readonly within parallel be sufficient? The examples that I can come up with right now would work fine with that.

whitequark, 2015-12-18:
> And are you saying this ability can stay? Or is it a challenge?

It stays. It was explicitly designed in.

> Is it the potential side-effects of these statements that are troublesome?

The only side effects that matter for interleaving is control flow and I/O delay. As long as a statement has no I/O delay and no control flow outside that statement, there is no trouble interleaving it.

> I was thinking of something like l = freeze(x*10) being a special (frozen) expression type or object.

That's not really a solution because what we want to rely on is that the *binding* (i.e. variable itself, not what it points to) doesn't change. That is, the number `1` is already frozen but `a` in `a = 1` is not.

> freeze(-self.latency_rise)?
> 
> Or would implicitly making all variables (except the special handling of now) readonly within parallel be sufficient? The examples that I can come up with right now would work fine with that.

(There is no more `now` global variable on the Python level, it was an implementation detail.)

It is not possible to make them readonly within the `with parallel:` block because of aliasing: it is possible to mutate local variables from a closure defined outside a `with parallel:` block and invoked inside a `with parallel:` block, however indirectly, as long as it does not advance timeline. And of course mutation of fields is not lexically restricted at all.

But that's not all, to interleave a timeline which depends on the value of `self.latency_rise`, the interleaver has to somehow guess what *one* object `self` will refer to at runtime and what the value of `self.latency_rise` would be, which is not a problem that can be reasonably solved.

whitequark, 2015-12-18:
> I was thinking of something like l = freeze(x*10) being a special (frozen) expression type or object.

Oh, in addition this adds a lot of boilerplate for the most common case, arguments.

jordens, 2015-12-18:
> The only side effects that matter for interleaving is control flow and I/O delay. As long as a statement has no I/O delay and no control flow outside that statement, there is no trouble interleaving it.

You have to clarify this. Every top-level statement that one would ever want to put inside `parallel`, uses `delay()`. Otherwise it gets terribly boring. `parallel` is all about parallelizing the `delay()s`.

> > Or would implicitly making all variables (except the special handling of now) readonly within parallel be sufficient? The examples that I can come up with right now would work fine with that.

> (There is no more `now` global variable on the Python level, it was an implementation detail.)

Except for aliasing, does it matter much whether it is a variable with accessor functions or an exposed variable?

> But that's not all, to interleave a timeline which depends on the value of `self.latency_rise`, the interleaver has to somehow guess what *one* object `self` will refer to at runtime and what the value of `self.latency_rise` would be, which is not a problem that can be reasonably solved.

Can't you demand that it must have been const-folded before interleaving?

Variables are important to me. Lets try to introduce another tool for a somewhat simpler case of parallelizing things:
Maybe we can use the above notion that `parallel` is all about parallelizing `delays` to reshape this entire thing. What if we give every statement at the top level within a `parallel` block (the scope where the timeline is parallelized so that delays in one statement do not affect the delays in another) its own `now`: After every statement in the block you would memoize the incremental delay it caused, then undo that delay, execute the next statement. And on closing the parallel block you affect the maximum of all delays caused. This is a bit similar to the dds batch. If you will run into time underruns earlier but to me this seems to be an acceptable trade-off.
If we can't figure out how to allow something like named expressions, I would probably want this interleave-in-rtio-fifo context in addition to the variable-free interleave-at-compile-time context.
Comments?

whitequark, 2015-12-18:
> You have to clarify this. Every top-level statement that one would ever want to put inside parallel, uses delay(). Otherwise it gets terribly boring. parallel is all about parallelizing the delay()s.

Sorry, let me be more precise. Inside `with parallel:`, the following statements are valid:
  * Expression, assignment statements. These can contain calls to functions that incur I/O delay.
  * `with sequential:` statements. The body is covered by these requirements recursively.
  * `with parallel:` statements.
  * `for` loops with a body that incurs an I/O delay and is covered by these requirements recursively. These  for loops must iterate over a range expression whose bounds are known.
  * `raise` statements.
  * Any other statements, provided they do not incur an I/O delay or transfer control in a way that other statements incurring I/O delay within this `with parallel:` block are not executed. E.g. `while cond: break` is fine, but `return` is not.

> Except for aliasing, does it matter much whether it is a variable with accessor functions or an exposed variable?

No. There is no deep meaning to this change, I am just mentioning that it has been done.

> What if we give every statement at the top level within a parallel block (the scope where the timeline is parallelized so that delays in one statement do not affect the delays in another) its own [...]

Unfortunately, this will not work with the current gateware design because the RTIO core uses a queue rather than associative memory and thus requires all requests to be submitted with a timestamp in nondescending order.

> I would probably want this interleave-in-rtio-fifo context in addition to the variable-free interleave-at-compile-time context.

We anticipate that a combination of techniques will be required. For one, associative memory in gateware is limited in size and the limit is relatively harsh, so some degree of interleaving will probably be always necessary in the compiler. Conversely, having this associative memory allows us to parallelize some statements that no amount of static analysis, no matter how advanced or hard to grasp, will support.

My prior experience writing compiler transformations based on static analysis suggests that the result will tend to exhibit wildly unpredictable behavior in corner cases (i.e. seemingly small changes will cause large amounts of code up or down the stack to gain or lose the potential for being statically interleaved) and be hard to debug, which is why I have so far been pushing for introducing restrictions for the language semantics that map most closely to the problem domain (i.e. implementing algorithms controlling an RTIO core with given hardware restrictions in a high-level way). It is unfortunate, but probably unavoidable, that the choice of a DSL that is a strict subset of Python is not very good at expressing code in this problem domain.

Adding some way of marking variable bindings as constant will reach some of your goals: this will allow to introduce names for expressions whose value is already statically known, as well as make the current usage of arguments sound (https://github.com/m-labs/artiq/issues/192). It will not, however, provide a construct analogous to `self.latency_rise`, and while I see some ways to implement that one, I'm not sure if any of them are good.

What is the anticipated usage of `self.latency_rise`? I.e. is it akin to a global constant, same for all instances of the type of `self` during a kernel run? Is it different for every object? Would `self` be always statically known in the caller? Etc.

whitequark, 2015-12-18:
> Can't you demand that it must have been const-folded before interleaving?

The issues with abstract interpretation (constant folding by itself is not generally powerful enough, you have to do a combination of DCE, SCCP, inlining, loop unrolling and tree-shaking at least) is unwieldy, if at all meaningful, diagnostics and fragile code, where changing some parts of it affects seemingly unrelated ones. I've elaborated on this elsewhere in this issue already.

jordens, 2015-12-18:
>  * `with sequential:` statements. The body is covered by these requirements recursively.

There is a difference between a function call and a sequential block even if they contain the exact same statements, just because of the different closures, the later closure being somewhat more "delicate"?

>  * `with parallel:` statements.

That would be a NOP, right?

> What if we give every statement at the top level within a parallel block (the scope where the timeline is parallelized so that delays in one statement do not affect the delays in another) its own [...]

> Unfortunately, this will not work with the current gateware design because the RTIO core uses a queue rather than associative memory and thus requires all requests to be submitted with a timestamp in nondescending order.

That's going to be just fine. Usage of `parallel` context is dominated by cases where actions on different channels are interleaved. The channels have independent FIFOs. We only have to be monotonic within a channel. DDS are a bit special here, but the consequences seem just fine.

> We anticipate that a combination of techniques will be required. For one, associative memory in gateware is limited in size and the limit is relatively harsh, so some degree of interleaving will [...]

I know little about associative memories. But I suspect that a small sorting tree could do this as well. Conceptually that might end up being very similar to an AM implementation...

> My prior experience writing compiler transformations based on static analysis suggests that the result will tend to exhibit wildly unpredictable behavior in corner cases (i.e. seemingly small [...]

Threading seems to be the answer people usually arrive at.

> What is the anticipated usage of `self.latency_rise`? I.e. is it akin to a global constant, same for all instances of the type of `self` during a kernel run? Is it different for every object? Would `self` be always statically known in the caller? Etc.

If these are not rhetorical questions: these things are globally constant but channel specific. And they do not need to be visible outside self.

whitequark, 2015-12-19:
> with sequential: statements. The body is covered by these requirements recursively.
> There is a difference between a function call and a sequential block even if they contain the exact same statements, just because of the different closures, the later closure being somewhat more "delicate"?

No, there is no difference between requirements that are placed by `with parallel:` for a sequential block and a body of an inlined function. However, to be inlined, the function has to be statically known.

> with parallel: statements.
> That would be a NOP, right?

No, e.g. consider this case:
```python
with parallel:
    ttl1.pulse(10)
    with sequential:
        ttl2.pulse(5)
        with parallel:
            ttl3.pulse(5)
            ttl4.pulse(5)
```
This cannot be reformulated with only one `with parallel:` block without adding explicit `delay` calls.

> That's going to be just fine. Usage of parallel context is dominated by cases where actions on different channels are interleaved. The channels have independent FIFOs. We only have to be monotonic within a channel. DDS are a bit special here, but the consequences seem just fine.

That is not the case. To make the invariant hold for the output of the interleaver transformation, now the interleaver must statically guess the channel numbers as well as track the position on the timeline. While right now you can use any value (e.g. `self.channel`) to get the channel number, this would not be the case if we added channel numbers to the analysis.

> Threading seems to be the answer people usually arrive at.

I'm afraid that thread-based parallelism in ARTIQ is a nonstarter. Python is a language based on shared mutable state. It ensures memory safety by using GIL. (Note that I am not yet talking about correctness; no matter how racy a program in CPython would be, it will never segfault.) Jython uses complex fine-grained locking. PyPy uses (or at least plans to; last time I looked it was not memory-safe in presence of data races) software transactional memory. Here's the thing: none of these techniques are compatible with realtime guarantees, as they all can and will block for unbounded amounts of time. Moreover, all of those Python implementations rely on a shared heap, which means an allocator and a GC, and a realtime multicore GC is not a project we have the expertise, capacity or time to implement. See also http://adriansampson.net/blog/parallelpypy.html and http://morepypy.blogspot.it/2012/08/multicore-programming-in-pypy-and.html for some details on Python and memory models.

Now in case anyone doubts the necessity of memory safety: first, it is a prerequisite for correctness in light of parallelism. Second, how are you going to debug segfaults arising from race conditions on the core device, and teach this to the people who expect to write a dialect of Python?

There are some other models of parallelism, e.g. fork-join parallelism, that are more tractable. However, they still require a nontrivial amount of compiler research and engineering to be usable--I anticipate that linear types and a borrow checker, much like what Rust uses, are a prerequisite, and this involves major language changes as well as a higher barrier to learning.

Finally, it might be possible to add even more restrictions to the semantics of code within `with parallel:` to make it automatically parallelizable across several CPUs, but that's also a research project and I am not yet convinced it can deliver the features you want.

> If these are not rhetorical questions

No; I have much less insight into how ARTIQ Python is actually used than I wish I had.

> these things are globally constant but channel specific. And they do not need to be visible outside self.

"Not visible outside self" still means a global analysis on every function which accepts this type of `self`, which means code will break when unrelated code changes, so that restriction is not much help.

The crux of the issue here is that we have to inline every invocation of a function that uses `self.latency_rise`, and we have to ensure that once we inline it, `self` we get is a known constant, and we have to do it in a tractable way. I will try to do something about this.

jordens, 2015-12-19:
> > with parallel: statements.
> > That would be a NOP, right?

> No, e.g. consider this case:
> This cannot be reformulated with only one `with parallel:` block without adding explicit `delay` calls.

`with parallel: with parallel:` is redundant. Deeper parallels are already covered by the recursive application of the rules.

> > That's going to be just fine. Usage of parallel context is dominated by cases where actions on different channels are interleaved. The channels have independent FIFOs. We only have to be monotonic within a channel. DDS are a bit special here, but the consequences seem just fine.

> That is not the case. To make the invariant hold for the output of the interleaver transformation, now the interleaver must statically guess the channel numbers as well as track the position on the [...]

No. You don't have to handle that in the compiler. Assume that affected statements in this flavor of parallel context only access mutually non-intersecting sets of channels, don't prove it. We generally only expose underruns (decreasing time events) at runtime and expect to have excellent tools to debug them.

whitequark, 2015-12-19:
> Assume that affected statements in this flavor of parallel context only access mutually non-intersecting sets of channels, don't prove it.

I'm not sure I correctly understand the model you are suggesting. Are you saying that all statements immediately inside `with parallel:` can be assumed to access disjoint sets of channels with respect to each other? In that case yes, the only thing `with parallel:` would do is to save/restore the timeline position between the statements in its body.

jordens, 2015-12-19:
```python
class Synchronous:  # or Simultaneous
    def __enter__(self):
        self.t_enter = now_mu()
        self.t_exit = []

    def rollback(self):  # or realign() or restart()
        self.t_exit.append(now_mu())
        at_mu(self.t_enter)

    def __exit__(self):
        at_mu(max(self.t_exit))

@kernel
def run():
    with Synchronous() as sync:
        ttl0.pulse(1)
        sync.rollback()
        ttl1.pulse(2)
        sync.rollback()
        ttl3.pulse(3)
```

Now the `rollback()` can become implicit and the `Synchronous()` can become `synchronous` and both be handled by the compiler. But that might not even need to be done. If we determine that the old `parallel` is either too difficult to get right or useless if appropriately constrained, both `sequential` and `parallel` can go. Then having such a transaction style parallelism with explicit rollback is pretty sweet. 
Subtransactions are free. And maybe this design aligns even further with databases (sessions, transaction managers etc).

The thing about the original `parallel` design is that resource conflicts and assumptions about state need to be handled manually anyway: if one thread sets a ttl high (for a pulse) it tends to expect it to stay high.

sbourdeauducq, 2016-01-19:
I like ``with parallel`` as it is more abstract and allows for different implementations or a combination of them:
1. the dumb implementation, where you store the time of entry into the ``parallel`` block, make it sequential and call ``at_mu`` before each of its branches. It is similar to the ``Synchronous`` above. Problems: each branch must use a different set of channels (but this is OK), may cause more underflows if e.g. a branch fills and blocks on the FIFO with events far in the future while another branch has shorter-term events to schedule.
2. the interleaving implementation, where a difficulty is determining what is constant at compile time. My compiler ran passes of constant propagation/loop unrolling/dead code elimination before interleaving, so that it handled nicely a number of practical cases. But this makes it difficult to guess in advance whether a particular piece of code will compile or not.
3. the coroutine implementation, which is difficult to implement without an unacceptable performance hit from coroutine switching and especially scheduling. The latter will certainly need gateware assistance.

For (2) if we cannot find a nice syntax for marking constants and if my solution is deemed unacceptable,  then (1) is probably the way to go.

whitequark, 2016-02-16:
I will proceed with the following syntax:
```
with const(expra, exprb, ...) as x, y...:
    ... delay(x)...
    self.ttl0.pulse(y * 100)
```

Formally speaking, the `with const` statement binds an identifier `x` to the result of the evaluation of `expra` within the body of the `with` statement. Any assignment to this specific instance of `x` bound by the `with const` statement is forbidden within it, although of course code can shadow `x` e.g. using a closure or a comprehension and assign to that, pertaining to Python's usual scoping rules.

In effect, `const` is a no-op and can be implemented as `def const(*args): return args`; the magic here is in the guarantees that untie the hands of the interleaver.

This will be enough to close #192 and #193, but not #236.

whitequark, 2016-02-16:
Actually, no, that's not strict enough to handle #192, e.g. because of
```
def foo(a):
  a = 3
  with const(a) as a:
    delay(a)
with parallel:
  foo(1)
  delay(2)
```
where `foo` will be executed after `delay(2)`. 

A stronger invariant is needed. Like I mentioned, what I actually want is bindings that are immutable right away when they're defined, but that's impossible to express in a sane way by abusing Python syntax... I will try to invent something.

sbourdeauducq, 2016-02-17:
Problem is acceptable for 1.0 with ``with interleave``.

sbourdeauducq, 2021-12-13:
``with interleave`` is abandoned, and NAC3 does explicit unroll (https://git.m-labs.hk/M-Labs/nac3/issues/63).