m-labs/artiq #198: round(x:float) doesn't work

whitequark, 2015-12-18:
Apparently we don't have round() on the core device, and this is what `@llvm.round.f64` intrinsic lowers to. @sbourdeauducq, should we find some variant of, and link libm?

sbourdeauducq, 2015-12-19:
Sounds overly complicated and then there will be a dangerous incentive to implement ``math.cos`` & friends. Just implement that round function somewhere.

jordens, 2015-12-19:
This is useful. Not as intrinsics but as part of the runtime API. Can't `libm` become part of the runtime?

whitequark, 2015-12-19:
@sbourdeauducq The implementation of `round` is simple-ish (https://github.com/JuliaLang/openlibm/blob/0036d707347b27c580bf25c28cfc43dcbe412437/src/s_round.c) but it depends on floor, which isn't... (https://github.com/JuliaLang/openlibm/blob/0036d707347b27c580bf25c28cfc43dcbe412437/src/s_floor.c). I'm not sure.

sbourdeauducq, 2015-12-19:
Why was that not a problem with my compiler? Note that Python's ``round`` returns an integer, not a float.

whitequark, 2015-12-19:
@sbourdeauducq Ok. By the way, what should be the behavior of round(1e100)? C makes it unspecified and also may set errno, and naturally Python doesn't have this problem.

sbourdeauducq, 2015-12-19:
Should be similar behavior as integer overflows. And I propose that ``round`` be 32-bit by default and take a width parameter like ``int``.

whitequark, 2015-12-19:
Do you mean wraparound? That is bound to produce some very obscure bugs...

sbourdeauducq, 2015-12-19:
Wraparound or undefined. I don't think the wrapping around of ints should be relied on by programs anyway, as this is not what Python does.

whitequark, 2015-12-19:
How about raising an exception on overflow? That would actually bring us back to the "strict subset of Python" semantics rather than the current "strict subset of Python except for integer literals"

sbourdeauducq, 2015-12-19:
I think performance of integer operations is more important (I know that mor1kx has hardware overflow detection hooked to an exception vector, but this is non-portable and somewhat complicated to use).

whitequark, 2015-12-19:
We can do it like Rust does: overflow in "release" mode, exception in "debug" mode.

whitequark, 2015-12-19:
Also, is it really complicated to use? All exception vectors are already hooked up to produce ARTIQ Python exceptions. Nonportable, yes, I agree.

commit d5216879 (2015-12-19): runtime: implement round() (fixes #198).

From https://github.com/bpowers/musl/blob/master/src/math/round.c.
closed 2015-12-19 by whitequark