m-labs/artiq #201: compiler assertion error in interleave

sbourdeauducq, 2015-12-21:
```python

from artiq import *

class Test(EnvExperiment):
	def build(self):
		self.setattr_device("core")
		self.setattr_device("ttl0")
		self.setattr_device("ttl1")
		self.setattr_device("ttl2")
		self.setattr_device("ttl3")

	@kernel
	def run(self):
		for i in range(10):
			with parallel:
				self.ttl0.pulse(3*ms)
				with sequential:
					self.ttl1.pulse(1*ms)
					self.ttl2.pulse(1*ms)
			self.ttl3.pulse(1.5*ms)

```

```
  File "/home/sb/anaconda3/lib/python3.5/site-packages/artiq/compiler/transforms/interleaver.py", line 36, in iodelay_of_block
    assert iodelay.is_const(folded_expr)
AssertionError
```

sbourdeauducq, 2015-12-21:
Still breaks when removing the for loop.

whitequark, 2015-12-21:
Crux of the issue is that `duration` after inlining is not constant unless a constant literal was used... I designed this in a wrong way. But this should be easy enough to fix

commit 082e9e20 (2015-12-25): compiler: do not associate SSA values with iodelay even when inlining.

Fixes #201.
closed 2015-12-25 by whitequark