m-labs/artiq #206: print from kernels should be a regular RPC

sbourdeauducq, 2015-12-24:
It is more convenient to have experiments print directly to the console when run, instead of having this go in the core device log.

It is not clear how to handle the core device log in the long term. Maybe keep it as it is, send the messages to the master (either with the current TCP-based protocol used by ``artiq_ctlmgr`` or with UDP; UDP is easier on the device but will be affected by #39 on the master and is less tolerant of packet losses), or send the messages to whatever is connected to the device. In the meantime, ``print`` as a normal RPC is more user-friendly.

Peter - I know you spent a lot of time implementing formatting on the device. This work can be put to good use to print debug messages into the VCD output through RTIO.

whitequark, 2015-12-24:
So the thing is, print can't be a regular RPC because RPCs cannot have a varying number of arguments.

whitequark, 2015-12-24:
But it should be easy to add another type of packet just for debug output, and even keep all that print machinery...

sbourdeauducq, 2015-12-24:
Would it be excessively complicated to support variable number of arguments in RPCs? My compiler did that...

sbourdeauducq, 2015-12-24:
The print machinery should be kept and go into the VCD system IMO.

whitequark, 2015-12-24:
Actually, yes. Your compiler also didn't do typechecking, and since unification-based type systems do not easily admit varargs functions, it means a lot of parts to be changed... essentially RPCs have to become something entirely different from functions.

sbourdeauducq, 2015-12-24:
Supporting ``print`` RPC with only one argument would be OK in the meantime I think.

sbourdeauducq, 2015-12-26:
The function ``rtio_log`` now prints to the VCD. It is usable as follows:

```python

@syscall
def rtio_log(time_mu: TInt64, message: TStr) -> TNone:
    raise NotImplementedError("syscall not simulated")

    @kernel
    def run(self):
        rtio_log(now_mu(), "foo:Hello World")  # foo is the VCD channel name (dynamically created), the rest is the string printed to it
```

@whitequark can you:
1. hook up the print machinery to it, e.g. make the API something like ``self.rtio_log.log("vcd_channel_name_as_string", any_printable_object)``
2. make ``print`` a regular RPC, limited to one argument
3. remove the ``lognonl`` function from the runtime

Note: due to a bug in the analyzer core, for now make sure that all runs produce an even number of analyzer messages. As far as the log is concerned, an analyzer message contains 4 characters of payload, the terminal 0 counts as one, and the payload of one log entry is padded to a message boundary.

whitequark, 2015-12-26:
1. How about just `log` or `rtio_log`? Builtins (and the polymorphic print has to be a builtin) currently cannot be embedded functions.
2. If I make it a regular RPC it will not accept any arguments because we do not support `*args` in RPCs. I can make it a different builtin though, easily one that accepts any number of arguments.
3. Sure, but I will also rename `log` to something else, because it is reserved by C99.

jboulder, 2016-01-15:
Any update on this?

whitequark, 2016-01-15:
@sbourdeauducq Ping on questions in https://github.com/m-labs/artiq/issues/206#issuecomment-167349524?

sbourdeauducq, 2016-01-15:
1. Ok.
2. The compiler should not know anything about the signature of host functions. Unpacking RPC arguments is the job of CPython.
3. Ok.

whitequark, 2016-02-13:
@sbourdeauducq Should the name be `log` or `rtio_log` ?

sbourdeauducq, 2016-02-13:
From ARTIQ Python? ``log`` sounds good.

whitequark, 2016-02-15:
Done.

sbourdeauducq, 2016-02-15:
There should be documentation for this functionality.

sbourdeauducq, 2016-02-15:
This should insert ``vcd_channel_name:`` in the string so that the decoder script creates the channels. The API should be ``rtio_log("vcd_channel_name_as_string", any_printable_object)``.

sbourdeauducq, 2016-02-15:
``:`` may be replaced by a more special character such as ``\n`` so that users are less likely to accidentally put it in the channel name.

whitequark, 2016-02-15:
I'll use `U+001E RECORD SEPARATOR` https://en.wikipedia.org/wiki/C0_and_C1_control_codes#Field_separators instead of a colon.

sbourdeauducq, 2016-02-15:
I assume loads of UTF-8 bugs, if not complete lack of support, in most VCD viewers. But it's fine as long as this UTF-8 character is only ARTIQ internal.

whitequark, 2016-02-15:
It's represented the same as the corresponding ASCII character, just a `0x1E` byte. I am just used to referring to characters by their Unicode name.

whitequark, 2016-02-15:
Where do I put the docs?

sbourdeauducq, 2016-02-15:
I have no particular opinion on this, as long as it has some consistency with the rest of the manual.

whitequark, 2016-02-15:
Let me rephrase: I have no idea where to put them in the manual. Seems equally inconsistent in any place I've considered.

whitequark, 2016-02-15:
Ok, I added channel name and wrote some docs.

sbourdeauducq, 2016-06-26:
``print`` is still not a regular RPC. This is user-visible when it is called with multiple arguments:

```
$ artiq_run.py repository/core_pause.py 
Core Device Traceback (most recent call last):
  File "repository/core_pause.py", line 13, in __modinit__ (RA=+0x4a8)
    print(1, 2)
  File "<artiq>/coredevice/comm_generic.py", line 454, in _serve_rpc
    result = service(*args, **kwargs)
builtins.TypeError(8): coredevice_print() takes 1 positional argument but 2 were given
Traceback (most recent call last):
  File "/home/sb/artiq/artiq/frontend/artiq_run.py", line 210, in <module>
    main()
  File "/home/sb/artiq/artiq/frontend/artiq_run.py", line 206, in main
    return run(with_file=True)
  File "/home/sb/artiq/artiq/frontend/artiq_run.py", line 192, in run
    raise exn
  File "/home/sb/artiq/artiq/frontend/artiq_run.py", line 185, in run
    exp_inst.run()
  File "/home/sb/artiq/artiq/language/core.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/sb/artiq/artiq/coredevice/core.py", line 116, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "/home/sb/artiq/artiq/coredevice/comm_generic.py", line 545, in serve
    self._serve_exception(embedding_map, symbolizer, demangler)
  File "/home/sb/artiq/artiq/coredevice/comm_generic.py", line 529, in _serve_exception
    raise python_exn
TypeError: coredevice_print() takes 1 positional argument but 2 were given
```

commit ef2af8c3 (2016-06-28): compiler.embedding: use the builtin print as RPC.

Fixes #206.
commit 6dc510a9 (2016-06-29): compiler.embedding: use the builtin print as RPC.

Fixes #206.
closed 2016-06-28 by whitequark