m-labs/artiq #208: kernel with simple parallel/sequential blocks fails to return

sbourdeauducq, 2015-12-26:
```python
class CreateTTLPulse(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_inout")

    @kernel
    def run(self):
        with parallel:
            self.ttl_inout.gate_both_mu(1020)
            with sequential:
                delay_mu(10)
                self.ttl_inout.pulse_mu(1000)
```

This worked with my compiler. Removing the parallel/sequential does not crash.

whitequark, 2015-12-26:
Bizarre--the IR is completely linear

whitequark, 2015-12-26:
Reduced repro:
```python
from artiq import *

class X(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def b(self):
        print("x")

    @kernel
    def a(self, x):
        delay_mu(x)
        self.b()

    @kernel
    def run(self):
        print("a")
        with parallel:
            self.a(1000)
            delay_mu(10)
```

This is a problem with inliner.

commit 82ec76af (2015-12-26): compiler.types: fix TFunction internal field order (closes #208).
closed 2015-12-26 by whitequark