m-labs/artiq #209: TTL loopback is broken

sbourdeauducq, 2015-12-26:
This used to work with my compiler, but prints 0 now.

```python
def xprint(x):
    print(x)

class CreateTTLPulse(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_inout")

    @kernel
    def run(self):
        self.ttl_inout._set_sensitivity(3)
        self.ttl_inout.output()
        delay_mu(1000)
        self.ttl_inout.pulse_mu(10000)
        delay_mu(1000)
        self.ttl_inout._set_sensitivity(0)
        xprint(self.ttl_inout.count())
```

whitequark, 2015-12-26:
coreanalyzer says this:
```
OutputMessage(channel=0, timestamp=96912675656416, rtio_counter=96912673497328, address=2, data=3)
OutputMessage(channel=0, timestamp=96912675656416, rtio_counter=96912673500416, address=1, data=1)
OutputMessage(channel=0, timestamp=96912675657416, rtio_counter=96912673506440, address=0, data=1)
OutputMessage(channel=0, timestamp=96912675667416, rtio_counter=96912673509992, address=0, data=0)
```
Is this the correct message sequence?

sbourdeauducq, 2015-12-26:
Actually no and this isn't a compiler problem, but a corner case with the RTIO core. It needs a delay between ``output`` and ``set_sensitivity`` as it can only process command per cycle; it works after adding it. AFAICT no real-world experiment will do this, so I don't think this case needs better handling.

closed 2015-12-26 by sbourdeauducq