m-labs/artiq #212: artiq_flash.sh not working with kc705

dleibrandt, 2015-12-30:
Installed artiq using conda following instructions in the manual, and am attempting to flash a kc705.  Command "artiq_flash.sh -m qc2" fails with error message below.  I've repeated about 5 times and always see the same error message (but the page it fails on is different).

Flashing FPGA bitstream...
XC3SPROG (c) 2004-2011 xc3sprog project $Rev: 774 $ OS: Linux
Free software: If you contribute nothing, expect nothing!
Feedback on success/failure/enhancement requests:
	http://sourceforge.net/mail/?group_id=170565 
Check Sourceforge for updates:
	http://sourceforge.net/projects/xc3sprog/develop

Using built-in device list
Using built-in cable list
Cable jtaghs1_fast type ftdi VID 0x0403 PID 0x6010 Desc "Digilent Adept USB Device" dbus data 80 enable 8b cbus data 00 data 00
Using Libftdi, Using JTAG frequency  30.000 MHz from undivided clock
JTAG chainpos: 0 Device IDCODE = 0x43651093	Desc: XC7K325T
Created from NCD file: bscan_spi_kc705-routed.ncd;UserID=0xFFFFFFFF
Target device: 7k325tffg900
Created: 2015/04/09 17:41:06
Bitstream length: 91548896 bits
done. Programming time 3431.5 ms
JEDEC: 20 ba 0x18 0x10
Found Numonyx N25Q Device, Device ID 0xba18
CFI: 00000000000000000000000000000000 
256 bytes/page, 65536 pages = 16777216 bytes total 
Created from NCD file: top;UserID=0XFFFFFFFF
Target device: 7k325tffg900
Created: 2015/12/26 23:02:01
Bitstream length: 91548896 bits
Erasing sector 175/175..Writing data page  44701/ 44702 at flash page  44701..
Maximum erase time 669.0 ms, Max PP time 66896 us
Verifying page  24974/ 44702 at flash page  24974
Verify failed  at flash_page  24975
read:0000400000000000000000000000000200080000000000000000000040004000100000000000400000000000000040000040000040000000000000000000000000000808000000000800020000000200000000000000000018000200020802000880000002002000088802800200000488000080020002080000008000000000000000000000000428000000200400082808000002002a080000000000000800280000000000020000080000000000000800080000000200080800002000020000002000000020000800000400000008001800002000000000000000010000000002000000000000000000002000200000040000200020002800008002000000
file:0000400000000000000000000000000200080000000000000000000040004000100000000000400000000000000040000040000040000000000000000000000000000808000000000800020000000200000000000000000018000200020802000880000002002000088802800200000488000080020002080000008000000000000000000000000428000000200400082808000002002a080000000000000800280000000000020000080000000000000800080000000200080800002000020000002000000020000800000400000008001800002000000000000000000000000002000000000000000000002000200000040000200020002800008002000000
...
USB transactions: Write 211546 read 205953 retries 219830

sbourdeauducq, 2015-12-31:
Someone should switch this script to openocd (@jordens ?), in the meantime can you use another kc705 that does not tickle this likely xc3sprog bug? (This is also happening on the Pipistrello, but I had not seen with the KC705 before)

sbourdeauducq, 2015-12-31:
See #103

dleibrandt, 2015-12-31:
Tried a second kc705 and artiq_flash.sh works.  Thanks.  For the record, both kc705s were new out of the box.

dleibrandt, 2015-12-31:
Next problem: when I try to run the led.py demo, I get the warning below and no leds change status.

WARNING:worker(36):artiq.coredevice.comm_generic:Mismatch between gateware (0.0+2302.g50a463a) and software (0+unknown) versions

sbourdeauducq, 2015-12-31:
Did you flash qc1 or qc2? If using qc2 you need to update device_db.

dhslichter, 2016-01-03:
On a related note, it's probably time to make qc2 the default configuration for the KC705.  This is obviously not a big issue for people to change the device_db, but going forward I don't think there will really be any users of the qc1 hardware with ARTIQ and there are at least two setups being put together right now with the qc2 hardware.

jordens, 2016-01-04:
I'll put porting that to openocd on my todo list. @whitequark, could you package openocd? There will be a few dependencies but it seems doable. Also https://github.com/timvideos/conda-hdmi2usb-packages/tree/master/openocd

dleibrandt, 2016-01-04:
I flashed qc2.  I just tried updating the device_db, and now I can turn the led on and off, but I still get the same mismatch warning message.  Should I worry about this?

Also, I agree with @dhslichter's comment above.  Additionally, it would be nice to add qc2 to the board ports man page (http://m-labs.hk/artiq/manual/core_device.html#board-ports).

jordens, 2016-01-05:
This should do it: [RFC] https://github.com/nist-ionstorage/artiq 
Packaging openocd does not seem necessary since we didn't package xc3sprog either (afaict).

sbourdeauducq, 2016-01-05:
@dleibrandt the warning message is due to #199 and you can safely ignore it.

sbourdeauducq, 2016-01-05:
@jordens Thanks, looks good, merged it.
@dleibrandt Can you try flashing the problematic KC705 with openocd/the new script and see if it works?

sbourdeauducq, 2016-01-05:
@jordens Your script does not look for flash proxy bitstreams where the conda package puts them.
https://github.com/m-labs/artiq/commit/57ae630c5c2059897c0cdbf8f16b3535486e2077

jordens, 2016-01-05:
The old one didn't. I am pretty sure mine checks that location first (https://github.com/m-labs/artiq/blob/master/artiq/frontend/artiq_flash.py#L83). But will check.

sbourdeauducq, 2016-01-06:
Yes, just tested and confirmed it works, but the script's help message doesn't mention this.

dleibrandt, 2016-01-06:
Yes, it also works on the board that didn't work with xc3sprog.  Thanks.

closed 2016-01-06 by jordens