m-labs/artiq #214: conda install fails on windows

dhslichter, 2016-01-06:
artiq installation on windows using conda fails with the following error message:
```
    c:\>conda install -c https://conda.anaconda.org/m-labs artiq
    Fetching package metadata: ..........
    Solving package specifications: ..............................
    Error: Could not find some dependencies for artiq: flterm, levenshtein
```
I thought that flterm had been removed?  And is there a win32 build of levenshtein?

sbourdeauducq, 2016-01-07:
Yes, flterm has been removed and I just double-checked that the latest uploaded package does not depend on it.

sbourdeauducq, 2016-01-07:
I checked ``noarch-artiq-0.0-py_2370+git73b4aff.tar.bz2`` to be precise.

whitequark, 2016-01-07:
I think there was some other issue with artiq on windows, quamash maybe? Anyhow, the reason it's broken is that Appveyor randomly decided to stop working and I haven't yet replaced it with Buildbot because Windows 8 doesn't work in a Xen VM.

dhslichter, 2016-01-08:
Currently it appears the issue is the levenshtein package, for which there is no win32 conda build available anywhere.  The latest build of artiq does appear to not depend on flterm, so that is an improvement.  For reference/duplication, there is no MSVC installed on this computer, and it needs to be possible to install on windows without having to install the full MSVC.

dhslichter, 2016-01-08:
Also, for reference we are running on Windows 7, and in fact Windows 8 is banned at NIST so it would probably be best to have Windows 7 in your VM for builds.

whitequark, 2016-02-08:
There is now levenshtein and Qt 5 on Windows but not yet binutils.

sbourdeauducq, 2016-02-09:
The Qt5 package does not work. ``from PyQt5 import QtCore`` results in ``DLL load failed``.

whitequark, 2016-02-11:
I've just set up a fresh miniconda install and the command you talk about works.

whitequark, 2016-02-11:
I verified that `from PyQt5 import QtCore` executes successfully on both win32 and win64.

jordens, 2016-02-11:
This is current artiq master with (AFAICT) the correct packages:

```
(py35)rj@tito:~/work/nist/artiq/examples/master (master)$ conda install qt5 pyqt5
Fetching package metadata: ............
Solving package specifications: ..............
# All requested packages already installed.
# packages in environment at /home/rj/src/conda/envs/py35:
#
pyqt5                     5.5.1                    py35_0    m-labs
qt5                       5.5.1                         0    m-labs
(py35)rj@tito:~/work/nist/artiq/examples/master (master)$ artiq_gui
This application failed to start because it could not find or load the Qt platform plugin "xcb".

Reinstalling the application may fix this problem.
Aborted (core dumped)
(py35)rj@tito:~/work/nist/artiq/examples/master (master)$
```

whitequark, 2016-02-11:
I was able to run artiq_gui on Windows just now. I did this by executing the following first

    QT_QPA_PLATFORM_PLUGIN_PATH=%QTDIR%\plugins\platforms\

where `%QTDIR%` is `c:\miniconda\envs\artiq\Library\lib\qt5`. @jordens, I imagine the same measure will help on Linux too. I will investigate why is this needed.

whitequark, 2016-02-11:
Ok, this is described in https://doc.qt.io/qt-5/windows-deployment.html and is basically the same for both OSes:

>The search path for Qt plugins is hard-coded into the QtCore library. By default, the plugins subdirectory of the Qt installation is the first plugin search path. However, pre-determined paths like the default one have certain disadvantages. For example, they may not exist on the target machine. For that reason, you need to examine various alternatives to make sure that the Qt plugins are found:
>
> * Using qt.conf. This approach is the recommended if you have executables in different places sharing the same plugins.
> * Using QApplication::addLibraryPath() or QApplication::setLibraryPaths(). This approach is recommended if you only have one executable that will use the plugin.
> * Using a third party installation utility to change the hard-coded paths in the QtCore library.

I will update artiq_gui to do addLibraryPath, I think.

whitequark, 2016-02-11:
Ah, no, no need. The package already has a qt.conf installed in an appropriate place but whoever wrote the recipe it did not set the prefix correctly.

sbourdeauducq, 2016-02-11:
@whitequark 
[image]

It's on VNC right now if you want to have a look.

sbourdeauducq, 2016-02-11:
Solution to this problem is to use conda's activate script.

whitequark, 2016-02-11:
This is actually what the manual says anyway.

whitequark, 2016-02-14:
conda install will now work on Windows.

closed 2016-02-14 by whitequark