m-labs/artiq #215: core device should be capable of sending 1MB in one RPC

sbourdeauducq, 2016-01-07:
See _PayloadCoreSend1MB in speed_benchmark.py. Fails with this log since the new comms code:
```
session.c: write overrun while trying to write 4 bytes (2 remaining)
Failed to send RPC request (service 2, tag Oli:n)
```

sbourdeauducq, 2016-01-07:
Also, we should probably unittest this.

whitequark, 2016-01-07:
Do you suggest just increasing buffer size, or something more clever? It looks like the coredevice code won't yield to the latter without some serious refactoring.

sbourdeauducq, 2016-01-07:
Then increase buffer size.

sbourdeauducq, 2016-01-07:
But at some point the comms code will have to handle 10MB chunks of data to load into core device storage...

whitequark, 2016-01-07:
This will be hard to implement. The crux of the issue is that the entire `session.c` has to be rewritten in using coroutines. C is not a language suited for doing this without allocation, quite ironically if you consider how it's supposed to be good for embedded work...

whitequark, 2016-01-07:
`receive_rpc_value` and `send_rpc_value` are the worst offenders because they are recursive, potentially with unlimited depth.

commit 027d54ca (2016-01-07): Enlarge coredevice buffers to 2.5MiB (fixes #215).

This should be enough for sending a 2MiB int32 list.
closed 2016-01-07 by whitequark