m-labs/artiq #216: Kick off buildbot on github push

whitequark, 2016-01-07:


whitequark, 2016-01-07:
@sbourdeauducq I can't do this, since I don't have admin access to m-labs repositories. Hypothetically it should be enough to point a custom hook to `http://101.78.236.68/buildbot/change_hook/github?project=PROJECT` where `PROJECT` is `artiq`, `misoc`, `migen`.

sbourdeauducq, 2016-01-08:
The server seems to return HTTP 500 to github.

whitequark, 2016-01-08:
Hm:
```
2016-01-08 11:25:26+0800 [HTTPChannel,13,] Attempting to load module buildbot.status.web.hooks.github
2016-01-08 11:25:26+0800 [HTTPChannel,13,] processing changes from web hook
        Traceback (most recent call last):
        Failure: exceptions.KeyError: 'payload'
```

sbourdeauducq, 2016-01-08:
Hmm maybe it works for actual pushes...

whitequark, 2016-01-08:
Yeah, I recall something like that. Try it.

sbourdeauducq, 2016-01-08:
Didn't work, still 500.

sbourdeauducq, 2016-01-08:
Should I use json content type or x-www-form-urlencoded?

whitequark, 2016-01-08:
`application/x-www-form-urlencoded`.

sbourdeauducq, 2016-01-08:
OK that works now, can you disable auto build for non-master branches?

whitequark, 2016-01-08:
Done

closed 2016-01-08 by whitequark