m-labs/artiq #217: AD9914 output frequency is half of the set value

dleibrandt, 2016-01-07:
I'm running artiq on kc705 with qc2 hardware.  I have a couple AD9914s, which are clocked at 3.5 GHz.  When I set the output frequency as follows

    self.dds11.init()
    delay(10*ms)
    self.dds11.set(200*MHz, amplitude=1.00)

, the actual output frequency is 100 MHz.

sbourdeauducq, 2016-01-08:
Have you set the correct sysclk in device_db?

dleibrandt, 2016-01-08:
Yes, here is the relevant entry in the device_db.  I've also checked on a spectrum analyzer that the actual clock going into the DDSs is 3.5 GHz.

    "dds11": {
        "type": "local",
        "module": "artiq.coredevice.dds",
        "class": "AD9914",
        "arguments": {"sysclk": 3.5e9, "channel": 1}
    },

sbourdeauducq, 2016-01-09:
Did you run ``dds11.init()`` at least once after power-up (e.g. at the beginning of your kernel, in another kernel, via the startup kernel)?

sbourdeauducq, 2016-01-09:
Does this happen with all DDS cards or only one?

dleibrandt, 2016-01-13:
@whitequark The root cause of this problem is your compiler breaks when computing the frequency tuning word. This is easily reproduced by running ``print(2**32*400e6/3.5e9)``: prints 245426702.514285 on the device, 490853405.25714284 in CPython.

whitequark, 2016-01-14:
@dleibrandt Yes, that's actually expected, since all integers are by default 32-bit. (Arguably it was a bug in the old compiler that by accident provided correct result.) This prints the correct value:

    print(int(2, width=64)**32*400e6/3.5e9)

sbourdeauducq, 2016-01-14:
It was not an accident, but requiring explicit size on all integers that require more than 32 bits is OK.

sbourdeauducq, 2016-01-14:
a9cf892 should fix the problem. @dleibrandt can you test?

whitequark, 2016-01-14:
@sbourdeauducq I'm aware that it worked because constant folding happened before inference of integer size. I still consider that a bug from language design perspective, because otherwise algebraically equivalent transformations would break code relying on that.

dleibrandt, 2016-01-15:
Yes, I just tested and the DDSs now output the correct frequency.

closed 2016-01-15 by whitequark